lvgl/lvgl #4960: Changing screens results in crash in version 8.3.11

supremeneuron, 2023-12-08:
### Perform all steps below and tick them with [x]
- [X ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [X ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ X] Update lvgl to the latest version
- [ X] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

In my code, I dynamically create and then delete screens as needed.  This process has worked flawlessly all the way up to version 8.3.10, but for some reason in version 8.3.11.  The system now crashes on the following piece of code.  

For reference, the coding is done in Arduino 2.21 and is compiled for the ESP32C3 using 2.0.14 board manager module.
### To Reproduce

### Expected behavior

I would expect that the screen changes would happen without crashing like they did with versions 8.3.10 and before.

### Screenshots or video

supremeneuron, 2023-12-08:
Here is a snippet that seems to cause issues....  It is a little much but this is where the whole thing crashes.

void setup()
{ // ..... basic hardware setup...
// in lv_conf.h #define LV_TICK_CUSTOM 1

   screen = lv_obj_create(NULL);
   lv_scr_load(screen);
   lv_example_tabview_1(); 

}

static void event_handler(lv_event_t * e)
{
    lv_event_code_t code = lv_event_get_code(e);
    lv_obj_t * obj = lv_event_get_target(e);
    if(code == LV_EVENT_CLICKED) 
    { 
       for (byte i=0;i<KEY_COUNT;i++)
          {
           if (strcmp(lv_list_get_btn_text(list1, obj), buttonlabels[i])==0)   
               { 
                Launch_app_id=i;
              }
          }
    }
}

void lv_example_tabview_1(void)
{
    /*Create a Tab view object*/
    tabview = lv_tabview_create(lv_scr_act(), LV_DIR_TOP, 50);

    /*Add 3 tabs (the tabs are page (lv_page) and can be scrolled*/
    lv_obj_t *tab1 = lv_tabview_add_tab(tabview, "Apps");
    lv_obj_t *tab3 = lv_tabview_add_tab(tabview, "About");
 
    /*Add content to the tabs*/
    lv_obj_t * label = lv_label_create(tab3);
    lv_label_set_text(label, "Proof of concept\n\n"
                             "using LVGL\n"
                             );

    //Create a list
    list1 = lv_list_create(tab1); 
    lv_obj_set_size(list1, 280, 320);
    lv_obj_center(list1);

    //Add buttons to the list
    lv_obj_t * btn;
 
    lv_list_add_text(list1, "Apps");
    
    for (byte i=0;i<KEY_COUNT;i++)
  {
    btn = lv_list_add_btn(list1, buttonicons[i], buttonlabels[i]);
    lv_obj_add_event_cb(btn, event_handler, LV_EVENT_CLICKED, NULL);  //add click event handler for each item in the list
  }
}
  
void loop()
{  
   lv_timer_handler(); /* let the GUI do its work */
   yield();  
 
   switch (ShowSelection){
 
    case FILEMANAGERSTART_FLAG:
      FileManager();  
      break;
    case FILEMANAGER_FLAG:  //This seems to be where the crashing occurs
      {lv_obj_del(screen1);
        screen = lv_obj_create(NULL);
        lv_scr_load(screen);
        FileManager();
      } 
       break;
    default:
        break;
   }
}

void FileManager(void)
  { 
  lv_obj_del(screen);
  screen1 = lv_obj_create(NULL);
  lv_scr_load(screen1);
  UpdateDirectoryList();
  }

void UpdateDirectoryList (void)
{   
  ShowBasic=IDLEMENU_FLAG; //clear the flag so we are not in a loop showing the directory...actually just set to an idle flag
  DisplayLoadingMessage();
  ShowDirectory();
  RemoveLoadingMessage();
}

void DisplayLoadingMessage(void)
{
  //Create a label that is superimposed on everything else to show that we are busy
   activity_label= lv_label_create(lv_layer_top());
    lv_obj_set_align(activity_label, LV_ALIGN_CENTER); 
    lv_label_set_text(activity_label,"Loading...");
    lv_task_handler();  // now take a moment to display the requested screen change
}
void RemoveLoadingMessage(void)
{
 //we are done so delete this object
   //because we are return to regular activity, there is no need to call the task handler
  lv_obj_del(activity_label);
}

  void ShowDirectory(void){
    
    //Show the files on the SD card
    //Create a list
    list3 = lv_list_create(lv_scr_act());   
    lv_obj_set_size(list3, LV_HOR_RES, LV_VER_RES);
    lv_obj_center(list3);
      //add a gesture call back
   lv_obj_add_event_cb(screen1, gesture_cb_event, LV_EVENT_GESTURE, NULL);
    //Add buttons to the list
    lv_obj_t * btn;

        lv_list_add_text(list3,CurrentDirectory);
}

...  I hope that offers some insights in the issue.  It *seems* that the creation and deletion of screens is where the issues surface; otherwise, things run smoothly as they always have.

Thanks...

supremeneuron, 2023-12-10:
After spending fair amount of time on this matter, I discovered that the problem appears to have been that I was deleting a screen before the new screen was finalized and active.  By simply moving the object delete to AFTER a new screen was created and active, there were no more crashes.  What I don't understand is how this worked for all the prior versions and failed on version 8.3.11, I don't know, but what I do know is that the code is now back to being functional.

ORIGINAL:
lv_obj_del(screen);
screen1 = lv_obj_create(NULL);
lv_scr_load(screen1);

CORRECTED (working):
screen1 = lv_obj_create(NULL);
lv_scr_load(screen1);
lv_obj_del(screen);

closed 2023-12-10 by supremeneuron

kisvegabor, 2023-12-10:
Deleting the active screen is really not a good idea. 

LVGL should print a warning log about it, if logging is enabled.