lvgl/lvgl #4966: About Multi Touch

IAMMX, 2023-12-09:
### Introduce the problem

Will LVGL consider supporting multi touch? For example, achieving dual finger scaling of images.
### Examples and cases

### Suggested solution

kisvegabor, 2023-12-11:
It's not on the near roadmap, but we might consider it in the future.

When v9 will be released (mid January) we will have a discussion to figure out the roadmap. Feel free to join and tell what feature you would like to see in LVGL.

closed 2023-12-12 by IAMMX

AZ-Renesas, 2024-06-20:
Hello,
As I understand, multi-touch is not currently supported in the latest version, right ?

kisvegabor, 2024-06-21:
Yes, it's not supported now. With @becseya we started to discuss it, but it's not on the near roadmap now. Would it be important for you or a customer?

AZ-Renesas, 2024-06-21:
I would consider using it on a demo we are working on EK-RA8D1.

kisvegabor, 2024-06-21:
What kind of multi touch gesture would be required for you?

AZ-Renesas, 2024-06-21:
zoom in and out with 2 fingers

kisvegabor, 2024-06-24:
I see, we will discuss it with @becseya tomorrow.

kashperanto, 2024-08-26:
Hey, has there been any progress on this?

I have a simple use case for a two-button press to unlock a screen.  This kind of thing (pressing some button combination to unlock something) is a relatively common feature on simple UIs in my experience, and it would be good to support for cases where we need to emulate the behavior of an existing "base model" interface that only has membrane keys (no touch screen input).

In the meantime, would there be a recommended way to work around this?  My plan is to filter out 2+ touch detections by my driver, and then detect and generate events in LVGL manually (I only have one such input to handle, so this is manageable).

kisvegabor, 2024-08-29:
I've added it to my LVGL v9.3 todo list. 

However for simple 2 button touch you can create 2 `lv_indev`s and use the first finger coordinates in one, and the second finger for the other. After that just create 2 `lv_button`s and see if they are pressed at the same time.

kashperanto, 2024-08-29:
@kisvegabor Cool, I'd be happy to help test it out when that time comes.  Also, thanks for the suggestion of two `lv_indev`instances.  That makes a lot more sense than what I was considering.

TheElectronicMusicProject, 2025-04-11:
> I've added it to my LVGL v9.3 todo list.
> 
> However for simple 2 button touch you can create 2 `lv_indev`s and use the first finger coordinates in one, and the second finger for the other. After that just create 2 `lv_button`s and see if they are pressed at the same time.

So when you say that multi touch is not supported yet, do you mean gestures? Because I understand that more buttons at the same time are detected (not tried), hence two sliders at the same time are supported too?

kisvegabor, 2025-04-15:
The multi touch gesture support is added, see here: https://docs.lvgl.io/master/details/main-modules/indev.html#multi-touch-gestures

@TheElectronicMusicProject 
For a simpler case you just create multiple indevs for multiple fingers and touch multiple widgets at the same time.