lvgl/lvgl #4967: can't decode jpg while use path file （version 8.3.7）

sam131208, 2023-12-09:
### Describe the bug
 I'm using lvgl on zephyr, the file system is littlefs, the file path is /lfs/test.jpg,  lv_img_set_src(image, "/lfs/test.jpg").

 LVGL returns LV_RES_INV by calling lv_img_decoder_open() in _lv_img_cache_open().

 The problem is that this code in lv_img_decoder_open() polls all the decoders:
```
_LV_LL_READ(&LV_GC_ROOT(_lv_img_decoder_ll), decoder) {
        /*Info and Open callbacks are required*/
        if(decoder->info_cb == NULL || decoder->open_cb == NULL) continue;

        res = decoder->info_cb(decoder, src, &dsc->header);
        if(res != LV_RES_OK) continue;

        dsc->decoder = decoder;
        res = decoder->open_cb(decoder, dsc);

        /*Opened successfully. It is a good decoder for this image source*/
        if(res == LV_RES_OK) return res;

        /*Prepare for the next loop*/
        lv_memset_00(&dsc->header, sizeof(lv_img_header_t));

        dsc->error_msg = NULL;
        dsc->img_data  = NULL;
        dsc->user_data = NULL;
        dsc->time_to_open = 0;
    }
``` 
  It first uses the decoder created by lv_split_jpeg_init() is ok, then the decoder created by _lv_img_decoder_init(), and in decoder->info_cb(decoder, src, &dsc->header), jpg != bin, returns LV_RES_INV.
  From this, lv_img_decoder_open() returns the LV_RES_INV and ends the decoding.

### Expected behavior
 The first time I used lvgl to decode an external image file, I'm not sure if it's this issue that causes the image to not display. But it's clearly a bug?

### Screenshots or video
<wrn> lvgl: (2.415, +0)       _lv_img_cache_open: Image draw cannot open the image resource  (in lv_img_cache.c line #125)
<wrn> lvgl: (2.415, +0)       lv_draw_img: Image draw error  (in lv_draw_img.c line #83)

laiwanz, 2023-12-15:
JPG images will not be unzipped if you perform additional operations on the image, such as zooming. Try showing only the image

sam131208, 2023-12-15:
Here's the code I use to display the image, which should just show the image:
```c
	lv_obj_t *image = lv_img_create(lv_scr_act());

	lv_img_set_src（image， "/lfs/test.jpg"）
	lv_obj_set_width(image, LV_SIZE_CONTENT);
	lv_obj_set_height(image, LV_SIZE_CONTENT);
	lv_obj_set_align(image, LV_ALIGN_CENTER);
	lv_obj_add_flag(image, LV_OBJ_FLAG_ADV_HITTEST);
	lv_obj_clear_flag(image, LV_OBJ_FLAG_SCROLLABLE);
``` 
  I'm not quite sure if the question raised above is the real reason that is preventing the image from showing.
  I'm still struggling with how to display the image correctly.

kisvegabor, 2023-12-15:
It seems you have not enabled LVGL's file system interface. 
See 
- https://docs.lvgl.io/8.3/overview/file-system.html
- https://docs.lvgl.io/8.3/libs/fsdrv.html

closed 2024-01-06 by lvgl-bot