lvgl/lvgl #4980: Migrate to C11(or use gnu99) to support `anonymous union`

XuNeo, 2023-12-11:
### Introduce the problem

Anonymous union makes code looks cleaner. It's supported in c11 or `gnu99` extension but not current `c99`.
Need to discuss the drawbacks to make such change.

### Examples and cases

### Suggested solution

kisvegabor, 2023-12-11:
I'm ok with it in general, but we need to check it with
- Arduino
- Microchip's XC16 compiler (last time I tried it was supporting some C features)
- MSVC

XuNeo, 2023-12-11:
For XC16, it does mention gnu99 is added. 
https://ww1.microchip.com/downloads/aemDocuments/documents/DEV/ProductDocuments/SoftwareTools/Readme_XC16.html

For others, it needs some time to confirm.

XuNeo, 2023-12-29:
Not stale.
Current code is NOT C99 compliant.

closed 2024-01-20 by lvgl-bot