lvgl/lvgl #4982: mem leak of x-axis values in chart for 8.3 branch

pfyra, 2023-12-11:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
Function lv_chart_remove_series does not free the memory allocated for the x-axis on the 8.3 branch. Problem is fixed on master branch.
Adding the line:
if(!series->x_ext_buf_assigned && series->x_points) lv_mem_free(series->x_points);
to:
https://github.com/lvgl/lvgl/blob/74d0a816a440eea53e030c4f1af842a94f7ce3d3/src/extra/widgets/chart/lv_chart.c#L391
fixes the problem.
The problem might exist on other branches as well, but not on master.

### To Reproduce
call lv_chart_remove_series  on 8.3 branch (perhaps a scatter chart is required)

### Expected behavior
memory is freed

### Screenshots or video
n/a

kisvegabor, 2023-12-11:
Thank you for the report! Could you send a Pull request?

pfyra, 2023-12-13:
Hello,

Sorry for the delay... Here is my pull request:
https://github.com/lvgl/lvgl/pull/5001

// Peter

kisvegabor, 2023-12-13:
The PR is merged, thank you! :+1:

closed 2023-12-30 by kisvegabor