lvgl/lvgl #4986: Issue with using the mouse and using the wheel to scroll.

kdschlosser, 2023-12-11:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
If you use the mouse and hover over the vertical scrolling list doesn't matter if you click to give focus or not if you use the house wheel to scroll everything that is scrollable will scroll not just the one thing that is being hovered over or has focus,

https://docs.lvgl.io/master/examples.html#flex

### To Reproduce
https://docs.lvgl.io/master/examples.html#flex

### Expected behavior
https://docs.lvgl.io/master/examples.html#flex

### Screenshots or video
https://docs.lvgl.io/master/examples.html#flex

kisvegabor, 2023-12-11:
Mouse wheel scrolling is not supported in LVGL yet, but it being discussed in https://github.com/lvgl/lvgl/issues/4909

kdschlosser, 2023-12-11:
well it's doing something with the examples...

kisvegabor, 2023-12-12:
Now the mouse wheel acts as an [ENCODER input device](https://docs.lvgl.io/8.3/porting/indev.html#encoder). Scrolling it changes the focus.

kdschlosser, 2023-12-27:
not stale

closed 2024-01-11 by kdschlosser