lvgl/lvgl #4992: Build error in cmsis-pack/lv_os_custom.c

kisvegabor, 2023-12-12:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

This [line](https://github.com/lvgl/lvgl/blob/master/env_support/cmsis-pack/lv_os_custom.c#L9) is causing build error in Eclipse. All source files are being compiled by Eclipse managed build and the linked header file is not found. 

### To Reproduce

Try https://github.com/lvgl/lv_port_pc_eclipse with the latest LVGL master.

@GorgonMeducer Is this file really required once we have `LV_OS_CMSIS_RTOS2` support?

GorgonMeducer, 2023-12-12:
This is a template for people to add their own RTOS support. 
Don't worry, I will fix this.

GorgonMeducer, 2023-12-12:
@kisvegabor  Let me explain how the lv_os_custom works:

When people using LVGL9 cmsis-pack and open the RTE window in MDK, the OS Abstraction Layer will be shown as a drop list:

[image]

If people cannot find the (RT)OS implementation in the list, for example, ucOS or some home-made ones:

[image]

They can choose the option "User Custom", and select the `OS Abstraction Layer`. By doing so, 

- A copy of the `lv_os_custom.c` and `lv_os_custom.h` will be added to the project
- Macro will be added to `RTE_Components.h` (which is included by `lv_conf_cmsis.h`):
```c
/* LVGL::LVGL&LVGL9:OS Abstraction Layer&User Custom@9.0.0 */
/*   Select an operating system to use. Possible options:
 * - LV_OS_NONE
 * - LV_OS_PTHREAD
 * - LV_OS_FREERTOS
 * - LV_OS_CMSIS_RTOS2
 * - LV_OS_RTTHREAD
 * - LV_OS_WINDOWS
 * - LV_OS_CUSTOM 
 */
#define LV_USE_OS               LV_OS_CUSTOM
#define LV_OS_CUSTOM_INCLUDE    "lv_os_custom.h"
```

So, our users can modify the `lv_os_custom.c` and `lv_os_custom.h` to add their own OSAL implementation.

kisvegabor, 2023-12-12:
Makes sense. Thank you for the explanation and the fix.

GorgonMeducer, 2023-12-12:
You are welcome.