lvgl/lvgl #4993: In v9 Parent and child elements of the same size, but the child element cannot completely cover the parent element

HongChao6, 2023-12-12:
### Describe the bug
Set the size of the parent element to 200X200, the background to blue, 
the child element to a 100X100 picture, the child element is scaled to 200X200, 
As a result, the child element does not completely cover the parent element.

```
void lv_image_test(void){
   int ww = 200;
   lv_obj_t* native = lv_obj_create(lv_scr_act());
   lv_obj_remove_style_all(native);
   lv_obj_set_width(native,ww);
   lv_obj_set_height(native,ww);
   lv_obj_set_pos(native, 100, 50); 

   lv_image_t* img_ = lv_img_create(native);
   lv_obj_remove_style_all(img_);
   lv_obj_center(img_);
   lv_img_set_src(img_, "/data/app/com.swiper.test/common/00.jpeg");
   lv_obj_set_width(img_,ww);
   lv_obj_set_height(img_,ww);

   lv_obj_set_style_bg_color(img_, lv_palette_main(LV_PALETTE_BLUE), LV_PART_MAIN);     
   lv_obj_set_style_bg_opa(img_, LV_OPA_80, LV_PART_MAIN); 

   uint16_t zoomHeight = ww * LV_SCALE_NONE / img_->h;
   img_->zoom = zoomHeight; 
  lv_img_set_size_mode(img_, LV_IMAGE_SIZE_MODE_REAL);
}
```

00.jpeg
[image]

### Expected behavior
The child element completely covers the parent element.

### Screenshots or video
[image]

XuNeo, 2023-12-12:
Could you create a minimal reproducible example on LVGL simulator?

sam0910, 2023-12-12:
Dear @HongChao6, sorry about out of issues, 
Could I ask you what kind of watch is running LVGL?

HongChao6, 2023-12-13:
This example is run on lvgl simulator
[image]

HongChao6, 2023-12-13:
> Dear @HongChao6, sorry about out of issues, Could I ask you what kind of watch is running LVGL?

That's just a picture, not running on the watch

kisvegabor, 2023-12-13:
Is it the same in v9 (`master` branch?)

XuNeo, 2023-12-13:
Yes, that can be reproduced in latest master branch with lvgl's simulator.

kisvegabor, 2023-12-14:
I could also reproduce it and will get back to this in January when hammering out v9 issues.

kisvegabor, 2023-12-18:
I've tried to fix it but I found that here 2 independent issues:

### Issue 1
LVGL calculated the zoom with a minor inaccuracy. It can be fixed by this:
```patch
diff --git a/src/draw/lv_image_buf.c b/src/draw/lv_image_buf.c
index 7f5f161cc..551c15db2 100644
--- a/src/draw/lv_image_buf.c
+++ b/src/draw/lv_image_buf.c
@@ -73,18 +73,18 @@ void _lv_image_buf_get_transformed_area(lv_area_t * res, int32_t w, int32_t h, i
 
     lv_point_t p[4] = {
         {0, 0},
-        {w - 1, 0},
-        {0, h - 1},
-        {w - 1, h - 1},
+        {w, 0},
+        {0, h},
+        {w, h},
     };
     lv_point_transform(&p[0], angle, scale_x, scale_y, pivot, true);
     lv_point_transform(&p[1], angle, scale_x, scale_y, pivot, true);
     lv_point_transform(&p[2], angle, scale_x, scale_y, pivot, true);
     lv_point_transform(&p[3], angle, scale_x, scale_y, pivot, true);
     res->x1 = LV_MIN4(p[0].x, p[1].x, p[2].x, p[3].x);
-    res->x2 = LV_MAX4(p[0].x, p[1].x, p[2].x, p[3].x);
+    res->x2 = LV_MAX4(p[0].x, p[1].x, p[2].x, p[3].x) - 1;
     res->y1 = LV_MIN4(p[0].y, p[1].y, p[2].y, p[3].y);
-    res->y2 = LV_MAX4(p[0].y, p[1].y, p[2].y, p[3].y);
+    res->y2 = LV_MAX4(p[0].y, p[1].y, p[2].y, p[3].y) - 1;
 }
```

### Issue 2
There are all sort of minor rounding error in the SW render. It'd require more time to better understand how to approach it, but if you are using a GPU for image transformation it should work. I suggest adding these to the image to reduce the possible rounding errors:
```c
   lv_image_set_align(img_, LV_IMAGE_ALIGN_TOP_LEFT);
   lv_image_set_pivot(img_, 0, 0);
```

XuNeo, 2023-12-18:
Hi Gabor,
Thank you for your quick response. We have tried solution for issue 1, and yes, it still has issue. So when there's no rotation, how the rounding error could happen? 

Say the image is 100x100, scale x2 should give us exactly 200x200. 
The method use for zoom is get transformed all four points (0,0), (99,0) .... 
Thus `99,0` is transformed to `198,0`. So the width becomes 198-0+1 = 199. Give us 1px error. I don't understand what's wrong here.

kisvegabor, 2023-12-18:
Do you see the issue with GPU rotation too? If so, do you see any strange numbers in the transformed area?

XuNeo, 2023-12-18:
This issue is reported on CPU only hardware. Thus no GPU is related.

kisvegabor, 2023-12-18:
It means we need to deeply look into the SW image transformation. I'll try fixing it again.

kisvegabor, 2023-12-18:
I should be fixed. Finally I added a special handling to image scale only (no rotation).

closed 2023-12-18 by kisvegabor

kisvegabor, 2023-12-18:
I've already fixed it: https://github.com/lvgl/lvgl/commit/7b9f9a9f9e5c5e57077e09e7549889df3520b8af

commit a720948a (2023-12-18): fix(draw): fix off by one scaling issue o nthe left and bottom

fixes #4993