lvgl/lvgl #4995: Make most of the draw_sw_image logic reusable

kisvegabor, 2023-12-12:
### Introduce the problem

We have just added support for [Renesas Dave2D](https://github.com/lvgl/lvgl/tree/master/src/draw/renesas/dave2d) and it found that [lv_draw_dave2d_image.c](https://github.com/lvgl/lvgl/blob/master/src/draw/renesas/dave2d/lv_draw_dave2d_image.c) is mostly the duplicate of [lv_draw_sw_img.c](https://github.com/lvgl/lvgl/blob/master/src/draw/sw/lv_draw_sw_img.c). This way it's hard to follow that changes and fixes on all the places. 

I believe this problem applies to other GPU implementation too.

### Examples and cases
See above

### Suggested solution
In [lv_draw_sw_img.c](https://github.com/lvgl/lvgl/blob/master/src/draw/sw/lv_draw_sw_img.c) the only SW render specif part is [img_draw_core()](https://github.com/lvgl/lvgl/blob/master/src/draw/sw/lv_draw_sw_img.c#L288-L505). The rest is about decoding and tiling. 

I think we should make `img_draw_normal` and `img_draw_tiled` public and add a callback parameter which is something like `img_draw_core` or it's platform specific equivalent.

What do you think?

cc @XuNeo @nicusorcitu @jeremy-baker

kisvegabor, 2023-12-12:
cc @GorgonMeducer

GorgonMeducer, 2023-12-12:
I agree with you. In LVGL8, I have encountered the similar problem and do the same thing. 
But before make this happen, we should fully review the structural implication.

kisvegabor, 2023-12-12:
>  review the structural implication

What do you mean by that exactly?

GorgonMeducer, 2023-12-12:
I mean 
a. whether a structural change is flexible enough for a range of cases. 
b. Why Renesas chose this way to add GPU support? Whether duplicating SW drawing is unavoidable?

kisvegabor, 2023-12-13:
I'll prepare a PR (hopefully) today to see it in the practice.

kisvegabor, 2023-12-13:
PR is sent: #5008

Please let me know what you think. Let's continue in the PR.

PR #5008 (2023-12-14): feat(draw): add generic draw image helpers
### Description of the feature or fix

Fixes #4995 

Nothing new, just made the static functions global.

### Checkpoints
- [ ] Run `code-format.py` from the scripts folder. [astyle](http://astyle.sourceforge.net/install.html) needs to be installed.
- [ ] Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed
- [ ] Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant. 
- [ ] Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- [ ] If you added new options to `lv_co [... cut]
closed 2023-12-13 by kisvegabor