lvgl/lvgl #5002: How to use STM32U599 NeoChrom GPU？

Fluency1224, 2023-12-13:
One of my projects uses LVGL again, and the MCU is the STM32U599 series. As far as I know, this chip has a unique GPU graphics acceleration chip, but I don't know how to use it. There are only DMA2D related routines in LVGL and no GPU2D related content. Who can tell me how to use this? Thank you very much!

The official introduction to GPU2D is here:https://blog.st.com/neochrom-gpu/

kisvegabor, 2023-12-13:
Hey,

It not supported now. For LVGL v9 we established a policy to support the GPU only of our official partners. Unfortunately ST is not our official partner yet, but I hope they will be soon :slightly_smiling_face:

closed 2023-12-13 by kisvegabor

AngryCarrot61, 2024-06-22:
Unfortunately, ST supports their GPU2D only for their offical partners, like STemWin.
Therefore, their GPU2D is not documented in the reference guides, leaving it worthless in the dust.
Their under-rated user community should leave the useless GPU2D disabled at all times in order to reduce power consumption, unless you are a happy STemWin user. We don't support obscured code and hardware, so we don't use it.

Note NXP has GPU2D in their ARM Cortex M7 cores, which are even more powerful than the ST counterparts (up to 1GHz).