lvgl/lvgl #5006: Roller Widget in infinite mode not working with more than ~90 items

Lerosen, 2023-12-13:
### Perform all steps below and tick them with [x]
- [X ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ X] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ X] Update lvgl to the latest version
- [ X] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

The roller widget in infinite mode does not render if given many items. In my example with 100 options, these are not displayed anymore. This behavior can be observed in the simulator but also on my device (ESP32-S3 based)
WIth 80 items the selection is misaligned and on my device with touchscreen it is hard to deteministically select anything.

### To Reproduce

### Expected behavior

### Screenshots or video

misaligned @ 80
[image]

not options @ 100 items
[image]

expected @ 10 items
[image]

supremeneuron, 2023-12-13:
I had a similar situation with 60 elements that the roller widget would not render accurately where it would properly snap to a set option.  I modified my lv_conf.h file as below and it seemed to solve the issue.

/*Extend the default -32k..32k coordinate range to -4M..4M by using int32_t for coordinates instead of int16_t*/
#define LV_USE_LARGE_COORD 1

Lerosen, 2023-12-26:
I can confirm it ist working fine with that flag, thanks for your answer! 

I would suggest to add a Note regarding that behavior in the documentation of the Roller Widget.

kisvegabor, 2023-12-30:
I updated the docs here: https://github.com/lvgl/lvgl/commit/68a7beb5d8e258d312f4099dbeb78eb1342870f2

FYI, in v9 it always works as always large coordinates are used.

Lerosen, 2023-12-30:
Great and thanks for the quick help! Closing this issue now...

closed 2023-12-30 by Lerosen