lvgl/lvgl #5007: How can I draw a TEXT onto a canvas's alpha (opacity) channel?

djixon, 2023-12-13:
If I create a canvas FULL transparent, there is no text at all, however if I make canvas full opaque, then I can draw text on it, but I can not remove the rest of background of the canvas to leave only text. 

I want  ALPHA channel on ARGB canvas be populated by LUMA component from color of font (LUMA component is calculated   as (max(r, g, b) + min(r, g, b)) / 2 ).  That is standard way to form ALPHA channel from colored image (with assumption that black level represents full transparency) , so doesn't matter if text has some color (or gradient) or if glyph contains anti-aliased sub pixels, LUMA will provide proper value for ALPHA channel even at the anti-aliased edges. So goal is to make destination CANVAS transparent where no text appear and opaque (with proper anti-aliased edges) where  text exists. RGB should be populated by TextColor. That is pretty straightforward way to do in many different libraries, however I can not make it works properly on LVGL v8.3 (but it worked on v7.0). I tried with all combinations of blending modes but closest I get is light TEXT surrounded with some ugly black outline (there should be anti-aliased pixels ) and rest of canvas is full transparent. But that BLACK outline around text is some bug probably related to  pre-multiplied /un-premultiplied  alpha or something like that.   NORMAL_BLENDING_MODE must not assume SRC and DST canvas (buffers) are opaque. When drawing TEXT onto "something" you MUST not assume, that, that "something" is opaque! If it is, for example fully (or semi) transparent canvas, ALPHA from text GLYPH has to be combined with DST ALPHA. So resulting "layer" in this case ARGB canvas should have all 4 components ARGB calculated like this (this example is in normalized range (0.0-1.0) and have to be scaled in range 0-255:

inv = 1.0 - text.pixel.alpha;
r = (text.pixel.r * text.pixel.alpha) + (canvas.pixel.r * canvas.pixel.alpha * inv);    //you must not assume it is opaque
g = (text.pixel.g * text.pixel.alpha) + (canvas.pixel.g * canvas.pixel.alpha * inv);    
b = (text.pixel.b * text.pixel.alpha) + (canvas.pixel.b * canvas.pixel.alpha * inv);    
a = max(text.pixel.alpha, canvas.pixel.a);

 or in case if text doesn't have alpha channel but is "rendered" on black background

luma = (max(text.pixel.r , text.pixel.g, text.pixel.b) + min(text.pixel.r , text.pixel.g, text.pixel.b )) / 2.0;
inv = 1.0- luma;
r = (text.pixel.r * luma) + (canvas.pixel.r * canvas.pixel.alpha * inv);    
g = (text.pixel.g * luma) + (canvas.pixel.g * canvas.pixel.alpha * inv);    
b = (text.pixel.b * luma) + (canvas.pixel.b * canvas.pixel.alpha * inv);    
a = max(luma, canvas.pixel.a);

Is there a way to achieve that in v8.3?  

Later, I want to transform (scale, rotate and translate) such prepared canvas and to bake it into some background image.  (that part works properly in v8.3)

kisvegabor, 2023-12-15:
Hey,

It's a very important topic, so sorry in advance for being annoying, but I would like to understand all the aspects of it very well :slightly_smiling_face: 

Would it be possible for you to move to v9? It will be released in mid January. 

> If I create a canvas FULL transparent, there is no text at all, however if I make canvas full opaque, then I can draw text on it, but I can not remove the rest of background of the canvas to leave only text.

In v9, drawing to a transparent canvas works. Try this
```c
void lv_example_canvas_4_modified(void)
{
    /*Create a buffer for the canvas*/
    static uint8_t cbuf[LV_CANVAS_BUF_SIZE(CANVAS_WIDTH, CANVAS_HEIGHT, 32, LV_DRAW_BUF_STRIDE_ALIGN)];

    /*Create a canvas and initialize its palette*/
    lv_obj_t * canvas = lv_canvas_create(lv_screen_active());
    lv_canvas_set_buffer(canvas, cbuf, CANVAS_WIDTH, CANVAS_HEIGHT, LV_COLOR_FORMAT_ARGB8888);
    lv_canvas_fill_bg(canvas, lv_color_hex3(0xccc), LV_OPA_TRANSP); 
    lv_obj_center(canvas);

    lv_layer_t layer;
    lv_canvas_init_layer(canvas, &layer);

    lv_draw_label_dsc_t dsc;
    lv_draw_label_dsc_init(&dsc);
    dsc.color = lv_palette_main(LV_PALETTE_RED);
    dsc.font = &lv_font_montserrat_18;
    dsc.decor = LV_TEXT_DECOR_UNDERLINE;
    dsc.text = "Hello";

    lv_area_t coords = {10, 10, 30, 60};

    lv_draw_label(&layer, &dsc, &coords);

    lv_canvas_finish_layer(canvas, &layer);
}
```

But I wonder what your exact use use case is.

---

It's also for v9 but later I would like to introuduce rendering to 
- `L8`: all color are converted to grayscale, it can used for gray scale displays or a mask
- `A8` all RGB info is dropped and only the alpha channel is rendered, also good for masks

The very specific problem that I would like to solve is having a image with a pattern or complex gradient and we mask out letters from it. Like this:
[image]

For this what we could is:
1. Create an `lv_label`
2. Tell LVGL to render it on a new `layer`, assign a static buffer for the `layer`, and don't blend the `layer`
3. Set `A8` format for the layer
4. Use the layer's buffer as a mask for the image. 

cc @XuNeo (We were talking about something like this earlier)

djixon, 2023-12-15:
Thank you for an extra explanation, but unfortunately I'm stick to v8.3 for this project. It would be too risky to try to move a lot of code to a new version one month till deadline. There is also qmsd library [https://github.com/smartpanle/QMSD-ESP32-BSP/tree/master](url) which is somehow stick to LVGL v8.3 version. You can see in their repository inside third-party-components there is their UI-engine tightly lock to any of those two LVGL versions they included in that folder V7 and V8.3.But their library contains all the drivers for those boards. 

My suggestion related to your question. Look, basically you chose in between two different concepts (don't mix them):
1. Use layers (canvas, buffers etc) with alpha channel included (in that case you do NOT use masks at all). Alpha within that layer or buffer or canvas is ALWAYS responsible for that how it is drawn on another layer (or how another layer is drawn onto it with combined Alphas)
2. use always ONLY opaque layers which ALWAYS requires masks for blending.

Basically those are two different concepts presented in most compositing software. You chose one or another but not both.
For your example in case with Alpha channel, yes, you would need some image with that texture in RGB channels, while in alpha channel you have "white" text and rest is standard blending function. Because where alpha is full (white, 255) that image use its own color while on places where alpha is "dark" (black , 0) it is transparent and layers "below" are visible

In case without alpha, but masks, mask determines where layer will be transparent. So pretty the same thing but working with an Alpha channel is more straight forward, because it is usually already allocated within pixel buffer especially for RGBA8888 where it provides ideal 4 byte alignment of each pixel so when pixel data is fetched (by 32 reads from buffer) Alpha information is already there too

XuNeo, 2023-12-16:
Canvas feature is described in https://github.com/lvgl/lvgl/issues/4910.

LVGL does not support A8 as dest buffer color format yet.

djixon, 2023-12-20:
@XuNeo well, you should consider merging things you call "color" and "opacity" into single thing called color.  In human naming conventions it maybe doesn't fit well but in graphic library those things are just channels (R G B A) and all behave the same way. You can write to them, read from them and each has its purpose. R determines intensity of "light" which will affect receptors for RED spectrum in retina of the human eye, G for green well Alpha for none but rather it determines blending behavior of that color.

So by having UNIQUE color_t which contains all 4 components R,G,B,A you are actually  allowing to setup both RGB and Opacity by single function call with single register passed. Also in blending (mixing) functions, in such a case you always deal with only two 32bit registers (Src and Dst) which both contain all 4 components.  By setting, lets say style parameter, you do not need separated functions for Colors and Opacity when both are within lv_color_t ...... Then COLOR itself (by having all 4 components) determines what are target channels in drawing routine. For example if I make color rgba(0,0,0,255) and I decided to draw text of that color onto some canvas it is obvious that "painting" will occur with BLACK in RGB and 255 (white) in alpha ..... or if you make color RGBA(128,128,0,128) it is semitransparent "some yellowish" color.  A lot of functions become more simpler by dealing with ALL 4 components within just one 32 bit register and lot of blending math of  two RGBA8888 (or ARGB8888 or whatever rearngement) can be performed in masking and shifting all components within 64 bit register. Similar way two 16 bit RGB565A8 blending is performed within two 32 bit registers... In general, whole library would be faster

kisvegabor, 2023-12-20:
On API level we havn't merged RGB and Alpha into one struct because the suer might want to set the RGB part or the Alpha part without affecting the other. E.g.
```c
lv_style_set_bg_color(&style1, red);
lv_style_set_bg_opa(&style2, 0x80);
``` 

We can't even store them together as once we store an `ARGB` color just for `RGB` in style we can't tell that Alpha is there but it's not really set and shouldn't be considered. (i.e. LVGL should keep looking for really set Alpha in an other style.)

djixon, 2023-12-21:
Well, you should consider merging those two components., seriously.  It is not an accident that all professional and commercial libraries, shading languages and renderers, trait COLOR as 4 component type. There are huge benefits behind that.     

> We can't even store them together as once we store an ARGB color just for RGB in style we can't tell that Alpha is there but it's not really set and shouldn't be considered

That is wrong way of thinking.  Tell me, how do you know the Blue component is there? Maybe user didn't  call the color  function at all. 

Simply, (for users thinking like you) you can make it default to value 255. But most of experienced programmers when thinking about color, always think about 4th component too.   

And all this is not about API at all. At API level, you can always provide separate functions (if you want to) to an end user. Suppose that lv_color_t contains all of 4 components. So some internal variable (inside some struct or what ever,  holded only 3 RGB values so far, now holds all 4 values). Your API then can
have both functions like it has so far. In such a case, LV_COLOR_RED would be defined as (255, 0, 0, 255) so user can call :

lv_style_set_bg_color(@style1, LV_COLOR_RED);
lv_style_set_bg_color(@style1, lv_color_make(255, 0, 0));
lv_style_set_bg_color(@style1, lv_color_make(255, 0, 0, 255));
and also
lv_style_set_bg_opa(@style1, 0x80);

Behind the "scene" the first function has already determined opacity by definition of LV_COLOR_RED with value of alpha set to 255. The second function lv_color_make() called with only 3 arguments "should know" that it has to setup Alpha to 255. However, if it is called with 4 arguments, then it "should know" in those 4 values, alpha is presented too. Function for opacity setup would only affect that byte where alpha is stored not "touching" RGB components. That way you can keep full compatibility with previous versions but also involve new behavior like
lv_style_set_bg_color(@style1, lv_color_make(R, G, B, A));

By other words now you have UNIQUE lv_color_t carring all 4 information.  

It can speed a lot of internal math (if both color and alpha are within the same 32 bit register). Writing to/reading from buffers (of the same format)  is then performed by single write or single read. Since "color" containing all 4 components in single register write/read the buffer by single memory access. Predetermined names for colors like LV_COLOR_RED are handled by macro lv_color_make (255,0,0,255) .... its not hard to make assumption that, if user pass only 3 arguments it "assumes" opaque behavior and then, Alpha is handled automatically by value 255. So by other words, if only 3 arguments are passed to any of  function expecting R,G,B channels, alpha is automatically set to 255. If 4 arguments are passed then those are carrying Alpha information too. Many graphic libraries including a lot of shading languages trait COLOR always as 4 component encapsulation (where default value for Alpha is 255 (1.0f for libraries using normalized intensities like 32 bit floats or 16 bit half-floats for channel values). Why would you involve two memory access to set up both (rgb) channels and alpha channel when it can be performed by single access?  Its not only about storing in 32 bit value where RGB and Opacity are stored, it is about all reading/writing to buffers containing  ARGB  channels. So such color_t (containing all 4 components) writes/reads from such a buffer by single memory access. 
There are more benefits (not only single memory access) from that:

1. If you analyze ANY blending or mixing function you will see that in any calculation, ALPHA is always involved in MULTIPLICATION of each of color channels (while inverse alpha (255-alpha) multiply second (buffer, pixel, color ...). So  once the lv_color_t  (with all 4 values) is set, an internal pre-multiplied value of the same lv_color_t can be formed which contains following 4 components (R*A, G*A, B*A, 255-A)  and such value is used in all blending/mixing routines saving  3 multiplication operations per each pixel.       
2. All drawing routines (which accept a color as an argument of calling function) now have new lv_color_t (with all 4 components included)  and they "draw" text, line, arc .... to a compatible buffer with also all 4 components. Each channel (coming from lv_color_t) goes straight to the buffer. So you do not need to use MASKS and draw separate to them and later perform mixing  ......  this way you ARE DRAWING the mask and the RGB simultaneously by single memory access. The rest is just blending function

djixon, 2023-12-21:
I forgot  to add one more extremely useful thing of such a concept. Since the all operations to/from a buffer are a single memory access operations with all 4 components involved, that allows SHUFFLING channels "on the fly" and other things of a huge benefit. Suppose, the WRITING to a buffer is described by let's say lv_buffer_mode_t  which is uint32_t  composed of 4 bytes where each is of lv_channel_mode_t  (uint8_t)  describing how 32bit color value (carrying all 4 components)  WILL be MERGED to the buffer. So bits of lv_channel_mode_t are described as:
```

typedef _lv_channel_mode {
 uint8_t merging_mode : 4 ;  // 4 lower bits defining merging operation 
 uint8_t shuffle: 4;  // 4 upper bits defining the various shuffle operations
} lv_channel_mode_t;

```
and whole buffer mode  is then:
```

typedef  _lv_buffer_mode {
 lv_channel_mode_t  R_mode;
 lv_channel_mode_t  G_mode;
 lv_channel_mode_t  B_mode;
 lv_channel_mode_t  A_mode;
} lv_buffer_mode_t;
```

Then such a 32bit lv_buffer_mode_t defines HOW PASSED VALUE in lv_color_t of all 4 components will affect each channel in buffer and "on the fly shuffling". So lets say shuffling is defined as:
```

ENUM lv_channel_shuffle_enum {
 NO_SHUFFLE = 0,  // default channel value
 USE_R, // force using value from R channel
 USE_G, // force using value from G channel
 USE_B, // force using value from B channel
 USE_A, // force using value from Alpha channel
 USE_MAX_OF_RGB,
 USE_LUMINESCENCE_OF_RGB,  //  luma = (max(R, G, B) + min(R, G, B)) / 2  
 USE_LUMINESCENCE_OF_RGB_MULTIPLIED_BY_A,  //  luma * A
 INV_NO_SHUFFLE = 8,  // default channel value inverted
 INV_USE_R, // force using value from R channel inverted
 INV_USE_G, // force using value from G channel inverted
 INV_USE_B, // force using value from B channel inverted
 INV_USE_A, // force using value from Alpha channel inverted
 INV_USE_MAX_OF_RGB  // 255 - max(r,g,b) 
 INV_USE_LUMINESCENCE_OF_RGB,  //  255 -  (max(R, G, B) + min(R, G, B)) / 2   };
 INV_USE_LUMINESCENCE_OF_RGB_MULTIPLIED_BY_A,  // 255 - luma * A

```
So 4th bit determines inverted behavior.  
This shuffling enum "tells" of HOW a passing value in lv_color_t (single 32 bit register composed of all 4 components) will be internally transformed into another 32 bit register lets call it TMP_CHNLS.  So ARGB from lv_color_t is transformed to internally ARGB value where that TMP_CHNLS is calculated based on those lv_buffer_mode_t which contains "shuffling" for each of 4 components. Then such an intermediate value (in single 32 bit register) is merged into buffer by following defined operations:   
```
ENUM merging_mode_e {
 LV_MRG_MODE_UNCHANGED = 0,  // this will leave value in buffer of that channel intact (unchanged)
 LV_MRG_MODE_OVERWRITE ,           // this will overwrite value in buffer (that is what LVGL does usually) from ch in TMP_CHNLS
 LV_MRG_MODE_BLEND_USING_COLOR_ALPHA,  // (value of this channel in buffer will be mixed with value from the TMP_CHNLS using ALPHA from TMP_CHNLS)  ch_in_buf = (ch_in_TMP_CHNLS * Alpha_in_TMP_CHNLS) + (ch_in_buf * (255 - Alpha_in_TMP_CHNLS))
 LV_MRG_MODE_BLEND_USING_BUFFER_ALPHA,  //  (value of this channel in buffer will be mixed with value from the same channel in TMP_CHNLS using ALPHA stored in buffer) ch_in_buf = (ch_in_TMP_CHNLS * Alpha_in_buffer) + (ch_in_buf * (255 - Alpha_in_buffer))
 LV_MRG_MODE_BLEND_USING_INV_COLOR_ALPHA,  //  (value of this channel in buffer will be mixed with value from the same  
channel in TMP_CHNLS using inverted ALPHA stored in TMP_CHNLS) ch_in_buf = (ch_in_TMP_CHNLS * (255 -Alpha_in_TMP_CHNLS)) + (ch_in_buf * Alpha_in_TMP_CHNLS)

LV_MRG_MODE_BLEND_USING_INV_BUFFER_ALPHA,  //  (value of this channel in buffer will be mixed with value from the same channel in TMP_CHNLS using inverted ALPHA stored in buffer) ch_in_buf = (ch_in_TMP_CHNLS * (255 -Alpha_in_buffer)) + (ch_in_buf * Alpha_in_buffer)

LV_MRG_MODE_ADD_CLAMPED, // value of this channel in buff will be added with value of the same channel in TMP_CHNLS  (will be saturated to 255 if sum is greater than 255) 
LV_MRG_MODE_ADD_UNCLAMPED, // value of this channel in buff will be added with value of the same channel in TMP_CHNLS    
(will roll over ignoring ninth bit so 256 becomes 0, 257 becomes 1 etc)
LV_MRG_MODE_ADD_NORMALIZED, // ch_in_buff = (ch_in_buf + ch_in_TMP_CHNLS) / 2
LV_MRG_MODE_MULTIPLY, // ch_in_buff = (ch_in_buf * ch_in_TMPS_CHNLS) >> 8
LV_MRG_MODE_ABS_DIFFERENCE, // ch_in_buff = abs(ch_in_buf - ch_in_TMPS_CHNLS)
etc.
};

```
Now, finally some lets say draw routine which writes to a buffer can look like
lv_canvas_draw_txt ( .... usual parms,  lv_color_t col, lv_buffer_mode_t mode)

because col contains all 4 components RGBA  and suppose we pass 0x01010161 as lv_buffer_mode_t
which means RGB will not be shufled (values RGB from col will OVERWRITE values in buffer, however ALPHA will be formed "on fly" from LUMA computed from RGB and also STORED in buffer. So you have text of desired color and buffer is opaque at those places while at the edges of the font it becomes semitransparent and finally full transparent where there is no text. That was simple example of  non_initilaized buffer because such operation will completely initialize it.  But now suppose some complex  example. If you already have a buffer of a lets say, some image, which contains rgba of that image and is part of hierarchy of multi layers. Now if you want for example to draw to such a buffer and you want to lets say draw a TEXT which wi [... cut ...]
tmp_b = ((bot & 0xFF) * tmp_b) >> 8) & 0x00FF00FF00FF00FF;  // here are all 4 multiplication of bottom channel done and values returned into 0-255 range and removed those lower bits after shifting so next subtraction can be performed
uint64_t res =  (((top & 0xFF) * (tmp_t - tmp_b)) >> 8) & 0x00FF00FF00FF0FF + tmp_b; // we are DONE here only we need is to bring back R and B component into lower part of 64 bit register
res = ((res >> 24) & 0xFF00FF00) | (res & 0x00FF00FF);

now lower 32 bits in res contains final result RGBA

closed 2024-01-15 by lvgl-bot