lvgl/lvgl #5009: This is HUGE unnecessary BUG in v8.3

djixon, 2023-12-14:
This math is WRONG!
`#elif LV_COLOR_DEPTH == 16
    /**
     * The floating point math for conversion is:
     *  valueto = valuefrom * ( (2^bitsto - 1) / (float)(2^bitsfrom - 1) )
     * The faster integer math for conversion is:
     *  valueto = ( valuefrom * multiplier + adder ) >> divisor
     *   multiplier = FLOOR( ( (2^bitsto - 1) << divisor ) / (float)(2^bitsfrom - 1) )
     *
     * Find the first divisor where ( adder >> divisor ) <= 0
     *
     * 5-bit to 8-bit: ( 31 * multiplier + adder ) >> divisor = 255
     * divisor  multiplier  adder  min (0)  max (31)
     *       0           8      7        7       255
     *       1          16     14        7       255
     *       2          32     28        7       255
     *       3          65     25        3       255
     *       4         131     19        1       255
     *       5         263      7        0       255
     *
     * 6-bit to 8-bit: 255 = ( 63 * multiplier + adder ) >> divisor
     * divisor  multiplier  adder  min (0)  max (63)
     *       0           4      3        3       255
     *       1           8      6        3       255
     *       2          16     12        3       255
     *       3          32     24        3       255
     *       4          64     48        3       255
     *       5         129     33        1       255
     *       6         259      3        0       255
     */

    lv_color32_t ret;
    LV_COLOR_SET_R32(ret, (LV_COLOR_GET_R(color) * 263 + 7) >> 5);
    LV_COLOR_SET_G32(ret, (LV_COLOR_GET_G(color) * 259 + 3) >> 6);
    LV_COLOR_SET_B32(ret, (LV_COLOR_GET_B(color) * 263 + 7) >> 5);
    LV_COLOR_SET_A32(ret, 0xFF);
    return ret.full;
#elif LV_COLOR_DEPTH == 32`

By such a math, you introduce a huge constant error in color calculation and beside that error such a math is CPU costly.
Look, color linearization in extended bits should be done more precisely and much cheaper in computational sense. You MUST NOT add constant values (those adders) because that way you affect the brightness of the converted color. I will explain math on simpler example. Suppose you have a VALUE of  BYTE (range 0-255) and you want to scale it to uint16_t. Sure, VAL<<8 is not good enough because lowest byte will always be zero. You need its linearization too. The best (and fastest) you can do is JUST SIMPLE COPYING of original byte into that lower byte so you have (VAL<<8) | VAL in 16bits. Perfect linearization!!! with extremely fast CPU operations. You can not achieve better linearization than you have already stored in VAL itself. So  the same way conversion from 5->8 bits and 6->8 bits have to be done. 
5to8:
VAL8 = (VAL5 << 3) | (VAL5 & 0x07);
So you composed resulting 8 bits value from starting 5 bits value just by shifting to the left for 3 bits and ORing lowest 3 bits from the original 5 bit value into that result. Those 3 bits represent LINEARIZATION of 3 lowest bits in extended result. So if they are 0 in original 5 bit value, they will stay 0 in the extended result, if they are 1 they stay 1 .... if they are 7 they stay 7. So you have perfect linearization based on them. Similar way is for 6->8   
6to8:
VAL8 = (VAL6 << 2) | (VAL6 & 0x03);
So you composed resulting 8 bits value from starting 6 bits value just by shifting to the left for 2 bits and ORing lowest 2 bits from the original 6 bit value into that result.

That is more accurate and much simpler in computational sense.
cheers

kisvegabor, 2023-12-14:
Hi @djixon 

Thank you for pointing it out. Could you send a Pull request with the suggested change? It would be easier to review it this way.

djixon, 2023-12-14:
@kisvegabor  Well I have never done that before. How can I send Pull request?
Here is the code I replaced and tested in the **lv_color.h**  v8.3 for function **lv_color_to32** (I left REMed old lines and added new ones below them)  

`    lv_color32_t ret;
//    LV_COLOR_SET_R32(ret, (LV_COLOR_GET_R(color) * 263 + 7) >> 5);
//    LV_COLOR_SET_G32(ret, (LV_COLOR_GET_G(color) * 259 + 3) >> 6);
//    LV_COLOR_SET_B32(ret, (LV_COLOR_GET_B(color) * 263 + 7) >> 5);
    LV_COLOR_SET_R32(ret, (LV_COLOR_GET_R(color) << 3) | ((LV_COLOR_GET_R(color) >> 2) & 0x07));
    LV_COLOR_SET_G32(ret, (LV_COLOR_GET_G(color) << 2) | ((LV_COLOR_GET_G(color) >> 4) & 0x03));
    LV_COLOR_SET_B32(ret, (LV_COLOR_GET_B(color) << 3) | ((LV_COLOR_GET_B(color) >> 2) & 0x07));
    LV_COLOR_SET_A32(ret, 0xFF);
`

kisvegabor, 2023-12-15:
See the guide in our docs about [Pull requests](https://docs.lvgl.io/8.3/CONTRIBUTING.html#pull-request).

I also think we can make good use of it v9 as well. E.g. [here](https://github.com/lvgl/lvgl/blob/master/src/misc/lv_color.h#L294).

djixon, 2023-12-15:
Sure, whenever some smaller range of values has to be extended into wider range, that math works perfectly. For example range of 4 bits (0 to 15)  into range of 8 bits (0 to 255):
4->8:
val8 = (val4 << 4) | val4;

or for example 2 bits ALPHA into 8 bits alpha
alpha4 = (alpha2 << 2) | alpha2;
alpha8 = (alpha4 << 2) | alpha4;
Do you see? If destination range is more than twice larger than source you have to populate all those extended bits to make them all properly distributed in new generated range. So linear distribution in newly generated range is always perfect.

etc.

closed 2024-01-06 by lvgl-bot