lvgl/lvgl #5011: About the style of the tileview

IAMMX, 2023-12-14:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest versionx
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
During the sliding process, the content in the tileview will cover the border. Please help fix it.

### To Reproduce
```c
void lv_example_tileview_1(void)
{
    lv_obj_t * tv = lv_tileview_create(lv_screen_active());
    lv_obj_set_size(tv,200,200);
    lv_obj_set_style_border_color(tv,lv_color_black(),LV_PART_MAIN);
    lv_obj_set_style_border_width(tv,4,LV_PART_MAIN);
    lv_obj_center(tv);

    /*Tile1: a button*/
    lv_obj_t * tile1 = lv_tileview_add_tile(tv, 0, 0, LV_DIR_RIGHT);

    lv_obj_t * button = lv_button_create(tile1);
    lv_obj_set_size(button, LV_SIZE_CONTENT, LV_SIZE_CONTENT);
    lv_obj_center(button);
    lv_obj_t * label = lv_label_create(button);
    lv_label_set_text(label, "right");

    /*Tile2: a list*/
    lv_obj_t * tile2 = lv_tileview_add_tile(tv, 1, 0, LV_DIR_LEFT);
    lv_obj_t * list = lv_list_create(tile2);
    lv_obj_set_size(list, LV_PCT(80), LV_PCT(80));
    lv_obj_center(list);

    lv_list_add_button(list, NULL, "One");
    lv_list_add_button(list, NULL, "Two");
    lv_list_add_button(list, NULL, "Three");
    lv_list_add_button(list, NULL, "Four");
    lv_list_add_button(list, NULL, "Five");
    lv_list_add_button(list, NULL, "Six");
    lv_list_add_button(list, NULL, "Seven");
    lv_list_add_button(list, NULL, "Eight");
    lv_list_add_button(list, NULL, "Nine");
    lv_list_add_button(list, NULL, "Ten");

}
```

### Expected behavior

### Screenshots or video
[image]

kisvegabor, 2023-12-14:
Hi, 
Try `lv_obj_set_style_border_post(tv, true, LV_PART_MAIN);`

IAMMX, 2023-12-14:
> Hi, Try `lv_obj_set_style_border_post(tv, true, LV_PART_MAIN);`

Can you write this **style** into **lv_theme_default.c** as the **default** **configuration** of **tileview?** I think tileview as a multi-interface control, sliding is inevitable, this setting is more reasonable as the default configuration.

kisvegabor, 2023-12-15:
As the tileview doesn't add a border by default I would like to not touch border related properties. In the default theme we don't have a a style to change only `border_post` and I would like to add only for a single purpose :slightly_frowning_face:

IAMMX, 2023-12-15:
Okay, I understand.

closed 2023-12-15 by IAMMX