lvgl/lvgl #5014: browser freezes while using mp_js_do_str at simulator

leidto, 2023-12-14:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
I'd like to interact with the simulator with mp_js_do_str. I can send a single command. It's fine.
When I send the commands continuously, one after the other, the browser freezes completely.
I've tried almost everything. (async and sync)
```
const sendCommand = Module.cwrap('mp_js_do_str', 'null', ['string'], { async: true })

const arr = Array.from({ length: 100 }, (_, i) => i)
const move = async () => {
   for await (const i of arr) {
      await delay(10)
      await sendCommand (`btn.set_pos(${i}, ${i})`)
   }
}
```

### To Reproduce
Follow the code above.

### Expected behavior
The browser doesn't freeze.

leidto, 2023-12-14:
We previously discussed similar issues with @embeddedt. Maybe he can help.

embeddedt, 2024-01-02:
I am actually not sure what causes this issue. I suspect what *might* be happening (although I am unsure how to verify it) is that Emscripten's async stuff is not atomic, and allows for your `mp_js_do_str` call to proceed while LVGL is already in the middle of rendering something else. In Firefox, when the browser freezes, I am eventually able to press a button to stop the script, and then the message

```
[Error]     (5.288, +5288)   _lv_inv_area: Asserted at expression: !disp->rendering_in_progress (Invalidate area is not allowed during rendering.) lv_refr.c:263
```

is printed. This would support my theory, since it's likely that the assert spinning in a loop prevents the browser from actually redrawing the message until the script is killed manually. The problem is that I am not sure how to fix this.

I've also attached the backtrace I see when stopping the script (this is from a build compiled with `make -j4 EM_DEBUG=1`, which enables debug info to be output into the WebAssembly file, and disables optimizations):

```
Script terminated by timeout at:
_lv_inv_area@http://localhost:8080/firmware.wasm:wasm-function[3970]:0x5b2ec9
lv_obj_invalidate_area@http://localhost:8080/firmware.wasm:wasm-function[3643]:0x52a1e8
lv_obj_invalidate@http://localhost:8080/firmware.wasm:wasm-function[3624]:0x52672c
lv_obj_refresh_style@http://localhost:8080/firmware.wasm:wasm-function[3751]:0x562991
lv_obj_set_local_style_prop@http://localhost:8080/firmware.wasm:wasm-function[3774]:0x56dce3
lv_obj_set_style_x@http://localhost:8080/firmware.wasm:wasm-function[3816]:0x580b54
lv_obj_set_x@http://localhost:8080/firmware.wasm:wasm-function[3599]:0x51944e
lv_obj_set_pos@http://localhost:8080/firmware.wasm:wasm-function[3598]:0x518f26
mp_lv_obj_set_pos@http://localhost:8080/firmware.wasm:wasm-function[1937]:0x2fc920
lv_fun_builtin_var_call@http://localhost:8080/firmware.wasm:wasm-function[1698]:0x2c27cf
mp_call_function_n_kw@http://localhost:8080/firmware.wasm:wasm-function[684]:0xe0eeb
mp_call_method_n_kw@http://localhost:8080/firmware.wasm:wasm-function[681]:0xe063f
dynCall_iiii@http://localhost:8080/firmware.wasm:wasm-function[7688]:0xd3576f
instrumentWasmExports/ret[x]@http://localhost:8080/micropython.js:8571:24
createExportWrapper/<@http://localhost:8080/micropython.js:673:20
invoke_iiii@http://localhost:8080/micropython.js:9480:10
instrumentWasmImports/imports[x]@http://localhost:8080/micropython.js:8549:24
mp_execute_bytecode@http://localhost:8080/firmware.wasm:wasm-function[1481]:0x24d0fd
fun_bc_call@http://localhost:8080/firmware.wasm:wasm-function[931]:0x14f9e3
mp_call_function_n_kw@http://localhost:8080/firmware.wasm:wasm-function[684]:0xe0eeb
mp_call_function_0@http://localhost:8080/firmware.wasm:wasm-function[683]:0xe0949
dynCall_ii@http://localhost:8080/firmware.wasm:wasm-function[7683]:0xd35260
instrumentWasmExports/ret[x]@http://localhost:8080/micropython.js:8571:24
createExportWrapper/<@http://localhost:8080/micropython.js:673:20
invoke_ii@http://localhost:8080/micropython.js:9447:10
instrumentWasmImports/imports[x]@http://localhost:8080/micropython.js:8549:24
do_str@http://localhost:8080/firmware.wasm:wasm-function[6448]:0xb55965
mp_js_do_str@http://localhost:8080/firmware.wasm:wasm-function[6449]:0xb564d3
instrumentWasmExports/ret[x]@http://localhost:8080/micropython.js:8571:24
createExportWrapper/<@http://localhost:8080/micropython.js:673:20
ccall@http://localhost:8080/micropython.js:8775:17
cwrap/<@http://localhost:8080/micropython.js:8796:10
move@http://localhost:8080/lvgl.html:166:18
async*window.startRunning/<@http://localhost:8080/lvgl.html:169:5
setTimeout handler*window.startRunning@http://localhost:8080/lvgl.html:157:14
async*mainProgram/<@http://localhost:8080/micropython.js:48:27
setTimeout handler*mainProgram@http://localhost:8080/micropython.js:48:13
doRun@http://localhost:8080/micropython.js:9593:46
run@http://localhost:8080/micropython.js:9606:3
runCaller@http://localhost:8080/micropython.js:9569:18
removeRunDependency@http://localhost:8080/micropython.js:632:4
receiveInstance@http://localhost:8080/micropython.js:767:22
receiveInstantiationResult@http://localhost:8080/micropython.js:775:18
promise callback*instantiateAsync/<@http://localhost:8080/micropython.js:741:18
promise callback*instantiateAsync@http://localhost:8080/micropython.js:739:6
createWasm@http://localhost:8080/micropython.js:785:18
@http://localhost:8080/micropython.js:9292:11
```

leidto, 2024-01-11:
Interesting. I have to do a quick check before every command: `if (window.Asyncify.currData) { ... }`, but it's very hacky and ugly :/

closed 2024-02-11 by lvgl-bot