lvgl/lvgl #5015: mixing of two RGB565 pixels without single multiplication (possible? ... sure) .

djixon, 2023-12-14:
In searching for highest speed of critical functions like mixing of two RGB565 colors (on ESP32S3 display) I've just finished complex calculation and need someone for speed testing in comparison to actual routine.  The well known function which uses 32 bit registers to do blending of two RGB565 colors, does extraction of G component into higher 16 bits word by utilizing shift 16 bits to the left and then makes proper masking by mask 0x07E0F81F leaving at least 5 bits to the left of each component for blending calculation. 
It can be improved even further. Since 8 bit mix value has sense only in 8 times smaller range (range 0 to 32) its possible to:

1. Make a switch() based on those 33 (0-32) exact possible values after (mix+4) >> 3
2. Use shifting (on flight) when 16 bit RGB565 values are stored into 32 bit registers (some CPUs can perform that in single cycle)
3. Make custom calc for each particular case of those 33 (allowing compiler to pre-calculate required masks at compile time for each case where it is possible, that way we save some clocks too)  
4. Only ADDITION and SHIFTING will be utilized in calculation (without single multiplication which even in hardware implemented multiplier takes approx 16 - 30 clock cycles sometimes even more, it depends on implementation)

In attachment is the full working and tested lv_color_mix() function for v8.3 However, it would be nice if someone could put it in v9.0 and make robust benchmarks with it on different platforms which use RGB565 color format. On ESP32S3 LCD module there is noticeable improvement in speed with transparent widgets. Further improvements would be only possible at assembly level but that would make a routine a very platform specific. So you can keep it as pure C and that way it will work on any platform.  Don't scare about its size. It is written almost like for assembly language and speed has costs, usually in terms of code size. So maybe it can be encapsulated into some configuration option switch something like LV_USE_EXTRA_FAST_MIX_RGB565_FUNCTION where user can chose in between that and old function . Here is the whole function code: 
[MixRoutineRGB565.txt](https://github.com/lvgl/lvgl/files/13678922/MixRoutineRGB565.txt)

kisvegabor, 2023-12-15:
Pretty interesting! My only concern is that the suggested implementation is much longer and this function is inlined on a lot of places. I wonder what is the performance difference if the function is not inlined.

Can you try this on your ESP platform?
```c
  lv_demo_widgets();
  lv_refr_now(NULL); /*Contains layout calculation too so not accurate*/

  /*Normal render time*/
  lv_obj_invalidate(lv_scr_act());
  volatile uint32_t t1 = lv_tick_get();
  lv_refr_now(NULL);
  t1 = lv_tick_elaps(t1);
  lv_obj_set_style_bg_opa(lv_layer_top(), LV_OPA_50, 0);

  /*Render time with overlay*/
  lv_obj_set_style_bg_opa(lv_layer_top(), LV_OPA_50, 0);
  volatile uint32_t t2 = lv_tick_get();
  lv_refr_now(NULL);
  t2 = lv_tick_elaps(t2);
```
On STM32F769 I got this:
- v9 original
  - t1: 29
  - t2: 44
  - mem: 410.82kB

- v9 mod inline
  - t1: 31
  - t2: 45 
  - mem: 412.69kB

- v9 mod no inline
  - t1: 32
  - t2: 46 
  - mem: 410.89kB

So the suggested algorithm is actually slower here :slightly_frowning_face:

djixon, 2023-12-15:
I made my first fork and posted request in master branch. With slightly optimized the size of code because in that v9.0 version ret value can be returned (without passing it to down-existed code which exists in v8.3 but is removed in v9). So for case 0 and case 32 the return is performed immediately from the switch (those are two most common cases). The rest of cases have 5 lines of code which are the common for all of them so now they are pulled from the switch because cases 0 and 32  do not pass the code after switch (in v8.3 they have to, because that routine in v8.3 has some more lines of code and true return was bellow it). So idea is to test it on different platforms. The also important thing for compilation is to force compiler to generate table of addresses  for switch() statement. Most of them do that by default but some of them generate branching with O(log N). The goal of that switch is exactly to avoid that. Based on mix value (0-32) jump to exact address of code (from that table) for each case should be performed without any additional branches. However some compilers do not generate table if there are more than 8 or 16 cases but rather generates code with O(log N) complexity branches. So you should check your generated assembly and see if the switch() is properly compiled to use address table. (maybe I can modify the code so address for each case is hardcoded in routine. That way calling (or jumping) can be performed (simulating switch statement) in the routine itself and is independent on compiler options.   

I am currently working on some project and whole setup (for multi devices) is locked to v8.3. So I can not test in latest v9.0 until I finish that project (middle of January). And I will actively participate in LVGL after that.  However, I can test it in v8.3 (but I must admit that I made a lot of changes on my local copy of that version mostly related to RGB565 color format, lot of function changed. So it wont be relevant until I sync to latest version.

kisvegabor, 2023-12-15:
> I made my first fork and posted request in master branch.

I can't see the Pull request [here](https://github.com/lvgl/lvgl/pulls). Could you check if it's really sent?

> And I will actively participate in LVGL after that. However, I can test it in v8.3 (but I must admit that I made a lot of changes on my local copy of that version mostly related to RGB565 color format, lot of function changed. So it wont be relevant until I sync to latest version.

We cannot add large scale change to v8, but I would be happy to hear your improvement proposals for v9.

djixon, 2023-12-15:
I made my own branch ... made changes and click send to master (or sync with master) ... something like that. I thought it is then completed.

djixon, 2023-12-15:
I see tons of errors, mainly because I copy-pasted code from my local v8.3 but in this last version you've changed types of those parameters passed to the function. lol. Now I corrected those and "fight" with that rigorous text formatter which complains about spaces .... anyway, I added into Kconfig switch LV_USE_ALTERNATE_MIX_ROUTINE_FOR_RGB565_COLOR_FORMAT but now those chekers complain that LV_USE_ALTERNATE_MIX_ROUTINE_FOR_RGB565_COLOR_FORMAT is not defined in that lv_color.h header. Where else should I define LV_USE_ALTERNATE_MIX_ROUTINE_FOR_RGB565_COLOR_FORMAT?

kisvegabor, 2023-12-16:
Thank you for the PR. Let's continue there
#5019

closed 2023-12-16 by kisvegabor

djixon, 2023-12-17:
I see, during Copy-Paste from v8.3 into that last branch git created for me, I had to manually edit function (without testing) because argument types have changed in v9. During that manual edit on Git I also "unrolled" math operation, because now in v9 cases for full-transparent and full opaque allow early exit which then allows the common part of blending for all other cases be pulled from each particular case and put after the switch statement. (in v8.3, that formed ret value was "passed" down to the rest of code inside that function (where it could be affected by that rest of code so instant return was not possible).  But that made thing worse. It created Interlocked execution pipeline, making the list of consecutive instructions highly dependent on each other.  And performs worse than original routine.  Once I finish my current project (deadline mid of January) I will take manual optimization in v9 and tune the arrangements of that math to bring back the performance gain.

kisvegabor, 2023-12-18:
Very nice! Looking forward to s the changes. I think I will learn a lot from it :slightly_smiling_face: