lvgl/lvgl #5021: There is a correct example for integrating the library into the STM32 Cube IDE?

AlekseyMamontov, 2023-12-16:
A__ttempts to integrate into a project lead to a lot of errors.
 https://docs.lvgl.io/8/get-started/stm32.html

What were
 Have the previous recommendations lost their meaning?__
 Or has it grown to such a size that it is no longer suitable for embedded systems))?
 
 _Add_ lvgl to your project
The graphics library itself is the lvgl directory. It contains a couple of folders but to use lvgl you only need .c and .h files from the src folder.__

What should this folder look like?
We put it in the project (for example, the drivers folder), change the config file, register this folder as a library in the project settings - the result is a huge number of errors.
Well, we only need files from the src/ folder with the extension .h and .c
- and they need to be pulled out of the folders and put together or left in those folders inside the src/ folder

and so and so a lot of errors ..

A lot of wasted time.... it was easier to write something of my own.

The library becomes very cumbersome - this is no longer for embedded systems.

kisvegabor, 2023-12-18:
You can take look at this project as an example: https://github.com/lvgl/lv_port_stm32f769_disco

AlekseyMamontov, 2023-12-18:
> You can take look at this project as an example: https://github.com/lvgl/lv_port_stm32f769_disco

Thank you very much for the example) I’ll study it,
But looking through the code more and more, I come to the understanding that it’s unlikely to start on STM32F072 - 91 (Or stm32f103)

kisvegabor, 2023-12-19:
STM32F072 is really to small for LVGL. Maybe you can make it work with a lot of tweaks, but LVGL is designed for a little bit more powerfule devices.

AlekseyMamontov, 2023-12-19:
Yes, I understand STM32F072CBT6 128kb and 16kb of RAM is probably not enough (although maybe not), although if you have access to an external sd_card (flash) then it’s probably possible).
I'll have to write something of my own))

kisvegabor, 2023-12-20:
There are simple UIs out there as well. But it's something for something. In a simpler lib there will be less widgets, styles and features, but they can fit into smaller MCUs too.

AlekseyMamontov, 2023-12-21:
_There are simple UIs out there as well. But it's something for something_.

Are these lightweight themes in this project?
or are you talking about some other projects?

kisvegabor, 2023-12-22:
I mean some other projects.

closed 2024-01-13 by lvgl-bot