lvgl/lvgl #5029: Dash width and gap are rendered incorrectly on horizontal lines.

philmoz, 2023-12-18:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
Horizontal lines with dash_width and dash_gap are not drawn correctly.

E.G. If dash_width = 1 and dash_gap = 1 then the result is a solid horizontal line instead of a dotted line.

Vertical lines are drawn correctly.

### To Reproduce
```c
  lv_style_t style;
  lv_style_init(&style);
  lv_style_set_line_width(&style, 1);
  lv_style_set_line_dash_width(&style, 1);
  lv_style_set_line_dash_gap(&style, 1);

  lv_point_t pts[2] = {{50, 50}, {250, 50}};

  lv_obj_t* line = lv_line_create(lv_scr_act());
  lv_obj_add_style(line, &style, LV_PART_MAIN);
  lv_line_set_points(line, pts, 2);
```

### Expected behavior
Dash settings should render correctly.

### Screenshots or video
dash_width = 1 and dash_gap = 1
[image]

dash_width = 2 and dash_gap = 2
[image]

The problem is in src/draw/sw/lv_draw_sw_line.c lines 171 - 172.
Should be:
```c
                        if(dash_cnt < dsc->dash_width) {
                            int16_t diff = dsc->dash_width - dash_cnt - 1;
```

kisvegabor, 2024-01-09:
Sorry, I missed this issue.

kisvegabor, 2024-01-09:
I can't look into it now, but will check it soon.

PR #5473 (2024-01-25): fix(draw_line): fix the issue where dash_dap equals 1 and cannot display properly when the line is a horizontal line
### Description of the feature or fix

fix #5029
fix(draw_line): fix the issue where dash_dap equals 1 and cannot display properly when the line is a horizontal line

### test code:
```c
void lv_line_dash_dap_test(void)
{
  static lv_point_precise_t line_points[2] = { {50, 50}, {250, 50} };

  lv_obj_t * line1;
  line1 = lv_line_create(lv_screen_active());
  lv_line_set_points(line1, line_points, 2);
  lv_obj_set_style_line_width(line1, 1, LV_PART_MAIN);
  lv_obj_set_style_line_ [... cut]
closed 2024-01-25 by kisvegabor