lvgl/lvgl #5033: [Feature Request] Allow to freeze widget property update

XuNeo, 2023-12-18:
### Introduce the problem

Normally, after widget is created, user then set lots of properties like image rotation, scale etc.
It takes time to calculate the new property and it's not necessary to update them immediately because of following properties will force to do another update.

### Examples and cases

`lv_image_set_rotation` and `lv_image_set_pivot` will both calculate transformed area to invalidate.

### Suggested solution

Add feature to freeze widget's property update and API to refresh all properties at once.

kisvegabor, 2023-12-18:
Are there any other affected widgets too, besides `lv_image`?

djixon, 2023-12-29:
All objects should have BEGIN_UPDATE and END_UPDATE methods. Those two do following:
1. Each object should have an internal uint32_t UpdateRequestCounter which is default to 0 upon object creation
2. Every call to BEGIN_UPDATE(lv_obj_t* obj) simply increase that counter by ONE
3. Every call to END_UPDATE(lv_obj_t* obj) simply decrease that counter by ONE, and test if the counter reached zero, perform forced refresh
4. Refresh of an object (and all of its children) is skipped if UpdateRequestCounter is larger than 0

That means, a user can call begin_update(lv_obj_t* MyList) which will increase that counter in MyList, and it is NOT going to be refreshed (nor any of its children) until user call end_update(lv_obj_t MyList) which will decrease that counter to zero which will perform "automatic" refresh. Such mechanism can be chained, because many "begin_updates" from different routines which works on lets say lot of items (member, style properties etc) can increase that counter several times in those internal calls. However, the only really matter is that EVERY begin_update() has its own end_update(). That way even in internal routines which require (or operates) over several dependent items (parameters ...) the same mechanism can be used without any limitation.  So it is universal for both, internal routines and a user routines. 

There are lot of different situations for requirement of such mechanisms. From setting a "billion" different properties related to style (or changing a lot of them during runtime) up to deleting many of Items in the list, or deleting lot of aligned children. Simply, you do not want to refresh all aligned children after each child (item, style property) is deleted (changer, repositioned) until you complete all changes (deletion, reposition or whatever). So user, can decide when he want to use that feature or simply ignore it in which case code will work like it works now.

kisvegabor, 2023-12-30:
> That means, a user can call begin_update(lv_obj_t* MyList) which will increase that counter in MyList, and it is NOT going to be refreshed (nor any of its children) [...]

Let's say the `lv_update_begin(list)` is called and and after that many buttons, labels, images, etc are created and in the end `lv_update_end(list)` is called. It means when a list button is created, it needs to check if the counter of any parent is not zero, and if not zero, skip the button's update. 

On update end, all children and grand children needs to be updated. 

However what if, the list is ready we call `lv_update_begin(list)` do a few things with the list only and call `lv_update_end(list)`. It will also update all the children for no reason.

---

It not fully clear to me what kind of updates we can prevent. I have a few ideas for generic features:
- style updates
- layout updates
- area invalidation

It's less clear how to handle widget specific properties. (E.g. slider range)

---

It could be used to replace the current postponed layout calculation too. If the users need to disable layout calculation they can call `lv_update_being/end` but without it LVGL will always recalculate the coordinates. (it's fast enogh for most of the use cases) It would fix the issue of `lv_obj_get_width` returns `0` after creating a widget which is quite counter intuitive and causing trouble for many users.

djixon, 2023-12-31:
Slider range is not big deal, you can always make a function which will accept all 3 parameters (min, max and value) and update all 3 parameters of the slider inside the self followed by single invalidate at the end. So you wont update (invalidate, refresh nor repaint) a slider, until all are three properties are modified. 

> However what if, the list is ready we call lv_update_begin(list) do a few things with the list only and call lv_update_end(list). It will also update all the children for no reason.

It depends. Suppose that one of "few things with the list" user want to change, is padding and suppose another is BorderWidth.  Without any mechanism which controls the update process, once you change the padding it will trigger updating of all the children anyway (content inside each child is not changed, however, by changing padding of container it will trigger repositioning of each child so in that sense each child coordinate has to be recalculated), but after that you change the BorderWidth and it also impact all the children of the list (in the same sense). So you have TWO "Full Items" repositioning and redrawn, but by involving lv_update_begin()/ end()  you have just ONE. Because, lv_update_begin(container) will "disable" repainting (internal calc still works, but repainting is disabled) of children, so changing value of padding will recalculate new positions (but they will not be reflected visually), then changing border width will be applied to already changed values and it will not trigger repainting. Su you can modify many properties of the container and those will not trigger repainting until you are done. Then just calling lv_update_end(container) will allow all those changes become reflected visually too, by calling single repaint (invalidate, refresh or whatever).    
However, if "few things with the list" are border color and BorderWidth, those two will make just single "Full items) refresh (because changing border color of container does not impact children at any way) so only border width does. In that cas you do not need to call begin_update / end_update at all.

kisvegabor, 2024-01-01:
So we need to record what kind of changes happened while the update was disabled. It seems we need to handle [these cases](https://github.com/lvgl/lvgl/blob/master/src/core/lv_obj_style.h#L43-L46).  (I've just noticed that the comments are incorrect)

djixon, 2024-01-02:
Well, if  we are talking strictly about the changes made to a style, then yes. You can have an ENUM, of all possible "update kinds" determining what an object has to do after different properties changed their values. For example, in the case of style properties like Padding and BorderWidth, only repositioning of children is required (and of course, re-painting of those elements on the container itself), so in private SETTER functions like: 
void SetStyleBorderWidth(int val){
if(val != ActualBorderWidth){
  ActualBorderWidth = val;
  Requires |= e_BORDER_REPAINT | e_REPOSITIONING_CHILDREN;   
 }
}   

while in the case of, let's say

void SetStyleBorderColor(lv_color val){
if(val != ActualBorderColor){
  ActualBorderColor = val;
  Requires |= e_BORDER_REPAINT;   
 }
}   

That way, any change, to ANY of style properties "records" what kind of updates are required once the UpdateCounter reaches zero and single refresh is called. Then REQUIRES can be cleared to 0 and so on.

kisvegabor, 2024-01-17:
It can be a good candidate for the next round of [shaping](https://docs.lvgl.io/master/CONTRIBUTING.html#feature-development-workflow).

closed 2024-02-11 by lvgl-bot