lvgl/lvgl #5037: `buf_clear_cb` and `buf_copy_cb` should not use GPU

XuNeo, 2023-12-18:
### Introduce the problem

User application could call lv_draw_buf_clear and renderer could also use GPU in the same time.

### Examples and cases

https://github.com/lvgl/lvgl/blob/3ececbfc93670af6bdaf4843ff3cb289a770fb04/src/draw/lv_draw.c#L389

### Suggested solution

Make it clear for GPU portings, take it into consideration or use CPU only.

cc @FASTSHIFT  Please comment if you have more questions.

FASTSHIFT, 2023-12-18:
Yes, the GPU API is not necessarily thread-safe, and multi-threaded use may cause thread-safety issues.

kisvegabor, 2023-12-18:
Besides we cannot start a GPU operation randomly as the GPU might work on a draw_task. 

I agree to keep it CPU only. What do you think about moving it to `lv_draw.c` as `lv_draw_clear_buf()`?

XuNeo, 2023-12-18:
Should we simply implement them in `lv_draw_buf.c`? 
And keep the original name `lv_draw_buf_copy` or `lv_draw_buffer_copy` if we decide to change name.

kisvegabor, 2023-12-18:
It also works, just remove the `callback` option (same as you did with the `go_to_xy`).

PR #5052 (2023-12-19): refactor(drawbuf): eliminate clear/copy buffer callbacks
### Description of the feature or fix

Potential hardware conflicts can occur when the clear/copy API is invoked concurrently with the renderer. To fully leverage all hardware capabilities, an alternative approach is required.

Fix #5037
### Checkpoints
- [ ] Run `code-format.py` from the scripts folder. [astyle](http://astyle.sourceforge.net/install.html) needs to be installed.
- [ ] Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed
- [ ] Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) [... cut]
closed 2023-12-19 by kisvegabor