lvgl/lvgl #5041: Support YUV color format

terry0012, 2023-12-18:
### Introduce the problem

To support video and YUV animation, we need to expand the color format. However, 

YUV is divided into many different formats according to different storage methods such as Planar/Semi-Planar/Packed. 

There are currently 10 commonly used YUV formats according to [https://github.com/lemenkov/libyuv/blob/main/include/libyuv/video_common.h](url), but the max number of color format in lv_image_header_t is 31 as below. 

```c
  typedef struct {
    uint32_t cf : 5;            /*Color format: See `lv_color_format_t`*/
    uint32_t always_zero : 3;   /*It the upper bits of the first byte. Always zero to look like a
                                  non-printable character*/

    uint32_t reserved_1: 8;     /*Reserved by LVGL for later use*/
    uint32_t flags: 16;         /*Image flags, see `lv_image_flags_t`*/

    uint32_t w: 16;
    uint32_t h: 16;
    uint32_t stride: 16;        /*Number of bytes in a row*/
    uint32_t reserved_2: 16;    /*Reserved to be used later*/
   } lv_image_header_t;
```

We cannot enumerate all YUV formats.

### Examples and cases

### Suggested solution

1. **expand color format.**
  But `always_zero` is used to differentiate between image src types, we can't expand cf.
```c
lv_image_src_t lv_image_src_get_type(const void * src)
{
    lv_image_src_t img_src_type = LV_IMAGE_SRC_UNKNOWN;

    if(src == NULL) return img_src_type;
    const uint8_t * u8_p = src;

    /*The first byte shows the type of the image source*/
    if(u8_p[0] >= 0x20 && u8_p[0] <= 0x7F) {
        img_src_type = LV_IMAGE_SRC_FILE; /*If it's an ASCII character then it's file name*/
    }
    else if(u8_p[0] >= 0x80) {
        img_src_type = LV_IMAGE_SRC_SYMBOL; /*Symbols begins after 0x7F*/
    }
    else {
        img_src_type = LV_IMAGE_SRC_VARIABLE; /*`lv_image_dsc_t` is draw to the first byte < 0x20*/
    }

    if(LV_IMAGE_SRC_UNKNOWN == img_src_type) {
        LV_LOG_WARN("unknown image type");
    }

    return img_src_type;
}
```
2.  as discussed in https://github.com/lvgl/lvgl/issues/4869, add `LV_COLOR_FORMAT_YUV` in `_lv_color_format_t` and put the detail information of YUV in buf_sup.
```c
typedef struct {
    lv_image_header_t header;           /*A header describing the basics of the image*/
    uint32_t data_size;                 /*Size of the data in bytes*/
    const void * buf;
    const void * buf_sup;           /*Optional supplementary buffer*/
}
```

How do you think ?

kisvegabor, 2023-12-18:
We could do this as well:
```c
    uint8_t magic; //Instead of cf + always_zero => 0b10101000
    uint8_t cf;    //8 bits
    uint16_t flags: 16;       

    uint16_t w;
    uint16_t h;
    uint16_t stride;   
    uint16_t reserved;
```

What do you think? 
cc @XuNeo

XuNeo, 2023-12-18:
Add a magic on header first byte could solve it. I agree with this solution. Should include all detailed YUV formats in lvgl?

terry0012, 2023-12-19:
> Add a magic on header first byte could solve it. I agree with this solution. Should we include all detailed YUV formats in lvgl?

I think support the primary YUV formats firstly is enough.

we need to regenerate all the image  in LV_IMAGE_SRC_VARIABLE format if we add magic number.

XuNeo, 2023-12-19:
The images in lvgl repo will be updated soon with latest script.
Below code allows us to replace all images gradually. Otherwise, merging this patch will break everything immediately. A warning log is included so we all know it.

https://github.com/lvgl/lvgl/blob/5ad6ca2eeb9f49a69a523ef464e672e196a892d4/src/libs/bin_decoder/lv_bin_decoder.c#L126-L135

PR #5051 (2023-12-19): refactor(image): add magic to image header

### Description of the feature or fix

Fix #5041
So the cf field is not limited to 5bits, to add more color format options. A workaround is included, so the existing bin images can still work.

### Checkpoints
- [ ] Run `code-format.py` from the scripts folder. [astyle](http://astyle.sourceforge.net/install.html) needs to be installed.
- [ ] Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed
- [ ] Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant. 
- [ ] Add [Tests](https://github.com/lvgl/ [... cut]
closed 2023-12-19 by kisvegabor