lvgl/lvgl #5044: New core member

kisvegabor, 2023-12-18:
Hi,

In the past half year we had weekly meetings with Xiaomi to discuss all kinds of development related questions. During this time I made sure of that they have a really good mindset and imaging the future of LVGL the same way as I do. Therefore, I have invited @FASTSHIFT to the LVGL repository with write permission. 

Starting in January, we'll work together to establish precise collaboration standards, ensuring a productive and aligned cooperation.

FASTSHIFT, 2023-12-19:
Thank you Gabor and everyone for your recognition and support. In the future, let's work together to create a better community and build a faster and stronger LVGL graphics library!

kisvegabor, 2023-12-21:
I go to Christmas holiday from tomorrow and will be back on January 3. Until that

### Mr. @FASTSHIFT, the bridge is yours! 

I'll check my mails in ~2-3 days. I don't anticipate any new larger updates and changes before releasing v9, so so feel free to merge to ongoing PRs (cache, image decoder related, crown) and the simpler new ones as well.

kisvegabor, 2023-12-21:
Just one note: please either `rebase` or `squash` the PRs to keep the linear and clean commit history.

FASTSHIFT, 2023-12-21:
@kisvegabor 
No problem~

kisvegabor, 2024-01-03:
I'm back to work.
Thank you for the great work! I was really happy to see that the development went very well in these days: :hugs:

kisvegabor, 2024-01-22:
Bringing @FASTSHIFT on board as a main team member has really helped us, making LVGL's development faster and better. Also, the new way we approve PRs now makes me feel a lot more sure about what we add to the project.

Lately, the work from @XuNeo, @PGNetHun, and @C47D has been amazing. They've done a lot in checking PRs, both in terms of  quality and activity. To recognize your great work and to help us work together even better, I'm happy to offer you **write access** too. 

What do you think? :slightly_smiling_face:

kisvegabor, 2024-01-22:
Absolute, not problem with that. Let's get back to it in a month.

PGNetHun, 2024-01-23:
Hello!
Thank you, but I can only help only in code reviews, investigations and small fixes, so I think write access to main LVGL repo is still early for me.
Helping in MicroPython binding is enough for me :-)

kisvegabor, 2024-01-23:
@PGNetHun Actually I feel like you are much better in adding reviews than me :smile: Having the write access wouldn't mean any extra expectations. The main difference would be only that your valuable feedback counted with a green check mark. 
By having  a core team consists of 5 people, we could increase the limit of required reviews to 2.

kisvegabor, 2024-01-24:
@XuNeo Amazing, thank you! I've just sent the invitation.

kisvegabor, 2024-02-12:
I keep it open until the Chines holiday ends.

By the way, 
## 新年快乐! :tada:

kisvegabor, 2024-03-16:
In light of https://github.com/lvgl/lvgl/issues/5791 I will have some news in this topic.

kisvegabor, 2024-03-18:
Hi All,

Welcome @liamHowatt, the new member of the LVGL team! Liam will help us maintaining LVGL by responding to issues, testing PRs, fixing bugs and implementing features. 

Liam is already familiar with LVGL, but probably a few weeks will be required to familiarize himself with the internals of the library.  Please help him to have a great on boarding experience. 😊

I'll be here too (more actively than in the previous weeks), so feel free to tag me if needed.

Liam, could you tell a few words about yourself?

liamHowatt, 2024-03-18:
Hey team!

Thanks Gabor! It's great to be here! About me: my education is in engineering and my professional experience and hobbies are in firmware. I'm based in Ontario, Canada. I think LVGL is a super exciting thing and I want to bring the best of what I can offer to this project! My experience with the members has been very positive and I am so thrilled to be a part of it! As Gabor said, I'm building up a full understanding of the internals while doing collaborative maintenance with the team and community!

kisvegabor, 2024-04-04:
Closing it for now. I will reopen this issue if needed.

kisvegabor, 2024-05-21:
@liamHowatt has been doing an awesome job these past few months and has proven to be a perfect fit for our maintainer team. To help him contribute even more, I’ve given him Write access. Now Liam can also approve pull requests. :tada:

kisvegabor, 2024-06-20:
Closing this issue for now.

kisvegabor, 2025-04-28:
Hi All,

I have good news! Probably you have already noticed the great reviews and contributions from@AndreCostaaa recently. We definitely did so we asked Andre if he want to join the LVGL team. 

Andre said yes, so from now he will officially help us to review issues and PRs. 🎉  

In light of that I granted write access to Andre.

AndreCostaaa, 2025-04-28:
Thank you so much! 🎉

I'm excited to officially join the team and help out more with reviews and contributions
Looking forward to working even more closely with everyone ! 🚀

kisvegabor, 2025-05-20:
Closing for now.

closed 2025-05-20 by kisvegabor