lvgl/lvgl #5047: Performance of 'lv_textarea'

philmoz, 2023-12-19:
### Introduce the problem

While working on the EdgeTX firmware I noticed that the lv_textarea object can be quite slow.
We are using lvgl  8.2 with some local patches. 
For the most part lvgl performs well; but lv_textarea is problematic.
We recently tried 8.3; but reverted because it was significantly slower than 8.2 for us. Having looked at the code in 8.3 I suspect it will have the same concerns for lv_textarea.

### Examples and cases

One example is our firmware running on a RadioMaster TX16S radio. One particular page takes around 750ms to build and render with a lot of this time being spent creating 7 lv_textarea based objects.

### Suggested solution

After doing some timing and experimentation a lot of the slow performance comes from these areas:
- lv_textarea_set_cursor_pos take a long time even with no text in the label - approx 9ms on the TX16S. It is called from the lv_textarea_constructor function as well as when text is added so it can get called multiple times during object construction. I'm wondering if lv_textarea_set_cursor_pos should return straight away if style_refr is false which might reduce the overhead while building an object.
- lv_textarea_constructor calls lv_label_create which call lv_obj_class_init_obj. Before the call to lv_label_create the style_refr variable is false so style changes do not cause a refresh; but after the call to lv_label_create is gets reset to true. So subsequent style changes in lv_textarea_constructor trigger refreshes. This is exacerbated if lv_text_area has been sub-classed as the style changes in the sub class also trigger refreshes.

Suggestions:
- Make style_refr a counter instead of a boolean value. It is incremented by lv_obj_enable_style_refresh(false) and decremented by lv_obj_enable_style_refresh(true) calls. Style refreshing only happens when style_refr == 0. This will prevent style refresh being reset on nested object creation.
- Add a helper function to return the state of style refreshing (true when style_refr == 0).
- Return immediately from lv_textarea_set_cursor_pos if style refresh is disabled.

After experimenting with these changes I can reduce the setup time of the page I mention above to ~450ms.

philmoz, 2023-12-19:
An alternative for nested object creation and style refresh might be to save the current value of style_refr at the start of lv_obj_class_init_obj and then call lv_obj_enable_style_refresh with the previous value instead of always calling with false (after object construction).

XuNeo, 2023-12-19:
> * lv_textarea_set_cursor_pos take a long time even with no text in the labe

Have you tried to disable default value by `LV_WIDGETS_HAS_DEFAULT_VALUE`? It could speed up a little with no default text.

I have encountered similar  problem, it looks like we should provide a solution for both. https://github.com/lvgl/lvgl/issues/5033

kisvegabor, 2023-12-19:
Hi,

How do you use the fonts and where do you store them? In internal flash, or are there any tricks? 

Normally it shouldn't take that much time, except if it's very slow to read the glyphs' (characters') data.

philmoz, 2023-12-19:
The default font used for lv_textarea is store in flash.
All our fonts are generated using lv_font_conv.

philmoz, 2023-12-19:
I've done some more timing tests and the time is being spent in the call to 'lv_obj_update_layout' within the 'lv_textarea_set_cursor_pos' function.

kisvegabor, 2023-12-20:
Could you send a small code example to reproduce the issue?

philmoz, 2023-12-20:
Running in the simulator on an iMac Pro - 3.2GHz 8-core Intel Xeon.

The code below takes 247us to create the lv_textarea. Of this 189us (76%) is spent in the lv_obj_update_layout function called from lv_textarea_set_cursor_pos (which itself is called 6 times while creating a single text area).

IMHO this is slow. When you move this to an MCU running at a couple of hundred MHz it takes a very long time.

```c
  lv_obj_t* ta = lv_textarea_create(lv_scr_act());
```

kisvegabor, 2023-12-20:
I've tested with LVGL v9 and this code on an STM32F769 (200MHz)
```c
	volatile uint32_t t = lv_tick_get();

	for(int i = 0; i < 100; i++) {
		lv_textarea_create(lv_screen_active());
	}

	t = lv_tick_elaps(t);  //158 -> 1.58ms per text area
```

And this this the profiler output:
[image]

So it's mainly getting style properties.

---

As a comparison this to the result for creating  100 sliders:
`t = 18` however the textarea is build from 2 widgets, so proportionally it's 36ms. Which is really much faster, but the text are is not that slow either.  
[image]

How much time does it take on your embedded hardware to create 100 or 10 textareas? What is clock speed of your device?

philmoz, 2023-12-20:
Running on STM32F429BIT6 at 168MHz with version 8.2.

100 textareas takes 1150ms = 11.5ms per textarea.

kisvegabor, 2023-12-20:
Have you enabled compiler optimization?

philmoz, 2023-12-20:
Everything is compiled with -Os optimisation.

philmoz, 2023-12-20:
-O2 gives only a slight improvement - 9.9ms per textarea. Unfortunately it also dramatically increases the overall build size which is a problem.

philmoz, 2023-12-21:
lv_textarea slows down even more if it is created inside a Flex layout - 16.5ms per textarea!

kisvegabor, 2023-12-21:
I'm really puzzled. We are testing on similar hardware (200 MHz vs 168MHz) and it's 10 times slower for you. The only difference I know is that I've tested with LVGL v9, but shouldn't matter much.

Some other questions:
- Have you enabled I and D cache?
- Are you sure the flash wait states are set correctly?
- Are you sure the clock speed is really set the 168MHz?

philmoz, 2023-12-22:
All caching is enabled. Flash has 5 wait states as recommended. Clock speed is 168MHz.

I have found that the parent object passed to lv_textarea_create has a significant effect on performance.
Our object hierarchy looks like this:
  Background (styles for bg color, bg opacity and size)
     Page (styles for bg color, bg opacity and size)
         Body (styles for scroll bar and padding)
            text area objects

In all previous tests lv_textarea_create is called with the Body object as the parent.
If, instead, I call lv_textarea_create(lv_layer_top()), then call lv_obj_set_parent(ta, Body) after finishing setting up the textarea object I get significantly better performance - 4.3ms per textarea instead of 11.5ms.

kisvegabor, 2023-12-25:
Please send a full code snippet with all the parents so that I can test the same thing.

C47D, 2023-12-25:
@kisvegabor maybe a dummy question, but the interface between MCU and screen is different between both projects?

philmoz, 2023-12-25:
> Please send a full code snippet with all the parents so that I can test the same thing.

Travelling at them moment. Will post something in a few days.

philmoz, 2024-01-02:
This is an example of what the layout in question would be (the actual code from our project would difficult to use here due to the complexity of the code).

```c
  static const lv_style_const_prop_t pad_small_props[] = {
    LV_STYLE_CONST_PAD_TOP(4),  LV_STYLE_CONST_PAD_BOTTOM(4),
    LV_STYLE_CONST_PAD_LEFT(4), LV_STYLE_CONST_PAD_RIGHT(4),
    LV_STYLE_CONST_PAD_ROW(4),  LV_STYLE_CONST_PAD_COLUMN(4),
    LV_STYLE_PROP_INV,
  };
  static LV_STYLE_CONST_INIT(pad_small, pad_small_props);

  static const lv_style_const_prop_t scrollbar_props[] = {
    LV_STYLE_CONST_BG_OPA(LV_OPA_50),
    LV_STYLE_CONST_PAD_TOP(3),  LV_STYLE_CONST_PAD_BOTTOM(3),
    LV_STYLE_CONST_PAD_LEFT(3), LV_STYLE_CONST_PAD_RIGHT(3),
    LV_STYLE_CONST_WIDTH(4),    LV_STYLE_PROP_INV,
  };
  static LV_STYLE_CONST_INIT(scrollbar, scrollbar_props);

  lv_obj_t* mainWin = lv_obj_create(lv_scr_act());
  lv_obj_set_size(mainWin, 480, 272);
  lv_obj_set_pos(mainWin, 10, 10);
  lv_obj_set_style_bg_color(mainWin, lv_palette_main(LV_PALETTE_GREY), 0);
  lv_obj_set_style_bg_opa(mainWin, LV_OPA_100, 0);

  lv_obj_t* pageWin = lv_obj_create(mainWin);
  lv_obj_set_size(pageWin, 480, 272);
  lv_obj_set_pos(pageWin, 0, 0);

  lv_obj_t* header = lv_obj_create(pageWin);
  lv_obj_add_style(header, &pad_small, 0);
  lv_obj_set_size(header, 480, 45);
  lv_obj_set_pos(header, 0, 0);
  lv_obj_set_style_bg_color(header, lv_palette_main(LV_PALETTE_BLUE), 0);
  lv_obj_set_style_bg_opa(header, LV_OPA_100, 0);

  lv_obj_t* body = lv_obj_create(pageWin);
  lv_obj_add_style(body, &pad_small, 0);
  lv_obj_add_style(body, &scrollbar, 0);
  lv_obj_set_size(body, 480, 227);
  lv_obj_set_pos(body, 0, 45);
  lv_obj_set_style_bg_color(body, lv_color_white(), 0);
  lv_obj_set_style_bg_opa(body, LV_OPA_100, 0);
  lv_obj_set_style_max_height(body,227, 0);
  lv_obj_set_scrollbar_mode(body, LV_SCROLLBAR_MODE_AUTO);

  lv_obj_set_flex_flow(body, LV_FLEX_FLOW_ROW_WRAP);
  lv_obj_set_flex_align(body, LV_FLEX_ALIGN_CENTER, LV_FLEX_ALIGN_START, LV_FLEX_ALIGN_CENTER);

  for (int i = 0; i < 100; i += 1)
  {
    lv_obj_t *ta = lv_textarea_create(body);

    lv_obj_set_size(ta, 60, 32);
    lv_textarea_set_password_mode(ta, false);
    lv_textarea_set_one_line(ta, true);
    lv_textarea_set_max_length(ta, 10);
    lv_textarea_set_text(ta, "TEST");
  }
```

kisvegabor, 2024-01-09:
Sorry for the late reply. I just debugged it and I can confirm that the layout update takes significant time in `lv_textarea_set_cursor_pos`.

The loop in your code runs in 900 ms for me. If I keep only `lv_textarea_create()` it's 400ms. If I also return immediately from `lv_textarea_set_cursor_pos`. it's 67ms. So 60 times faster. 

The problem is that `lv_textarea_set_cursor_pos` needs to update layouts to figure out where the the character of the text are.

What can we do:
- Make layout calculation much faster: complicated, and probably can't be 60 times faster
- Allow temporarily disabling the layout calculations: simple, and solves other issues too.
- Calculate the cursor position only during rendering if it was updated: not too complicated and transparent to the user, but related only this specific issue.