lvgl/lvgl #5054: Use `image_buf` instead of `draw_buf`

XuNeo, 2023-12-19:
### Introduce the problem

The newly introduced `draw_buf` has identical definition with `lv_image_dst_t`. 
We can use the existing `lv_image_buf` naming. The `lv_image_buf_free` is currently not used by `lvgl`.

> `lv_image_create()` is already used by the `lv_image` widget, however `lv_image_buf_*` could work. 
> 
> It really opens to door for some simplification and removal of duplication. 
> 
> Let's try it, if it really looks good we are still on time to add this change. 
> 
> I merge it now, to no block the development of with nitpicking on naming.

_Originally posted by @kisvegabor in https://github.com/lvgl/lvgl/issues/5034#issuecomment-1860407825_
            
### Examples and cases

### Suggested solution

XuNeo, 2023-12-29:
cc @FASTSHIFT .
I think currently people cannot distinguish draw buffer from image. 
See the title of this issue. https://github.com/lvgl/lvgl/issues/5127

Separating draw buffer with image seems not a good choice from my point of view.

kisvegabor, 2023-12-30:
I agree to unify `lv_image_buf` and `lv_draw_buf`. Less building blocks is better.

kisvegabor, 2024-01-14:
no stale

kisvegabor, 2024-01-30:
Being discussed in #5496 too

closed 2024-02-18 by XuNeo