lvgl/lvgl #5074: Request for Enhanced Scroll Prediction in LVGL

XuNeo, 2023-12-21:
### Introduce the problem

The existing scroll behavior in LVGL lacks the desired responsiveness, falling short when compared to standard PC or smartphone scrolling experiences. The scroll distance doesn't adapt to the user's scroll speed very well.

### Examples and cases

Scroll a long list in LVGL and compare the experience to scroll a long page in browser.

### Suggested solution

I don't know how to improve it yet. I hope somebody knows and we can discuss it.

kisvegabor, 2023-12-22:
I agree that it could be improved. 

Now it works like this:
1. Calculate a vector:
```c
    /*Calculate the vector and apply a low pass filter: new value = 0.5 * old_value + 0.5 * new_value*/
    indev->pointer.vect.x = indev->pointer.act_point.x - indev->pointer.last_point.x;
    indev->pointer.vect.y = indev->pointer.act_point.y - indev->pointer.last_point.y;

    indev->pointer.scroll_throw_vect.x = (indev->pointer.scroll_throw_vect.x + indev->pointer.vect.x) / 2;
    indev->pointer.scroll_throw_vect.y = (indev->pointer.scroll_throw_vect.y + indev->pointer.vect.y) / 2;
```
2. Move the widget with the vector and reduce the size of the vector:
```c
int32_t scroll_throw = indev->scroll_throw;
...
indev->pointer.scroll_throw_vect.x = indev->pointer.scroll_throw_vect.x * (100 - scroll_throw) / 100;
```

I don't know what could be a better algorithm. :slightly_frowning_face:

xlbao123, 2023-12-25:
I'd like to share a similar logic I've implemented before:
1. Given the friction coefficient, provide an api for users to set it within the range of 0 to 1, with a default of 0.96f.
2. Clear the previous speed when a touch press occurs.
3. When there is a touch move or touch release, calculate the real-time speed in the x and y directions (assuming a tick is 16ms at 60Hz):
```
speed.x = (point.x - last_point.x) / MS2_2_TICKS(point.ts - last_point.ts)
```
Update the speed using a weighted average:
```
current->speed.x = 0.2f * current->speed.x + 0.8f * speed.x
```
4. Upon touch release, calculate the scroll distance and the duration ticks in the x/y directions based on friction and the latest speed:
```
distance = 0;
 if (fabs(speed) > EPSILON) {
     ticks = (int)floor(log(1.0f / fabs(speed)) / log(friction));
     for (int i = 0; i < ticks; i++) {
         distance += (int)speed;
         speed *= friction;
     }
 }
```
5. The calculated `ticks` can be converted into a duration in milliseconds. Then, based on the x and y direction distances and duration, set up the animation accordingly.

kisvegabor, 2023-12-25:
Hi,

Thanks for sharing. Unless I miss something, we are doing something similar now. I think the main difference between this and phones is phones keep the speed for some time, and start to slow down only after that. 

In LVGL the speed vs time look like this:
```
********
        *******
               *****
                    ***
                       **
                         *
```

On phones
```
****************************
                            *******
                                   *****
                                        ***
                                           **
                                             *
```
The length of constant speed is proportional to the size of the scroll vector. It's not that difficult to implement, but I'm not that's what we need.

XuNeo, 2023-12-29:
We can custom an anim-path instead of `lv_anim_path_ease_out`.
It's important to try the new path on touch screen instead of simulator with mouse. It may feel different.

> we are doing something similar now

I agree with `[xlbao123](https://github.com/xlbao123)`. 
I think the main difference is the scroll predictor should output both distance and speed. 

The anim-path should take both into consideration. Scroll speed should starts with current calculated and slow down gradually(like using ease-out path). Calculate the distance by integrating speed over time.

kisvegabor, 2023-12-30:
After some Googling I found only [this](https://medium.com/homullus/recreating-native-ios-scroll-and-momentum-2906d0d711ad). TLDR: in iOS `new_speed = old_speed * 0.95`

kisvegabor, 2024-01-14:
Not stale

W-Mai, 2024-01-23:
I think the action of throwing hands is like throwing a square, so the algorithm should be consistent with the algorithm of throwing a square in the physical world.

$$ x = \frac{1}{2} * a * t ^ 2 + v_0 * t $$

We make differentiation on it. So, 

$$ \Delta x = a * t * \Delta t + v_0 * \Delta t = (a * t + v_0) * \Delta t $$

We can simplify the equation to

$$ x = k * t ^ 2 + v_0 * t$$

$$ \Delta x = (a * t + v_0) * \Delta t $$

So, 
- the `x` is the total distance it will scroll. 
- `Δx` is the delta value each iteration the obj will be moved.
- `Δt` is the minimum time tick or the time delta got from lv_get_tick() between each iteration.
- `a` is a value depending on the speed of throwing the hand and the possibility is related to the size of the component (weight)
- `v_0` is the initial speed

I think following the laws of physics will make the UI more comfortable. 😁

XuNeo, 2024-02-11:
Not stale.

kisvegabor, 2024-03-16:
@W-Mai what is plan with this feature?

closed 2024-05-27 by lvgl-bot