lvgl/lvgl #5075: lv_async_timer_cb seems running twice somtimes for one lv_async_call.

lanistor, 2023-12-21:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

I run lv_async_call in child thread, using `xTaskCreatePinnedToCore` in `ESP32`, `lv_async_timer_cb` seems running twice somtimes for one `lv_async_call`.

Here is my log code in `lv_async.c`:
```c
static void lv_async_timer_cb(lv_timer_t* timer) {
  printf("#### lv_async_timer_cb 1 \n");
  lv_async_info_t* info = (lv_async_info_t*)timer->user_data;
  printf("#### lv_async_timer_cb 2 \n");
  if (info && info->cb) {
    info->cb(info->user_data);
    printf("#### lv_async_timer_cb 3 \n");
    lv_mem_free(info);
    timer->user_data = NULL;
  }
  printf("#### lv_async_timer_cb 4 \n");
}
```
and here is the log:
```c
#### lv_async_timer_cb 1
#### lv_async_timer_cb 2
#### lv_async_timer_cb 3
#### lv_async_timer_cb 4
#### lv_async_timer_cb 1
#### lv_async_timer_cb 2
#### lv_async_timer_cb 4
```

If don't add `timer->user_data = NULL;`, this will cause crash with log: 
```
Guru Meditation Error: Core  0 panic'ed (InstructionFetchError). Exception was unhandled.

Core  0 register dump:
PC      : 0x3f803014  PS      : 0x00060e30  A0      : 0x801593e2  A1      : 0x3ffdefe0
A2      : 0x3fa61ce4  A3      : 0x00000000  A4      : 0x4015818c  A5      : 0x00060c23
0x4015818c: lv_async_timer_cb at /Users/lanistor/C/flicker/Hardware-EInk_Desktop_Display/components/lvgl/src/misc/lv_async.c:100

A6      : 0x00060c20  A7      : 0x00000001  A8      : 0x8015819b  A9      : 0x3ffbbec8
A10     : 0x3f803014  A11     : 0x00004272  A12     : 0x00000000  A13     : 0x00060c23
A14     : 0x00060c23  A15     : 0x00000001  SAR     : 0x0000001d  EXCCAUSE: 0x00000002
EXCVADDR: 0x3f803014  LBEG    : 0x4008488d  LEND    : 0x40084895  LCOUNT  : 0x00000027
0x4008488d: esp_timer_impl_get_counter_reg at /Users/Shared/SDK/esp/esp-idf/components/esp_timer/src/esp_timer_impl_lac.c:118 (discriminator 2)

0x40084895: esp_timer_impl_get_counter_reg at /Users/Shared/SDK/esp/esp-idf/components/esp_timer/src/esp_timer_impl_lac.c:119 (discriminator 1)
```
The probability of occurrence is accidental, i think maybe the task runned throw two`lv_task_handler`?

### To Reproduce

Create a child task like this:
```cpp
xTaskCreatePinnedToCore(&_async_hardware_cb, name.c_str(), stack_size, arg,
                            priority, &arg->xHandle, core);
```
call `lv_async_call` in this child task and do something.

### Expected behavior

No crash, `lv_async_timer_cb` run only once.

### Screenshots or video

XuNeo, 2023-12-24:
LVGL is not thread safe. If multiple thread is needed, you can provide a global lock to lvgl. Get the lock before accessing any lvgl API.

lanistor, 2024-01-06:
> LVGL is not thread safe. If multiple thread is needed, you can provide a global lock to lvgl. Get the lock before accessing any lvgl API.

I run `lv_task_handler` just in main loop thread, but run `lv_async_call` in multiple threads.
I think `lv_async_call` just push the callback to the run list, then the callback of `lv_async_call` only run in main thread, so this is not multiple-thread scene ?

kisvegabor, 2024-01-07:
> I think lv_async_call just push the callback to the run list, then the callback of lv_async_call only run in main thread, so this is not multiple-thread scene ?

`lv_async_call` creates a  new one shot timer, so if it's called from an other thread it can mess up the timer list if the main thread if also using it.

lanistor, 2024-01-07:
Got it, thanks.

closed 2024-01-07 by lanistor