lvgl/lvgl #5090: Setting LV_DRAW_BUF_STRIDE_ALIGN to a value rather than 1 will cause display glitches

GorgonMeducer, 2023-12-24:
### Perform all steps below and tick them with [x]
- [ ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
I do not fully understand the purpose of the option `LV_DRAW_BUF_STRIDE_ALIGN`, but when I set it to `4`, there are obvious display glitches.

### To Reproduce

### Expected behavior

### Screenshots or video

XuNeo, 2023-12-24:
`LV_DRAW_BUF_STRIDE_ALIGN` determines the bytes alignment of draw buffer stride.
E.g. if `64` is used, for a draw buffer with 100x100, ARGB8888, stride will be `ALIGN_UP((100*4), 64)` = 512.

For CPU draw unit and GPU that has no such requirement, `LV_DRAW_BUF_STRIDE_ALIGN` can be configured to 1.

GorgonMeducer, 2023-12-24:
The problem is, 4 is a valid opinion but it causes glitches.

XuNeo, 2023-12-25:
Could you show the steps to reproduce it?

GorgonMeducer, 2023-12-25:
Set the macro I mentioned to 4 and run benchmark test. 

If you turn on performance monitor, you will saw similar glitches very often.

kisvegabor, 2023-12-30:
It works well in the [Eclipse simpualtor](https://github.com/lvgl/lv_port_pc_eclipse) so I assume that it's a driver issue.
Do you consider the stride align = 4 in the flush function too?

GorgonMeducer, 2023-12-31:
@kisvegabor, The driver supports aligned access. As long as there is no bus fault or hardfault, even an unaligned address is used, the data would be written to the correct address.

GorgonMeducer, 2023-12-31:
Happy new year!

kisvegabor, 2024-01-01:
Happy new year :slightly_smiling_face: 

In PARTIAL mode let's say LVGL wants to update a 13px wide are, which is 26 bytes with RGB565 color format. However this area also respects the stride, so the rendered image will have 28 byte stride (assuming 4 byte alignment) with 26 bytes of real data.

GorgonMeducer, 2024-01-01:
@kisvegabor 
I see the problem. And this is a problem we have to face:

- when we pass arguments to `disp_flush`
```c
static void disp_flush(lv_display_t * disp_drv, const lv_area_t * area, uint8_t * px_map)
{

}
```
the `area` contains the x1, y1, x2, y2 but not the stride information.  Where can we get the stride information?

- A lot of LCD has a window mode, which allows you to set a rectangular area on the LCD display ram and you can fill the memory continuously.

```c
        GLCD_SetWindow(x, y, width, height);

        wr_cmd(0x22);
        wr_dat_start();
        
        
        uint32_t i = width * height;
        do {
            wr_dat_only(*ptr_bmp++);
        } while(--i);
```

[image]

For supporting the window mode:
- We have to setup a window which has the stride as the width 
    - So not only the valid area is included, but also
    - the content undefined area is also included.

GorgonMeducer, 2024-01-01:
I have checked the SDL driver:

https://github.com/lvgl/lvgl/blob/4018ef89e14b2d2dae52006c21787f7bed0175fd/src/dev/sdl/lv_sdl_window.c#L186-L215

It doesn't handle the fb stride alignment correctly.

XuNeo, 2024-01-02:
What we need to do is pass the `lv_draw_buf_t` instead of `px_map` to flush_cb. Driver will need to decide how to deal with case of `stride != width*pixel_byte`.

Also we need to add a new API to return a window of draw buffer. This will be used in non-directmode. I have included it in the original PR, but remove at the end for simplification.

https://github.com/lvgl/lvgl/commit/24783005eeeb0bdfd406a2a794413e1abd2165b0#diff-58fe28c3aa2448a713b0e2a82dfe922a43457d06ff4d3ff40314fb5624ad79db

```c
/**
 * Return a window to the main draw buffer.
 * It always have same stride parameter as main buffer, thus can navigate to next line
 * of window by adding stride to current pointer.
 * @param buf   The main draw buffer
 * @param area  The area of the window inside the draw buffer
 * @return      A new draw buffer instance point to the window
 */
lv_draw_buf_t lv_draw_buf_window(lv_draw_buf_t * buf, const lv_area_t * area);
```

GorgonMeducer, 2024-01-02:
> What we need to do is pass the lv_draw_buf_t instead of px_map to flush_cb. Driver will need to decide how to deal with case of stride != width*pixel_byte.

> Also we need to add a new API to return a window of draw buffer. This will be used in non-directmode. I have included it in the original PR, but remove at the end for simplification.

This is also a good solution.

GorgonMeducer, 2024-01-03:
I manage to temporarily fix this issue by applying the following patch to the driver:

```c
static void disp_flush(lv_display_t * disp_drv, const lv_area_t * area, uint8_t * px_map)
{

    if (disp_flush_enabled) {

        do {
            lv_color_format_t color_format = lv_display_get_color_format(disp_drv);
            int32_t width = area->x2 - area->x1 + 1;
            int32_t height = area->y2 - area->y1 + 1;
            int32_t stride = lv_draw_buf_width_to_stride( width, color_format);
            width *= lv_color_format_get_bpp(color_format) >> 3;
            
            if (width == stride) {
                break;
            }
            
            uint8_t *src_ptr = px_map;
            uint8_t *des_ptr = px_map;
             
            for (int y = 0; y < height; y++) {
                memcpy(des_ptr, src_ptr, width);
                
                src_ptr += stride;
                des_ptr += width;
            }
        } while(0);

        GLCD_DrawBitmap(area->x1,               //!< x
                        area->y1,               //!< y
                        area->x2 - area->x1 + 1,    //!< width
                        area->y2 - area->y1 + 1,    //!< height
                        (const uint8_t *)px_map);
    }

    /*IMPORTANT!!!
     *Inform the graphics library that you are ready with the flushing*/
    lv_display_flush_ready(disp_drv);
}
```

I think it could be embedded into `call_flush_cb`, for example:

```c
static void call_flush_cb(lv_display_t * disp, const lv_area_t * area, uint8_t * px_map)
{
    LV_PROFILER_BEGIN;
    LV_TRACE_REFR("Calling flush_cb on (%d;%d)(%d;%d) area with %p image pointer",
                  (int)area->x1, (int)area->y1, (int)area->x2, (int)area->y2, (void *)px_map);

    lv_area_t offset_area = {
        .x1 = area->x1 + disp->offset_x,
        .y1 = area->y1 + disp->offset_y,
        .x2 = area->x2 + disp->offset_x,
        .y2 = area->y2 + disp->offset_y
    };

    
    do{
        lv_color_format_t color_format = lv_display_get_color_format(disp);
        int32_t width = offset_area->x2 - offset_area->x1 + 1;
        int32_t height = offset_area->y2 - offset_area->y1 + 1;
        int32_t stride = lv_draw_buf_width_to_stride( width, color_format);
        width *= lv_color_format_get_bpp(color_format) >> 3;

        if(width == stride) {
            break;
        }

        uint8_t *src_ptr = px_map;
        uint8_t *des_ptr = px_map;

        for(int y = 0; y < height; y++) {
            lv_memcpy(des_ptr, src_ptr, width);
              
            src_ptr += stride;
            des_ptr += width;
        }
    }while(0);

    lv_display_send_event(disp, LV_EVENT_FLUSH_START, &offset_area);
    disp->flush_cb(disp, &offset_area, px_map);
    lv_display_send_event(disp, LV_EVENT_FLUSH_FINISH, &offset_area);

    LV_PROFILER_END;
}
```

kisvegabor, 2024-01-03:
Great, now we know that the problem was really this.

@XuNeo 
> int32_t stride = lv_draw_buf_width_to_stride( width, color_format);

With the most recent stride related updates where there is no concept for global stride how can LVGL set a stride for the main draw buffer and the layers?

It seems we still need some kind of global stride, but it might be set in `lv_display_t`. 

@GorgonMeducer 
> I think it could be embedded into call_flush_cb, for example:

I think we shouldn't do this by default because it has a significant performance cost and it's not sure that all driver needs it. For example:
- In DIRECT or FULL_REFRESH modes, the LCD controller can candle the stride
- In PARTIAL mode 
    - if GPU copies to a the frame buffer, it can handle the stride too
    - if the user sends the image via SPI, stride can be handled by having an interrupt after each line  
    - if the user memcpy-s the image to the frame buffer, stride can be handled on the fly

XuNeo, 2024-01-03:
> It seems we still need some kind of global stride, but it might be set in `lv_display_t`.

There's no need for global stride. Every draw buffer has a stride in header. @FASTSHIFT is updating `lv_display_t` to use draw buffer in https://github.com/lvgl/lvgl/pull/5113. The PR is dropped for now.

See https://github.com/lvgl/lvgl/pull/5113/files#diff-67fd1b48a2bb4b3f48b602ce63408bcc5c729d6193bf05735905a7e74c0fa50cL1011

kisvegabor, 2024-01-03:
[Here](https://github.com/lvgl/lvgl/pull/5113/files#diff-67fd1b48a2bb4b3f48b602ce63408bcc5c729d6193bf05735905a7e74c0fa50cR628)  `lv_draw_buf_width_to_stride` is used to set the buf stride which assumes that there is a global stride setting which can convert a width and a color format to stride bytes.  Now `LV_DRAW_BUF_STRIDE_ALIGN` is used for this. (That's what I mean by global stride) I think we should keep it for these kind of cases. Do you agree? Independently of that GPUs can decode images with any stride they like.

XuNeo, 2024-01-03:
I don't get it. This case is perfectly covered by the draw buffer idea, where display and layer has a `lv_draw_buf_t` that contains rendered result that ready to be flushed to screen.
Driver should take the stride into consideration because it says GPU has limitation. If no GPU is used, just set LV_DRAW_BUF_STRIDE_ALIGN to 1 and driver can ignore all stride parameters.

`LV_DRAW_BUF_STRIDE_ALIGN` alone is not enough to determine the stride width. See vg_lite:

https://github.com/lvgl/lvgl/blob/master/src/draw/vg_lite/lv_vg_lite_utils.c#L477-L484

kisvegabor, 2024-01-03:
I think we are talking about the same thing :slightly_smiling_face: 

What I mean is keeping and using `lv_draw_buf_width_to_stride` (which uses `LV_DRAW_BUF_STRIDE_ALIGN` internally unless the user overwrites `width_to_stride_cb` in `lv_draw_buf_handlers_t`) to get the stride of the draw buffers, layers, font bitmaps, or on all places where we need to have a stride no related to any GPU specially.

GorgonMeducer, 2024-01-04:
Since I get what I wanted, I would close this issue.

closed 2024-01-04 by GorgonMeducer