lvgl/lvgl #5092: load png from filesystem

Maverik1408, 2023-12-24:
### Introduce the problem

Hi everyone, I'm trying to load a png from the filesystem of an esp32 but it keeps returning "lv_fs_open: Can't open file (F:wink.png): unknown driver letter"
this is my code:

static void *sd_fs_open(lv_fs_drv_t *drv, const char *path, lv_fs_mode_t mode)
{
  Serial.println("FILE OPEN");
  File *fp = (File *)lv_mem_alloc(sizeof(File));
  if (fp == NULL)
    return NULL;
  File mySDfile = *fp;

  // Make the path relative to the current directory (the projects root folder)
  char buf[256];
  sprintf(buf, "/%s", path);

  if (mode == LV_FS_MODE_WR)
  {
    // Open a file for write

    mySDfile = SPIFFS.open(buf, FILE_WRITE);
  }
  else if (mode == LV_FS_MODE_RD)
  {
    // Open a file for read
    mySDfile = SPIFFS.open(buf, FILE_READ);
  }
  else if (mode == (LV_FS_MODE_WR | LV_FS_MODE_RD))
  {
    // Open a file for read and write
    mySDfile = SPIFFS.open(buf, FILE_WRITE);
  }

  // make sure at the beginning
  mySDfile.seek(0);

  *fp = mySDfile;

  return fp;
}
static lv_fs_res_t sd_fs_close(lv_fs_drv_t *drv, void *file_p)
{
  Serial.println("FILE CLOSE");
  File *fp = (File *)file_p;

  fp->close();

  delete (fp); // when close
  return LV_FS_RES_OK;
}

static lv_fs_res_t sd_fs_read(lv_fs_drv_t *drv, void *file_p, void *fileBuf, uint32_t btr, uint32_t *br)
{
  lv_fs_res_t res = LV_FS_RES_NOT_IMP;

    return res;

}

static lv_fs_res_t sd_fs_seek(lv_fs_drv_t *drv, void *file_p, uint32_t pos, lv_fs_whence_t whence)
{

   Serial.println("FILE SEEK");
  lv_fs_res_t res = LV_FS_RES_OK;

  File *fp = (File *)file_p;
  fp->seek(pos);
  return res;
}

static lv_fs_res_t sd_fs_tell(lv_fs_drv_t *drv, void *file_p, uint32_t *pos_p)
{
  Serial.println("FILE TELL");
  File *fp = (File *)file_p;
  *pos_p = fp->position();

  return LV_FS_RES_OK;
}

void spiffs_drv_init()
{
  /* create the drive */
  lv_fs_drv_t sd_drv;
  lv_fs_drv_init(&sd_drv);
  sd_drv.letter = 'F';
  sd_drv.cache_size = 0;
  sd_drv.open_cb = sd_fs_open;
  sd_drv.close_cb = sd_fs_close;
  sd_drv.read_cb = sd_fs_read;
  sd_drv.seek_cb = sd_fs_seek;
  sd_drv.tell_cb = sd_fs_tell;
  sd_drv.write_cb = NULL;
  sd_drv.dir_close_cb = NULL;
  sd_drv.dir_open_cb = NULL;
  sd_drv.dir_read_cb = NULL;

  lv_fs_drv_register(&sd_drv);
  Serial.println("Registred Driver");
}

`spiffs_drv_init();`
`lv_img_set_src(ui_Image1, "F:/wink.png");`

### Examples and cases

### Suggested solution

kisvegabor, 2023-12-25:
You should add `static` here:
```c
static lv_fs_drv_t sd_drv;
```

Maverik1408, 2023-12-26:
thank you but now i have another error 
`"[Error](0.826, +826)    remove_free_block: Asserted at expression: prev && "prev_free field can not be null"  (in lv_tlsf.c line #590)"` 
you can help me with this error?

XuNeo, 2023-12-26:
`fp` is malloc'ed by lv_mem_alloc, but free'd by `delete`. The API should be in pair.
```
static void *sd_fs_open(lv_fs_drv_t *drv, const char *path, lv_fs_mode_t mode)
{
  Serial.println("FILE OPEN");
  File *fp = (File *)lv_mem_alloc(sizeof(File));
  if (fp == NULL)
    return NULL;
  /* ... */
}

static lv_fs_res_t sd_fs_close(lv_fs_drv_t *drv, void *file_p)
{

  delete (fp); // when close
  return LV_FS_RES_OK;
}
```

Maverik1408, 2023-12-26:
I changed the close function

```
static lv_fs_res_t sd_fs_close(lv_fs_drv_t *drv, void *file_p)
{
  File *fp = (File *)file_p;
  fp->close();
  //delete (fp); // when close
  lv_mem_free(fp);
  return LV_FS_RES_OK;
}
```
but the code not arrived at close function
```
FILE OPEN --> start function
FILE OPEN --> end function
FILE SEEK --> start function
FILE SEEK --> end function
FILE READ --> start function
FILE READ --> end function
[Error] (0.602, +602)    lv_tlsf_free: Asserted at expression: !block_is_free(block) && "block already marked as free"         (in lv_tlsf.c line #1167) 
```

XuNeo, 2023-12-26:
From provided code, I can not see the issue. The error reported is typically double-free issue.

Maverik1408, 2023-12-26:
How can I explain the problem better to you. I tried removing `lv_img_set_src(ui_Image1, "F:/wink.png");` and I don't get any errors

kisvegabor, 2023-12-30:
Please add a lot of `printf`s to see exactly where the issue happens.

Maverik1408, 2024-01-09:
this is my code but not working :(

this is error 
```
ESP32 Chip: ESP32-D0WDQ6
Arduino Version: 2.0.11
ESP-IDF Version: 4.4.5
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 4194304 bytes, speed 80 MHz
CPU speed 240 MHz
touch_calibrate
Calibration File Loaded
Registred Driver
FILE OPEN
FILE SEEK
[Error] (0.601, +601)    lv_fs_read_cached: Asserted at expression: file_p->cache->buffer != NULL (Out of memory)  (in lv_fs.c line #183)
```

this is code
```
#include "Arduino.h"
#include "logdebug.h"
#include "WiFi.h"
#include <SD.h>
#include <TFT_eSPI.h>
#include "SPI.h"
#include "lvgl.h"
#include "gui/ui.h"

TFT_eSPI tft = TFT_eSPI(); // Initializza TFT

/*Change to your screen resolution*/
static const uint16_t screenWidth = 320;
static const uint16_t screenHeight = 240;
static lv_disp_draw_buf_t draw_buf;
static lv_color_t buf[screenWidth * screenHeight / 10];

/*****************************************************************************************************************************************************
 *                                                              L V G L                                                                              *
 *****************************************************************************************************************************************************/
#if LV_USE_LOG != 0
void my_print(const char *buf)
{
  Serial.printf(buf);
  Serial.flush();
}
#endif
/* Display flushing */
void my_disp_flush(lv_disp_drv_t *disp_drv, const lv_area_t *area, lv_color_t *color_p)
{
  uint32_t w = (area->x2 - area->x1 + 1);
  uint32_t h = (area->y2 - area->y1 + 1);

  tft.startWrite();
  tft.setAddrWindow(area->x1, area->y1, w, h);
  tft.pushColors((uint16_t *)&color_p->full, w * h, true);
  tft.endWrite();

  lv_disp_flush_ready(disp_drv);
}
/*Read the touchpad*/
void my_touchpad_read(lv_indev_drv_t *indev_drv, lv_indev_data_t *data)
{
  uint16_t touchX, touchY;

  bool touched = tft.getTouch(&touchX, &touchY, 600);

  if (!touched)
  {
    data->state = LV_INDEV_STATE_REL;
[... 229 lines cut ...]
  disp_drv.flush_cb = my_disp_flush;
  disp_drv.draw_buf = &draw_buf;
  lv_disp_drv_register(&disp_drv);

  /*Initialize the (dummy) input device driver*/
  static lv_indev_drv_t indev_drv;
  lv_indev_drv_init(&indev_drv);
  indev_drv.type = LV_INDEV_TYPE_POINTER;
  indev_drv.read_cb = my_touchpad_read;
  lv_indev_drv_register(&indev_drv);

  LogDebug::DEBUG_PRINTF("Ui Init");
  ui_init();

  spiffs_drv_init();
  lv_img_set_src(ui_Image1, "F:/wink.png");

}

void loop()
{
  lv_timer_handler();
  delay(5);
}
```

kisvegabor, 2024-01-10:
> Please add a lot of printfs to see exactly where the issue happens.

Without adding some extra debug info I can't really tell what could be the problem :slightly_frowning_face:

Maverik1408, 2024-01-10:
now i setting 
`#define LV_MEM_CUSTOM 1` it was before `#define LV_MEM_CUSTOM 0`
after i change 
`lv_img_set_src(ui_Image1, "F:/pause.png");` to `lv_img_set_src(ui_Image1, "F:pause.png");`

now i don't have any error but not showing image 

this is log

```
Calibration File Loaded
FILE: wink.png
FILE: pause.png
FILE: touchCal
Registred Driver
/pause.png451
OPEN
OPEN
FILE SEEK
FILE READ
FILE CLOSE
/pause.png451
OPEN
OPEN
FILE SEEK
FILE READ
FILE CLOSE
```

this line `/pause.png451` is this `Serial.println(mySDfile.size());`

kisvegabor, 2024-01-11:
The string that you would pass to `SPIFFS.open` without LVGL shall be after `F:`.

Are you sure it works without LVGL is you call `SPIFFS.open` directly?

Maverik1408, 2024-01-11:
first code part:

```
  lv_fs_file_t f;
  lv_fs_res_t res;
  res = lv_fs_open(&f, "F:pause.png", LV_FS_MODE_RD); // system crash here
  if (res != LV_FS_RES_OK)
    Serial.println("don't work nothing open");
  if(res == LV_FS_RES_OK)
    Serial.println("file open");
  File *ftest = (File *)f.file_d;
  Serial.print("lv_fs_open File Size: ");
  Serial.println(ftest->size());
  Serial.print("lv_fs_open File Name: ");
  Serial.println(ftest->name());
```
the result on debug is

```
file open
lv_fs_open File Size: 451
lv_fs_open File Name: pause.png
```
second code part
```
File testFile = SPIFFS.open("/pause.png", "r");
  Serial.print("spiffs File Size: ");
  Serial.println(testFile.size());
  Serial.print("spiffs File Name: ");
  Serial.println(testFile.name());
```
the result on debug is
```
spiffs File Size: 451
spiffs File Name: pause.png
```

3th part:
`lv_img_set_src(ui_Image1, "F:/pause.png");`

the result on debug is

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d3ba3  PS      : 0x00060230  A0      : 0x800eba1a  A1      : 0x3ffc93d0  
A2      : 0x3ffcfc00  A3      : 0x3f400bd1  A4      : 0x00000002  A5      : 0x3ffc600c  
A6      : 0xb17b341a  A7      : 0xff000000  A8      : 0xbaad5678  A9      : 0x3ffc93b0  
A10     : 0x3ffcfc00  A11     : 0xffffffff  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x3ffbd888  A15     : 0x00000000  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xbaad567c  LBEG    : 0x40086289  LEND    : 0x400862ad  LCOUNT  : 0x800e3301  

Backtrace: 0x400d3ba0:0x3ffc93d0 0x400eba17:0x3ffc9560 0x400e3323:0x3ffc9590 0x400e2596:0x3ffc95e0 0x400f10c7:0x3ffc9600 0x400d3fdd:0x3ffc9640 0x4010194e:0x3ffc96e0

  #0  0x400d3ba0:0x3ffc93d0 in sd_fs_open(_lv_fs_drv_t*, char const*, unsigned char) at /Users/ChicchiDani/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/xtensa-esp32-elf/include/c++/8.4.0/ext/atomicity.h:53
      (inlined by) __atomic_add_dispatch at /Users/ChicchiDani/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/xtensa-esp32-elf/include/c++/8.4.0/ext/atomicity.h:96
      (inlined by) std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_add_ref_copy() at /Users/ChicchiDani/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/xtensa-esp32-elf/include/c++/8.4.0/bits/shared_ptr_base.h:139
      (inlined by) std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count(std::__shared_count<(__gnu_cxx::_Lock_policy)2> const&) at /Users/ChicchiDani/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/xtensa-esp32-elf/include/c++/8.4.0/bits/shared_ptr_base.h:735
      (inlined by) std::__shared_ptr<fs::FileImpl, (__gnu_cxx::_Lock_policy)2>::__shared_ptr(std::__shared_ptr<fs::FileImpl, (__gnu_cxx::_Lock_policy)2> const&) at /Users/ChicchiDani/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/xtensa-esp32-elf/include/c++/8.4.0/bits/shared_ptr_base.h:1165
      (inlined by) std::shared_ptr<fs::FileImpl>::shared_ptr(std::shared_ptr<fs::FileImpl> const&) at /Users/ChicchiDani/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/xtensa-esp32-elf/include/c++/8.4.0/bits/shared_ptr.h:129
      (inlined by) fs::File::File(fs::File const&) at /Users/ChicchiDani/.platformio/packages/framework-arduinoespressif32/libraries/FS/src/FS.h:47
      (inlined by) sd_fs_open at src/main.cpp:177
  #1  0x400eba17:0x3ffc9560 in lv_fs_open at .pio/libdeps/esp32doit-devkit-v1/lvgl/src/misc/lv_fs.c:85
  #2  0x400e3323:0x3ffc9590 in decoder_info at .pio/libdeps/esp32doit-devkit-v1/lvgl/src/extra/libs/png/lv_png.c:81
  #3  0x400e2596:0x3ffc95e0 in lv_img_decoder_get_info at .pio/libdeps/esp32doit-devkit-v1/lvgl/src/draw/lv_img_decoder.c:99
  #4  0x400f10c7:0x3ffc9600 in lv_img_set_src at .pio/libdeps/esp32doit-devkit-v1/lvgl/src/widgets/lv_img.c:107
  #5  0x400d3fdd:0x3ffc9640 in setup() at src/main.cpp:368
  #6  0x4010194e:0x3ffc96e0 in loopTask(void*) at /Users/ChicchiDani/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42

ELF file SHA256: d599fe8c33cb2083

```

where am I wrong

i have settings

```
  lv_init();
  lv_png_init();
```

in alternative i can use one of this, as settings?
```
#define LV_USE_FS_STDIO 0
#define LV_USE_FS_POSIX 0
#define LV_USE_FS_WIN32 0
#define LV_USE_FS_FATFS 0
```

kisvegabor, 2024-01-11:
I don't understand:
```c
 res = lv_fs_open(&f, "F:pause.png", LV_FS_MODE_RD); // system crash here
```
The comment says it crashes here, but the log seems ok.

PierreRambaud, 2024-01-25:
If it can help you, this is what I use to register a new driver reading a littlefs partition:
https://github.com/PierreRambaud/pokegotchi/blob/master/lib/lv_conf/lv_port_fs_littlefs.cpp

Maverik1408, 2024-01-25:
thank you i solved problem ! :)

PierreRambaud, 2024-01-26:
> @PierreRambaud if you have a LittleFS layer for LVGL could you send a PR? [We have it in v8](https://github.com/lvgl/lvgl/blob/release/v8.3/src/extra/libs/fsdrv/lv_fs_littlefs.c), but it's not ported to v9 yet.

I'll try to do it next week, but I can't promise you. I'm not as good as all of you in C :sweat_smile:

kisvegabor, 2024-01-29:
No worries, we are happy to help :slightly_smiling_face:

PierreRambaud, 2024-01-29:
Seems like I have a lot of things to do on my project to make it work with the v9, I'll let you know if I can make LittleFS working

kisvegabor, 2024-01-30:
Okay, thanks! Feel free to ask if any issues comes up during porting.

yunusemreucan, 2024-01-31:
Hi @Maverik1408  , how did you solve the problem i need the same thing, Can you share code thanks

PierreRambaud, 2024-01-31:
@kisvegabor It seems my LittleFS driver is working well, I'm able to read or write file, where do you want the PR? Is there a directory dedicated to this kind of driver in this repository, or should it be put into the lv_drivers one?

kisvegabor, 2024-02-01:
Cool! You can check out [this driver](https://github.com/lvgl/lvgl/blob/master/src/libs/fsdrv/lv_fs_stdio.c) as a reference.

Maverik1408, 2024-02-02:
Hi everyone, so will the driver be integrated with the new version?

PierreRambaud, 2024-02-18:
Three more drivers will coming for the next release, one for littlefs based on lfs.h, one based on Arduino littlefs. And the last one is about the Arduino SD.h :)

manhkamui0502, 2024-03-14:
> thank you i solved problem ! :)
can i have the code? I'm working on a project that displaying image from .png file store in esp32. My problem is picture not showing. Pls help @_@

manhkamui0502, 2024-03-14:
@Maverik1408 
can i  have your project?

Maverik1408, 2024-03-17:
i find this [https://github.com/100askTeam/esp-arduino-learn/blob/master/examples/08_integrated/01_lcd_sd_card_fc_joypad_fs_lv_lib_100ask/01_lcd_sd_card_fc_joypad_fs_lv_lib_100ask.ino#L47](url) and start

PierreRambaud, 2024-04-04:
LFS and Arduino littlefs drivers have been merged. Arduino SD driver is in review FYI.

kisvegabor, 2024-06-20:
As it should be resolved by @PierreRambaud's built-in driver I'm closing this issue.

closed 2024-06-20 by kisvegabor

nikunj8780, 2024-10-23:
`simple how to save image (png) file in our esp LittleFS storage and how to using in Arduino IDE to display this image in any display like ILI9488 ; LIL9441 any display but give atlist one for Arduino IDE with LVGL 9.2.0 and TFT_eSPI 4.5.43`