lvgl/lvgl #5096: Establish API retirement policy

kisvegabor, 2023-12-25:
### Introduce the problem
Now we don't have any standards to deprecate and retire old APIs. It causes trouble mostly (but not only) on major version changes.

### Examples and cases

- major version change
- typo fixes
- API changes in minor versions

### Suggested solution

We need to establish standards like:
- Support old APIs with `define`, `static inline` function, `typedef` etc
- Add optional warning logs when old API is used. 
- Allow enabling/disabling old API
- Keep the old API for 1 major release or at least one year
- Announce the API changes in release notes, and newsletter
- Add CI test for API to see if all the required functions are still available
- Add more `*_private.h` files to better control what is considered public API.
- Establish rules for internal or less critical APIs (E.g. image decoder or cache API)

What do you think?

cc @XuNeo @FASTSHIFT

bjsylvia, 2023-12-26:
Some ideas:
- Deprecated APIs should also be marked `deprecated ` in the API documentation.
- If there are new APIs or implementations available for the same functionality as the deprecated API, the documentation should describe the officially recommended replacement reference codes.
- For new API, It is recommended to also indicate the version in which this API is supported.

XuNeo, 2023-12-29:
APIs start with `_` should be used internally only. 
We can use mark APIs not ready in this way too. `_` can mean this API is subject to change in any time.

kisvegabor, 2023-12-30:
@bjsylvia 
> For new API, It is recommended to also indicate the version in which this API is supported.

Do you mean in the docs? If so, we have `release/vX.Y` branches so it's if the users look into the docs of thier version they see which functions are available. 

In function comments and docs I would rather add when we will remove the deprecated functions. 

@XuNeo 
> APIs start with _ should be used internally only.

I think establishing the practice of `*_private.h` files would be cleaner. With this we could say what is availabel from `lvgl.h` is considered public API.

XuNeo, 2024-01-02:
I thinks it's a good time to explain the usage of `_` now. 
It seems the APIs are only for lvgl internally use, but some APIs should also user visible.

E.g.
```c
bool _lv_area_intersect(lv_area_t * res_p, const lv_area_t * a1_p, const lv_area_t * a2_p);
void _lv_area_join(lv_area_t * a_res_p, const lv_area_t * a1_p, const lv_area_t * a2_p);
bool _lv_ll_is_empty(lv_ll_t * ll_p);
```

kisvegabor, 2024-01-03:
Originally we have introduced `_` to lower the public API size for the MicroPythin binding, where each public function increases flash usage. 

However I think it's not best way to do it because 
- `_` function causes confusion
- doesn't reduce the binding size too much
- it's a constant decision making on function level

Instead, by introducing `*_private.h` we could exclude even the struct `lv_obj_t` from the binding, and keep only the API functions.  @kdschlosser, what do you think about it?

So in the end, gradually, I would convert the `_` functions back to normal and use much more private headers.

kdschlosser, 2024-01-03:
If you can push internal functions into their own private header files that would be ideal. If there are structures or unions that you consider to be private but get used in public functions whether they are used as a return value or as a parameter these structures and unions are not really private. Having them defined in private header files can cause issues with creating a binding (CPython) where the size of the structure needs to be known in order to create a correct representation of the structure/union. 

I understand the want to have that kind of stuff tucked away but when creating a binding that uses CFFI it becomes impossible to do so unless those private header files get included when generating the binding. 

Now if there is a size issue as with the MicroPython Binding there are a lot of ways the binding can be reduced in size that doesn't include hiding structures in private header files. There is a massive amount of repeated code in the MicroPython binding that can be reduced thus decreasing the size of the binding. a fantastic example are the "print" functions. there are 231 of them that I am sure can be reduced in some way. 

The functions are generated in this manner
```c
STATIC mp_obj_t mp_lv_obj_get_style_border_side(size_t mp_n_args, const mp_obj_t *mp_args, void *lv_func_ptr)
{
    const struct _lv_obj_t *obj = (const struct _lv_obj_t *)mp_to_lv(mp_args[0]);
    uint32_t part = (uint32_t)mp_obj_get_int(mp_args[1]);
    lv_border_side_t _res = ((lv_border_side_t (*)(const struct _lv_obj_t *, uint32_t))lv_func_ptr)(obj, part);
    return mp_obj_new_int_from_uint(_res);
}
```

There are a total of 130 functions that are wrapped in this manner. there are 178 functions that are like this.

```c
STATIC mp_obj_t mp_lv_obj_get_style_flex_flow(size_t mp_n_args, const mp_obj_t *mp_args, void *lv_func_ptr)
{
    const struct _lv_obj_t *obj = (const struct _lv_obj_t *)mp_to_lv(mp_args[0]);
    uint32_t part = (uint32_t)mp_obj_get_int(mp_args[1]);
    lv_flex_flow_t _res = ((lv_flex_flow_t (*)(const struct _lv_obj_t *, uint32_t))lv_func_ptr)(obj, part);
    return mp_obj_new_int(_res);
}
```

I am sure there is a way to reduce the amount of this kind of repetition and that will reduce the size of the firmware. The amount of size reduction that would be had from making a structure private is going to be very minimal compared to eliminating the code repetition. 

I don't know C code well enough to be able to craft a way to do this. if there is someone that would like to look over the attached C source file and come up with ways to eliminate the repetitive code I can implement it into the binding generation. 

[lv_mp.zip](https://github.com/lvgl/lvgl/files/13819480/lv_mp.zip)

If I turn off everything in LVGL that has the ability to be turned off except for a single font when I compile the binding LVGL and the binding code add 500K to the firmware size. I am pretty sure that is a bit on the high side. If I turn on all of the typical stuff used it takes up 1mb of flash size.

kdschlosser, 2024-01-03:
IDK how C treats things prefixed with a double underscore. in Python those are the truly private things. single underscore prefixed items are only considered to be for internal use not private. There is a big difference between the 2.

Take this as an example.

```python
class Test:

    def _internal_function1(self):
        print('this is an internal function')

    def _internal_function2(self):
        print('internal function calling a private function')
        self.__private_function()

    def __private_function(self):
        print('this is a private function')

test = Test()

test._internal_function1()
test._internal_function2()
test.__private_function()
```

and the output is this.

```python
this is an internal function
internal function calling a private function
this is a private function
Traceback (most recent call last):
  File "\scratches\scratch_891.py", line 18, in <module>
    test.__private_function()
AttributeError: 'Test' object has no attribute '__private_function'. Did you mean: '_Test__private_function'?
```

the `__private_function` is not accessible publicly. Well... I should say not easily accessible.

kisvegabor, 2024-01-03:
> Having them defined in private header files can cause issues with creating a binding (CPython) where the size of the structure needs to be known in order to create a correct representation of the structure/union.

If a structure is private its fields don't need to be represented in the binding. E.g.
```c
//lv_display_private.h
typedef struct _lv_display_t {
  ...a lot of fields...
};

//lv_display.h
struct _lv_display_t;
typedef struct _lv_display_t lv_display_t;

//main.c
#include "lvgl.h" //includes lv_display.h, but NOT lv_display_private.h

lv_display_t d1; //Doesn't work as C doesn't know the size of lv_display_t
lv_display_t * d1_p  //Works, as it's a just a pointer which always has the same size
d1_p->hor_res = 100; //Doesn't work as the fields are no visible 
```

So basically all you can do is using `lv_display_t *`, which is the same as `void *` but type checked in compile time.

kdschlosser, 2024-01-03:
I am going to try a couple of things to see if I am able to convince python ctypes to handle the pointer as a simple integer (memory address) and then create a ctypes pointer object from that address when passing it back in a function call. IDK if it will do that properly or not.

kdschlosser, 2024-01-03:
It's not that the fields need to be defined in a binding. It's in my use case it is to know the upper boundry of the pointer.

Think of it this way. When you create an array in C and that array gets passed to a function call there are 2 things that need to be known in order to access that array properly. You need to know the start memory address and you also need to know the size of the array right? Otherwise you have no way of knowing what the lower and upper bounds are. It's the same thing when passing a single structure to a function call. What the binding needs to know is what the starting point is and also what the stopping point is. Without knowing the size of what is being passed there is no way for it to know when to stop.

The size is determined by looking at the types that are stored in the structure. That is how the size gets determined. There needs to be some kind of representation of the object in the binding. It doesn't matter if the fields are defined or not It could be a simple catchall a single field that has a uint8_t array that has the same size as the undefined object. Without being able to see the structure at all by reading the header files I am not able to calculate what the size of the structure is.

Maybe I could add functions the the shared library that would return the size of those objects at runtime. then i could build a representation of the structure or union when the binding first gets imported. That's about the only way I can think of possibly doing it.

kisvegabor, 2024-01-03:
Why do you need to access the structure fields? It's just a pointer to who-knows-what. Anything you do with the struct should be done via API functions. E.g. `lv_display_set_user_data(display, p)`.

kisvegabor, 2024-01-14:
I think the next steps here are to 
- Create some more `private` header files to better see what is public and what isn't
- Better define what `_` means (IMO we should get rid of it.)
- Add CI test for API to see if all the required functions are still available
- Document what our actual API retirement policy is 
   - Support old APIs with define, static inline function, typedef etc
   - Add optional warning logs when old API is used.
   - Allow enabling/disabling old API
   - Keep the old API for 1 major release or at least one year
  - Announce the API changes in release notes, and newsletter
  - Deprecated APIs should also be marked deprecated  in the API documentation.
  - If there are new APIs or implementations available for the same functionality as the deprecated API, the documentation should describe the officially recommended replacement reference codes.

XuNeo, 2024-01-29:
Not stale. It's ongoing.

kisvegabor, 2024-02-19:
Still not fully established

kdschlosser, 2024-03-14:
Still not fully established

kisvegabor, 2024-04-03:
Let's continue in #6011.