lvgl/lvgl #5097: Establish merging policy

kisvegabor, 2023-12-25:
### Introduce the problem

As LVGL grows I cannot be the single decision maker (even if all major things as discussed in issues) because all the decisions and changes might affect important technologies and businesses. Therefore the decisions and changes need to be well grounded and supported by multiple core people. 

### Examples and cases

- Merging directly to `master` is problematic
  - follow up commits like "fix warning" or "code formatting" pollutes the commit history
  - there is no way to comment on the change  
- PRs should be reviewed by multiple people before merging to increase the chance of spotting issues

### Suggested solution

- Make the [`master` branch protected](https://docs.github.com/en/repositories/configuring-branches-and-merges-in-your-repository/managing-protected-branches/managing-a-branch-protection-rule), so only PRs can be merged and not direct commit is allowed 
- Allow merging PRs only with `X` number of reviews. `X` should be at least 2: one will be usually me, and the other(s) should be other members of the core LVGL team. It's important to set this number according to the actual activity of the core team, so I wonder from the current people with write access to LVGL repository who is interested in regular code reviews: @embeddedt, @FASTSHIFT, @pete-pjb, @turoksama, @xiaoxiang781216. 

If there is anything you would add, please comment. 🙂

C47D, 2023-12-25:
Hi @kisvegabor , my 2 cents:

> follow up commits like "fix warning" or "code formatting" pollutes the commit history

What about blocking the merge button if the code formatting or compilation CI fails?

> PRs should be reviewed by multiple people before merging to increase the chance of spotting issues

We use this at work, when we are in tight schedules we have time issues because:
- Not all reviewers have to to review a PR
- They're in different time zones

PGNetHun, 2023-12-26:
Hello @kisvegabor ,
1. I also support reviewing PRs by more devs: allow merging after at least 2 reviewer approved: 1 approver should be in the group "maintainer"; so PRs can be reviewed by external people also, and at least 1 maintainer (who has the main knowledge) -> we have this policy also in our company.
2. Merging is allowed only when CI builds finish successfully, and at least 2 approves (1 from a maintainer)
3. No direct commit to `main` branch.
4. I propose squash merging, to avoid polluting git history.
5. PR description: we should review that also, because I see a lot of merged PRs without filling check-bullets, and some PRs without tests added, although it could be tested (for example bugfixes, or refactors).

XuNeo, 2023-12-29:
I propose a clear git history and avoid squashing unrelated changes.  Do squash commits like `code formatting`, `fix CI problem`.

kisvegabor, 2023-12-30:
Thank you for the comments! To kick it off I set up these rules for the `master` branch:
[image]
[image]

We can adjust these after seeing how it works in the practice.

kisvegabor, 2024-01-01:
I lowered the required approval count to 1, because I can't approve my own PR and there are no 2 other people with write access who reviews PR on a daily basis. 

Unfortunately reviews from people with no write access don't add to the count. :slightly_frowning_face:

kisvegabor, 2024-01-14:
2 weeks were passes and I think now we can look back to evaluate how the new rules were working. 

My feelings are positive. The required approvals lowered the follow-up commints in the `master` branch and increased the quality of the merged PRs as well. I'm really happy that people without write access also joined the review process and left valueable feedback. Special thanks for @XuNeo, @PGNetHun, and @C47D in this regard.

Probably due to this higher activity and the write access of @FASTSHIFT  I also felt like the development were speeded up.  So in summary I consider it as a good practice that we should keep. 

The only issue I see that is some larger features became fragile and hard track (at least for me). I belive it will be resolved when we start formalizing the developing using the process described [here](https://docs.lvgl.io/master/CONTRIBUTING.html#feature-development-workflow). I've received the feedback from @XuNeo that this process sounds complicated. To mitigate it I'll update the docs to summarize it shorter in a less pragmatic way. After the release of v9 I'll create issues for some planned features to see how i works in the practice. Of course, as usual we can tailor it as needed.

What do you think? How did is feel for you? What can we improve?

PGNetHun, 2024-01-14:
Hello!
Thank you for the fast PR reviews, that is also **important** to have fluid work on a project.

Just one suggestion regarding PR reviews: 
when there are feedbacks (and feedback status is "Request changes"), and those are addressed by PR author (new commit is added), then **PR author should re-request** reviews to notify ALL reviewers that new changes are made, and PR should be **reviewed again**. Without it reviewers don't know whether author is still working on the fixes, and needs more time, or has already finished, and is waiting for feedbacks.
Please add this step also in the document.
https://github.blog/changelog/2019-02-21-re-request-review-on-a-pull-request/

C47D, 2024-01-14:
Hi,

Always happy to help in any way.

Offtopic:
Will we have an issue containing the new features for the next major release? Maybe we could do one and pin it.

kisvegabor, 2024-01-15:
@PGNetHun 
Good idea, see #5332 

@C47D 
Once we can get started with shaping-pitching it should naturally show what are planned for **minor** releases. To be honest still I have no idea for v10 :smile:

kisvegabor, 2024-01-30:
As the Chinese New Year is approaching we can either
1. temporarily remove the required PR approval
2. add new members with write access
3. do not merge PRs for 2 weeks

I think 2) would be the best to increase the diversity of LVGL. 
 @PGNetHun @C47D It would be amazing if you could jump in even for this short time. I understand if you say no and I don't want to push it too far. If you have any concerns please let me know. I'm sure that we can  figure out a solution for them.

PGNetHun, 2024-01-30:
Hello @kisvegabor ,
I’m happy to help, even if I can only review PRs now, so yes, please grant me write access to the repo, and increase required PR approves to 2.
Thank you!

kisvegabor, 2024-01-30:
Amazing! Thank you! I've just sent the invitation.

kisvegabor, 2024-01-30:
Let's keep the approval count at 1 for the time of the Chinese holiday. At that time both @FASTSHIFT and @XuNeo will be away.

After that, we can increase it to 2.

C47D, 2024-01-31:
Can we have a list of topics and who is familiar with it? Example, we got a error related to SDL, who can take a look at it? Who is familiar so they can review a change related to it? etc

kisvegabor, 2024-01-31:
GitHub's [codeowner](https://docs.github.com/en/repositories/managing-your-repositorys-settings-and-features/customizing-your-repository/about-code-owners) feature can be helpful here. This is how [ESP-IDF applies it](https://github.com/espressif/esp-idf/blob/master/.gitlab/CODEOWNERS).

C47D, 2024-02-01:
I didn't knew that feature, it looks really useful

kisvegabor, 2024-02-19:
@XuNeo 
Happy new year! 

Let  me know if the holiday time is over and you are back to normal work. When it happens (as we have discused) I'll increase the approval count to 2.

XuNeo, 2024-02-20:
Hi Gabor,
We are back to work. Please do increase the count.

XuNeo, 2024-02-20:
> I propose a clear git history and avoid squashing unrelated changes. Do squash commits like `code formatting`, `fix CI problem`.

Hi Guys,

I want to mention this policy again. Contributor should squash the small fixes to one commit and do separate unrelated changes to different commits. 
For changes not related to current topic, separate them to another PR.

kisvegabor, 2024-02-20:
The required approval count is increased to 2.

kisvegabor, 2024-03-12:
I close this issue. 2 approval counts seems working in the practice.

closed 2024-03-12 by kisvegabor