lvgl/lvgl #5098: Establish clean release policy

kisvegabor, 2023-12-25:
## Introduce the Problem 
This will be a long one 🙂

As LVGL gains more users, having a transparent method to plan, select, develop, and release features becomes increasingly important.

## Suggested Solution
Issues to solve:

- Medium and large-scale features should be discussed and planned first.
- Features should have clear goals and responsibilities.
- Have frequent and predictable release cycles.
- Have a clear timeframe for features.
- Implement a clear feature stop for testing before releases.
- Handle stale feature requests.
- Allow all community members to share their opinions/ideas during the planning of a feature or to open new feature requests.
- Enable all community members to easily contribute to the development of a feature.

After conducting some research and getting inspired by [Shape Up](https://www.productplan.com/glossary/shape-up-method/) (mostly, please read the linked summary first) and [Scrum](https://www.productplan.com/glossary/scrum-agile-framework/) (a little), I devised a method that seems effective and flexible for us. These methodologies weren’t designed for open-source projects, but with some tailoring, we can adapt them. It might sound complex at first, but I truly believe we need something like this to be transparent and have a solid foundation for the development of LVGL.

**So, this is what I have in mind:**

### Terminology
(These are taken from Shape Up)

- **Shaping**: Convert raw ideas into something that can be developed. It includes examining opportunities, finding similar feature requests, roughly planning how it shall be implemented, tested, and taught through docs and examples.
- **Pitch**: A document that is the result of shaping.

### Schedule
Work in 8-week cycles, each cycle starting at the beginning of odd months (e.g., a cycle could be from March to April).

- **Weeks 1-2: Cool Down** (A relaxed period to step back, plan, fix bugs, and make a release):
    - Close the previous development phase: Feature stop, bug fixes, release, discuss anything that can be improved.
    - Prepare for the next development phase: Ideas, shaping, pitch creation.
- **Weeks 3-8: Develop** any planned pitches.

### Key Components
Feature Development Issue

- For each raw idea, there is a dedicated issue where it's being "shaped".
- Anyone can comment with ideas to help the shaping process.
- Use GitHub Labels to show the state of the issue:
    - `shaping`: Still discussing how to approach it.
    - `pitched`: We have come to a conclusion and created a pitch.
    - `request-change(<user-name>)`: Each core member has a dedicated label, and when they request a change, they add their label.
    - `under development`: It's being developed by someone (can be developed only if no change is requested by the end of the cool-down phase).
    - `ready`: The pitch is developed and merged into master.
    - `stale`: Inactive, can be warmed up.
    - `rejected`: Not interested in this feature for some reason.
Pitch Files

- There is a `pitches` branch in the LVGL repo (unrelated to the `master` branch) that contains `<pitch-name>.md` files.
- A pitch file contains the following:
    - Link to the Issue.
    - Estimated time: Assuming someone works on it 2 hours per day (a pitch should fit into a 6-week development phase).
    - Problem to solve: Describe the problem in ~1 paragraph, ideally with real-life cases.
    - Success criteria: Something measurable, e.g., make something 20% faster/smaller/shorter/etc.
    - Solution outline: A rough description of how to implement it.
    - Rabbit holes: Possible issues and difficulties that we already see.
    - Considerations: Other approaches that we have considered but dropped (just for the record).
    - Testing: How will we test this feature?
    - Teaching: How will we teach this feature/how will the user know about it (docs, examples, etc)?

### Workflow
1. Someone opens a new issue with an idea.
2. Others jump in to comment on it. Core members play an important role in really understanding the problem and possible solutions, and request changes if needed. This is the shaping phase.
3. It might take any length of time to shape an idea, but finally, it will be either converted to a pitch (created by a core member or a community member) or will become stale and closed.
4. Pitches are typically created during the cool-down phase. Once we have a pitch, the development of the pitch can be undertaken by anyone (original author of the idea, other community member, core member). If someone applies for the development of the pitch, he/she will be marked as Responsible in the issue.
5. Create a new branch for the developed pitch and continuously merge there via PRs.
6. A pitch is ready if tests, docs, and examples are added.
7. Finally, a PR is created against the master branch, which needs to be approved by the core members. This PR should have a template like this:
    - Pitch file and issue.
    - Success criteria checkboxes.
    - Tests added checkboxes.
    - Teaching checkboxes.
    - Coding standard link (API, formatting, etc).
8. Anything merged by the end of the Development phase will be part of the release. Anything not ready can be continued; however, if it takes too long, it should be reviewed and shaped again.
9. If a pitch is not taken by anyone in the next cool-down phase, it will be closed as stale.

---

So that's it. I'm really curious what you think. It really might sound complex, however once everything will be in place (templates, issue labels, docs, first pitches), I believe it will go smoothly.

If you agree in general we can start working on this bases from January. Probably the first cycle in January-February will be a transitional period after which we can refine our workflow gradually. 

However if you have something very different in mind or see fundamental flaws in the above method, don't hesitate to share it.

kisvegabor, 2023-12-30:
It might not the most interesting topic, but it's very important as it declares our development and release policies for the future. 

So could you please comment on it. An "ok for me" would be perfect too, so that I can see that you read it and there is no objection. If it will be really okay for you too, I'll prepare it at the very beginning of January. 
 
@XuNeo @bjsylvia @FASTSHIFT @C47D @PGNetHun

C47D, 2023-12-30:
For planning we can use the [Project](https://docs.github.com/en/issues/organizing-your-work-with-project-boards/managing-project-boards/about-project-boards) tab,

PGNetHun, 2023-12-30:
Hello!
It's a good idea to shape the development life cycle.

Some comments:
1. Due to the Github branching strategy, the `main` branch should contain ALWAYS stable version (thx for PR CI builds)
2. All new feature/pitch branches should contain tests, added/updated documentations, and examples. I think it is better to fully finish a new feature (code, tests, documentation, examples), than partially merging unusable code. So community can use it immediately as soon it is added to `main` branch. But if a feature is urgent, and if it's possible, then a **lean** (AND working) version can be merged (including: code, tests, docs, examples), and later it can be extended. But anything goes to `main` branch, should work. I'm not a fan having a "feature flag" to turn off **merged**, but half-done features till the full feature is finished. Either it is done and can be used, or not. If not, then continue implementing it on the feature/pitch branch.
3. Keep focus on having stable LVGL code on `main` branch
4. My suggestion for releases: use Git tags to mark a `main` branch commit as release. If a hotfix is needed, then create a `release/vX.XX-hotfix-YYY` branch from that commit, so it will not contain breaking changes. But I think hotfixes are rare in LVGL :)
5. No automatic version bump -> keep manual semantic versioning (major.minor.patch)
6. MicroPython binding: same release strategy: tag `main` branch commit as release -> align MicroPython binding release to LVGL release. The MicroPython version is mentioned anyway in it's version file, but it is good to know with which LVGL version is the firmware created. Same with other bindings.
7. MicroPython binding: we could create (with github automation) some release firmwares for most used ports and boards, so community devs do not need to pull git repo, install requirements, and build firmware, instead they just download it, deploy to device, and can start developing their MicroPython project. For example these 2 boards are common: ESP32-GENERIC and ESP32-GENERIC-SPI. There could be a public voting to see which ports & boards community wants.
8. It would be good to add "LVGL version or commit hash" to Git issue template to know which version is affected by the issue (v8 or v9 or reporter uses an old version, and issue is already fixed?)
9. This is also important (for me at least): keep LVGL a **GUI library**. Meaning: it should contain **ONLY** GUI related code, libraries, and nothing else. For example no file system drivers (FAT, LittleFS, etc), non-UI drivers (USB, Bluetooth, etc) or scanning libraries (QR code scanning) or camera drivers, or anything else that are not GUI related. Please firmly reject those feature requests, and close the github issue (no stale). Keep LVGL simple, GUI-specific, and not a bloated library (handling everything).

+1: PR code review: it would be good to test (manually) the PR change BEFORE merging, to see that it really works as expected. (I don't trust unit tests, 'cause those are created also by developers, and some bugs appear only when tested manually)

kisvegabor, 2023-12-30:
Thanks for chiming in! :blush: 

@C47D 
I've just check the new "Projects" (there is classic too) and it's really amazing. I was thinking about using only Labels first, but we can explore how to use Projects as well.

@PGNetHun  
I mostly agree. To phrase it differently, we should try to select smaller targets that we fully finish (with tests, examples, docs). It should be part of the shaping process to set the goals accordingly. (We already had some great but never ending updates which were never merged :slightly_frowning_face: )

> My suggestion for releases: use Git tags to mark a main branch commit as release. If a hotfix/patch is needed, then create a release/vX.YY.ZZZ branch from that commit, so it will not contain breaking changes. But I think hotfixes are rare in LVGL :)

We already have a branches for all minor versions where we can add bugfix versions, independently of the other versions. E.g. we have `v8.3.11` in the `release/v8.3` branch. This way we can support all minor versions independently of the others. 

> MicroPython binding: [...]

I agree with both, but let's treat these as a separate topics.

> It would be good to add "LVGL version or commit hash" to Git issue template to know which version is affected by the issue (v8 or v9 or reporter uses an old version, and issue is already fixed?)

I've just sent [a PR](https://github.com/lvgl/lvgl/pull/5140) to update the issue template. 

> This is also important (for me at least): keep LVGL a GUI library.

What about interfaces to 3rd party libs such as LitteFS, FATFS, libpng, etc?

> PR code review: it would be good to test (manually) the PR change BEFORE merging, to see that it really works as expected. (I don't trust unit tests, 'cause those are created also by developers, and some bugs pop up only when feature is tested manually)

I strongly agree.

kisvegabor, 2024-01-03:
Based on the above discussion I created this PR #5170

closed 2024-01-03 by kisvegabor

kisvegabor, 2024-01-16:
I've created [3 issues](https://github.com/lvgl/lvgl/labels/shaping) for to get started with the new development process. For medium and large scale issues and features let's start following the same practice to better oversee the goals of each feature.