lvgl/lvgl #5104: lv_obj_get_width and lv_obj_get_height  always return 0

liuaoalan, 2023-12-26:
version ： lvgl 8.2.0
environment ： codeblocks   

sample code like ：

   lv_obj_t * obj = lv_obj_create(lv_scr_act());
   lv_obj_set_pos(obj, 500, 500);
    //lv_obj_set_width(obj, 300);
    //lv_obj_set_height(obj, 500);
    lv_obj_set_size(obj,300, 300);

    printf("w:%d\n", lv_obj_get_width(obj));
    printf("h:%d\n", lv_obj_get_height(obj));
}

after run this sample  ，print this ：

[User]  (10067.062, +10067062)   WinMain: LVGL initialization completed!        (in main.c line #56)
w:0
h:0

XuNeo, 2023-12-26:
Only after `lv_obj_update_layout`, you can get the actual `width` and `height`.
You may want to try `lv_obj_get_style_width` which is the width you set.

liuaoalan, 2023-12-27:
> Only after `lv_obj_update_layout`, you can get the actual `width` and `height`. You may want to try `lv_obj_get_style_width` which is the width you set.

Thanks for help

closed 2023-12-27 by liuaoalan