lvgl/lvgl #5105: lv_example_get_started_1.c.o: in function `lv_scr_act': undefined reference to `lv_disp_get_default'

raphael10-collab, 2023-12-26:
### What MCU/Processor/Board and compiler are you using?
x86_64
OS: Ubuntu 22.10
LVGL: 8.3

###  What do you want to achieve?
A working plain simple starter example

`lv_example_get_started_1.h` : 

```
#ifdef __cplusplus
extern "C" {
#endif

void lv_example_get_started_1(void);

#ifdef __cplusplus
}
#endif
```

`lv_example_get_started_1.c` : 
```
#include "../libs/lvgl/lvgl.h"
#include "../libs/lvgl/src/core/lv_disp.h"
void lv_example_get_started_1(void)
{
    lv_obj_t * btn = lv_btn_create(lv_scr_act());     /*Add a button the current screen*/
}
```

`basic.cc` : 
```
#include "iostream"
#include "c_file.h"
#include "lv_example_get_started_1.h"
int main(int argc, char* argv[]) {
    int di = doubleFunct(2);
    std::cout << "di: " << di << std::endl;

    lv_example_get_started_1();
    return 0;
}
```

In `CMakeLists.txt` : 
```
cmake_minimum_required(VERSION 3.24)
project (webviewPrj LANGUAGES C CXX)

add_executable(basic
    src/basic.cc
    src/c_file.c
    src/lv_example_get_started_1.c
)
```

I get this error message:
```
cmake --build cmakebuilddir
[ 12%] Building C object CMakeFiles/basic.dir/src/lv_example_get_started_1.c.o
[ 25%] Linking CXX executable basic
/usr/bin/ld: CMakeFiles/basic.dir/src/lv_example_get_started_1.c.o: in function `lv_scr_act':
lv_example_get_started_1.c:(.text+0x5): undefined reference to `lv_disp_get_default'
/usr/bin/ld: lv_example_get_started_1.c:(.text+0xd): undefined reference to `lv_disp_get_scr_act'
/usr/bin/ld: CMakeFiles/basic.dir/src/lv_example_get_started_1.c.o: in function `lv_example_get_started_1':
lv_example_get_started_1.c:(.text+0x28): undefined reference to `lv_btn_create'
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/basic.dir/build.make:146: basic] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:87: CMakeFiles/basic.dir/all] Error 2
gmake: *** [Makefile:91: all] Error 2
```

How to make it work?

PS:
Due to this message in LVGL Forum (I previously created other topics which I deleted from the Forum)

    An error occurred: We appreciate your enthusiasm! That said, for the safety of our community, you’ve reached the maximum number of topics a new user can create on their first day. Please wait 16 hours and you’ll be able to create more new topics.

I had to ask this question here

kisvegabor, 2023-12-30:
Note that LVGL is just a library. You need to provide the main.c file, drive initialization, etc.
See this project to get started: https://github.com/lvgl/lv_port_pc_eclipse

raphael10-collab, 2023-12-30:
Thank you @kisvegabor

closed 2023-12-30 by raphael10-collab