lvgl/lvgl #5110: about label txt and png btn effect

docterling, 2023-12-27:
i have an app use lvgl,but there are different effects in two lvgl versoin
1.ver 8.3.11
[image]
conf:
[lv_conf-v8.3.11.txt](https://github.com/lvgl/lvgl/files/13778882/lv_conf-v8.3.11.txt)

2.ver 8.1.0-dev
[image]
conf:
[lv_conf-v8.1.0-dev.txt](https://github.com/lvgl/lvgl/files/13778886/lv_conf-v8.1.0-dev.txt)

3.btn png
[image]

as the pics show,there's some artifact behide the label in ver 8.3.11 and it's fine in ver 8.1.0-dev,and the btn is abnormal too,the app about the label and btn codes are the same just different so lib,what led to such a result.

kisvegabor, 2023-12-30:
Hi,

- How did you created the label?
- Can you reproduce it in a simulator? E.g. [here](https://github.com/lvgl/lv_port_pc_eclipse)

docterling, 2023-12-31:
Hi
1.create the label like this:
```
static lv_color_t bg_color ;
static lv_color_t text_color ;
static lv_img_dsc_t *index_bg = load_png_from_file();
static lv_img_dsc_t *icon_record_r = load_png_from_file();
static lv_img_dsc_t *icon_record_p = load_png_from_file();
static lv_obj_t *record_obj = NULL;
static lv_obj_t *main_view_obj = NULL;
static lv_obj_t *main_view_bg_obj = NULL;

bg_color.full = 0x00000000;
text_color.full = 0xffffffff;

main_view_bg_obj = lv_obj_create(lv_scr_act());
lv_obj_set_size(main_view_bg_obj, LV_HOR_RES , LV_VER_RES);
lv_obj_align(main_view_bg_obj, LV_ALIGN_BOTTOM_RIGHT, 0, 0);
lv_obj_set_style_bg_color(main_view_bg_obj, bg_color, 0);
lv_obj_set_style_border_color(main_view_bg_obj, bg_color, 0);
lv_obj_set_style_radius(main_view_bg_obj, 0, 0);

main_view_obj = lv_img_create(lv_scr_act());
lv_img_set_src(main_view_obj, index_bg);

date_obj = lv_label_create(main_view_obj);
lv_obj_set_pos(date_obj, x, y);
lv_obj_set_style_text_font(date_obj, ttf_info_24.font, 0);
lv_obj_set_style_text_color(date_obj, text_color, 0);

time_obj = lv_label_create(main_view_obj);
lv_obj_set_pos(time_obj, x1, y1);
lv_obj_set_style_text_font(time_obj, ttf_info_24.font, 0);
lv_obj_set_style_text_color(time_obj, text_color, 0);

record_obj = lv_imgbtn_create(main_view_obj);
lv_imgbtn_set_src(obj, LV_IMGBTN_STATE_RELEASED, NULL, icon_record_r, NULL);
lv_imgbtn_set_src(obj, LV_IMGBTN_STATE_PRESSED, NULL, icon_record_p, NULL);
lv_obj_set_pos(obj, xx,yy);
lv_obj_set_size(obj, 96 , 96);
```
index_bg.png is a 1280x720 totally transparent png,icon_record_r is the btn png file upthair.

2.i havn't tried that yet

kisvegabor, 2024-01-01:
It seems like LVGL thinks the background of the screen is transparent or at least it has issues with mixing colors. Does it make any difference if you enable `LV_COLOR_SCREEN_TRANSP` in `lv_conf.h`?

docterling, 2024-01-02:
as you can see in the lv_conf.h attachment of the ask,LV_COLOR_SCREEN_TRANSP is already set to 1
if i set the LV_COLOR_SCREEN_TRANSP to 0 and save lvgl display buffer,the red dots become black,and the there are some black artifacts behind the date time text just like the ask attachment pic show.
[image]
when i blend with another buffer,the black things are show
[image]
and no matter how i memset the lvgl display buffer to 0x00 or 0xff,no difference

kisvegabor, 2024-01-03:
To exclude a few possible issues please try to reproduce it in [the simulator](https://github.com/lvgl/lv_port_pc_eclipse/tree/release/v8.3). Be sure to use the `relaese/v8.3` branch. If you see the issue there as well, I can help to find and fix the problem.

docterling, 2024-01-06:
Hi,I've reproduced the problem in lv_sim_visual_studio simulator in release/v8.3 branch
as the code show above if set the 2 lines
```
lv_obj_set_style_bg_color(main_view_bg_obj, bg_color, 0);
lv_obj_set_style_border_color(main_view_bg_obj, bg_color, 0);
```
the result pic is:
[image]

and if comment out the 2 lines,the result pic is:
[image]

in my hardware board there are 2 planes,video plane in the bottom,ui plane on top of it,so i must set the 2 lines or else the ui plane will  cover the video plane,but that's cause the problem!

kisvegabor, 2024-01-07:
Can you share the simulator project (upload to GitHub?) so that I can see all the settings.

docterling, 2024-01-07:
i use Microsoft Visual Studio Enterprise 2019,version 16.11.17
down and rename the file to
[image]
[lv_sim_visual_studio_2.7z.003.zip](https://github.com/lvgl/lvgl/files/13853526/lv_sim_visual_studio_2.7z.003.zip)
[lv_sim_visual_studio_2.7z.004.zip](https://github.com/lvgl/lvgl/files/13853527/lv_sim_visual_studio_2.7z.004.zip)
[lv_sim_visual_studio_2.7z.005.zip](https://github.com/lvgl/lvgl/files/13853529/lv_sim_visual_studio_2.7z.005.zip)
[lv_sim_visual_studio_2.7z.006.zip](https://github.com/lvgl/lvgl/files/13853531/lv_sim_visual_studio_2.7z.006.zip)
[lv_sim_visual_studio_2.7z.002.zip](https://github.com/lvgl/lvgl/files/13853524/lv_sim_visual_studio_2.7z.002.zip)
[lv_sim_visual_studio_2.7z.001.zip](https://github.com/lvgl/lvgl/files/13853523/lv_sim_visual_studio_2.7z.001.zip)

kisvegabor, 2024-01-08:
unzip-ing showed an error :slightly_frowning_face: 

As I'm on Linux I can't test in VisualStudio, but if you send only your
- lv_conf.h
- lv_drv_conf.h
- main.c (with driver init and demo UI)

it should be enough for me to reproduce in Eclipse as well.

docterling, 2024-01-09:
in the project,i set the hor_res 1280 ver_res 720
the display dev things are all about windows
in win32drv.c file,fun lv_win32_display_driver_flush_callback it's disp_flush in linux
i use the click cb to triger the line  and date time drawing and in the flush cb to save the raw rgb file
[win32drv.zip](https://github.com/lvgl/lvgl/files/13873446/win32drv.zip)

kisvegabor, 2024-01-11:
Ah it just came to my mind that I saw this issue specific to the windows driver in LVGL v8.3. 

@MouriNaruto, do you know why the windows driver can't handle transparent screens?

@docterling I don't know why, but I can't extract this ZIP either. It shows "An error occurred".  Uploading the full working project to GitHub would be really helpful to easily test it.

docterling, 2024-01-11:
i had met the problem in linux,and reproduce in windows simulator,so maybe it doesn't specific to windows
i will upload the project to github later

kisvegabor, 2024-01-11:
If you upload a Linux project, I can test it too.

MouriNaruto, 2024-01-11:
@kisvegabor 

> I don't know why, but I can't extract this ZIP either. It shows "An error occurred". Uploading the full working project to GitHub would be really helpful to easily test it.

You need to remove the “.zip” suffix which is for bypassing GitHub uploading. (It's a 7-Zip multi-archive.)

Kenji Mouri

docterling, 2024-01-11:
@MouriNaruto 
you should uncomment the two lines in main in  file LVGL.Simulator.cpp and try again,be care the text_color
```
lv_obj_set_style_bg_color(main_view_bg_obj, bg_color, 0);
lv_obj_set_style_border_color(main_view_bg_obj, bg_color, 0);
```

MouriNaruto, 2024-01-11:
@docterling 

After I uncomment these two lines. The UI will black. I think it may caused by many APIs from Windows GDI is not support transparent. I will try to find some workaround for that.

Kenji Mouri

MouriNaruto, 2024-01-11:
But I think LVGL should provide a proceed image for HAL. So, I will wait for other news.

I also see something like https://github.com/lvgl/lvgl/issues/5110#issuecomment-1872657790 when I using AlphaBlend instead of  StretchBlt for supporting the transparent.

Also, the Windows backend basically only put the LVGL framebuffer to the screen. So in https://github.com/lvgl/lvgl/issues/5110#issuecomment-1872657790, @docterling save the framebuffer which only touched by LVGL.

Kenji Mouri

docterling, 2024-01-12:
@kisvegabor
you can try this demo,it depends on libpng,libpng depend on zlib
just comment and uncomment line num 275~276 in file LVGL.Simulator.cpp
and use [YUView](https://github.com/IENT/YUView) view the saved lvgl disp raw buf,try invert check box on the right side of the app to see different color
[lvgl_demo.zip](https://github.com/lvgl/lvgl/files/13915788/lvgl_demo.zip)

closed 2024-02-05 by lvgl-bot