lvgl/lvgl #5119: warning: bitwise operation between different enumeration types ... is deprecated

peteDDD, 2023-12-28:
### Perform all steps below and tick them with [x]
- [x ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x ] Update lvgl to the latest version
- [x ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

### To Reproduce

##Error generated:
```
.pio/libdeps/cyd/lvgl/src/core/lv_obj_style.h:94:48: warning: bitwise operation between different enumeration types '<unnamed enum>' and '<unnamed enum>' is deprecated [-Wdeprecated-enum-enum-conversion]
   94 |     lv_obj_remove_style(obj, NULL, LV_PART_ANY | LV_STATE_ANY);
```

### Expected behavior

### Screenshots or video

XuNeo, 2023-12-29:
Could you prepare a fix about the warning? It could be resolved by converting them to unsigned before using `bitwise`.

peteDDD, 2023-12-29:
@XuNeo are you asking me to fix it?  There are dozens of commands that use this structure.

kisvegabor, 2023-12-30:
I think it's not reasonable to fix it by casting as it would pollute the API and the examples.  

We could move `lv_part_t` and `lv_state_t` into the same `enum`? E.g.
```c
/**
 * Possible states of a widget.
 * OR-ed values are possible
 */
typedef enum lv_style_selector_t {
    LV_STATE_DEFAULT     =  0x0000,
    LV_STATE_CHECKED     =  0x0001,
    LV_STATE_FOCUSED     =  0x0002,
    LV_STATE_FOCUS_KEY   =  0x0004,
    LV_STATE_EDITED      =  0x0008,
    LV_STATE_HOVERED     =  0x0010,
    LV_STATE_PRESSED     =  0x0020,
    LV_STATE_SCROLLED    =  0x0040,
    LV_STATE_DISABLED    =  0x0080,
    LV_STATE_USER_1      =  0x1000,
    LV_STATE_USER_2      =  0x2000,
    LV_STATE_USER_3      =  0x4000,
    LV_STATE_USER_4      =  0x8000,

    LV_STATE_ANY = 0xFFFF,    /**< Special value can be used in some functions to target all states*/
    

    LV_PART_MAIN         = 0x000000,   /**< A background like rectangle*/
    LV_PART_SCROLLBAR    = 0x010000,   /**< The scrollbar(s)*/
    LV_PART_INDICATOR    = 0x020000,   /**< Indicator, e.g. for slider, bar, switch, or the tick box of the checkbox*/
    LV_PART_KNOB         = 0x030000,   /**< Like handle to grab to adjust the value*/
    LV_PART_SELECTED     = 0x040000,   /**< Indicate the currently selected option or section*/
    LV_PART_ITEMS        = 0x050000,   /**< Used if the widget has multiple similar elements (e.g. table cells)*/
    LV_PART_CURSOR       = 0x060000,   /**< Mark a specific place e.g. for text area's cursor or on a chart*/
    
    LV_PART_CUSTOM_FIRST = 0x080000,    /**< Extension point for custom widgets*/

    LV_PART_ANY          = 0x0F0000,    /**< Special value can be used in some functions to target all parts*/
};

/*Just the better indicate in API what the expected parameter is*/
typedef int32_t lv_part_t;
typedef int32_t lv_state_t;
```

However it would break the MicroPython examples and codes where it's used like `lv.STATE.PRESSED` and `lv_PART.KNOB`. (The grouping is generated automatically from the type names) 

So first, we need to understand how big problem it is. If I recall correctly this is the first time it was reported, so I assume it's not a problem affecting a lot of users.
- Which MCU/target do you use?
- Have you enabled some special compiler flags?
- How many warnings do you see from this kind?

peteDDD, 2023-12-30:
@kisvegabor 

"So first, we need to understand how big problem it is. If I recall correctly this is the first time it was reported, so I assume it's not a problem affecting a lot of users."

The way I view this is that the warning of deprecated code is giving us time to address the issue before a future release turns this into an error or completely ignores the code.  So, it is not a show-stopper issue today but may well become one in the future, as this warning gives us fair notice.

"Which MCU/target do you use?"  ESP32
"Have you enabled some special compiler flags?" No.
"How many warnings do you see from this kind?" dozens of times for each app as this construct is used extensively in LVGL and by the code generated by SLS.

Thank you.

XuNeo, 2024-01-02:
The warning is default ON with g++20. https://gcc.gnu.org/onlinedocs/gcc/C_002b_002b-Dialect-Options.html#C_002b_002b-Dialect-Options
```
-Wno-deprecated-enum-enum-conversion (C++ and Objective-C++ only)
Disable the warning about the case when the usual arithmetic conversions are applied on operands where one is of enumeration type and the other is of a different enumeration type. This conversion was deprecated in C++20. For example:

enum E1 { e };
enum E2 { f };
int k = f - e;
-Wdeprecated-enum-enum-conversion is enabled by default with -std=c++20. In pre-C++20 dialects, this warning can be enabled by -Wenum-conversion.

```

LVGL has use so many of it. Maybe you can try to disable the warning with `-Wno-deprecated-enum-enum-conversion`.

kisvegabor, 2024-01-03:
If it's a C++ warning I wonder why it is reported from the C files of LVGL.

derekmulcahy, 2024-01-12:
I began to see this warning when I included a component which was written in C++.

kisvegabor, 2024-01-14:
Which component?

derekmulcahy, 2024-01-14:
[Connection Helper](https://components.espressif.com/components/johboh/connectionhelper), it implements OTA updates using Arduino OTA push style instead of pulling from a web server.

kisvegabor, 2024-01-15:
I can't see adding an independent module made LVGL to show warnings.

derekmulcahy, 2024-01-15:
The module is written in C++. In order to use it I renamed my main.c to main.cpp (with some tweaks to use external “C” functions). I guess the ESP IDF build then runs everything through CPP to allow correct handling of C/C++ linkage.

kisvegabor, 2024-01-17:
Hmm, C files still should be compiled with `gcc`. Anyway if it was problem for you it could a problem for other too, however right now I cannot see a simple non breaking solution. :slightly_frowning_face:

hunterramp, 2024-01-18:
We are also experiencing this issue in our ESP IDF project which uses LVGL. The following line is an example of .cpp code that is causing the warning:

`lv_obj_remove_style(my_obj, NULL, LV_PART_MAIN | LV_STATE_DEFAULT);`

Ideally, we would not disable the -Wdeprecated-enum-enum-conversion. Compiler warnings are there for a reason and help developers find potential issues to ensure the reliability and robustness of their code.

kisvegabor, 2024-01-19:
Does it affect LVGL itself asa well or only the user code?

hunterramp, 2024-01-19:
> Does it affect LVGL itself as well or only the user code?

@kisvegabor, it applies to user code and LVGL itself. Here's an example of code in LVGL that causes the warning:
https://github.com/lvgl/lvgl/blob/f470539f4c732ac9e5984e30448727a8336a8229/src/core/lv_obj_style.c#L255

kisvegabor, 2024-01-24:
Thank you, however I'm still puzzled. It should be valid in C and no C++ warnings should be generated from C files.

closed 2024-02-18 by lvgl-bot

kschwan, 2024-03-11:
Hi,
Just chiming in...

> Thank you, however I'm still puzzled. It should be valid in C and no C++ warnings should be generated from C files.

When including lvgl headers from C++ files (compiled with g++) that's when the warnings appear. For example here: https://github.com/lvgl/lvgl/blob/9ec3b8706b098f60940cee6ac211d3cb3141aedf/src/core/lv_obj_style.h#L94

This particular place looks to have been changed since 8.3.9. The same kind of thing appears in our application code also however.

Best,
Kim

kisvegabor, 2024-03-12:
Hey,

So it's something that we can fix in our header files. If so, could send the full build log to see all the warnings?

bryghtlabs-richard, 2024-08-02:
Here's the log for one of our c++ file including lv_obj_style.h, LVGL version 8.3.11, g++ 12.2.0:

```
[1557/1574] Building CXX object esp-idf/ui_event_handlers/CMakeFiles/__idf_ui_event_handlers.dir/ui_events_battery_component.cpp.obj
In file included from c:\projects\redacted\esp32\managed_components\lvgl__lvgl\src/core/lv_obj.h:140,
                 from c:\projects\redacted\esp32\managed_components\lvgl__lvgl\lvgl.h:35,
                 from C:/Projects/redacted/ESP32/components/ui_event_handlers/ui_events_battery_component.cpp:3:
c:\projects\redacted\esp32\managed_components\lvgl__lvgl\src/core/lv_obj_style.h: In function 'void lv_obj_remove_style_all(_lv_obj_t*)':
c:\projects\redacted\esp32\managed_components\lvgl__lvgl\src/core/lv_obj_style.h:94:48: warning: bitwise operation between different enumeration types '<unnamed enum>' and '<unnamed enum>' is deprecated [-Wdeprecated-enum-enum-conversion]
   94 |     lv_obj_remove_style(obj, NULL, LV_PART_ANY | LV_STATE_ANY);
      | 
```

It's a little odd though - isn't this valid 'C' code? `LV_PART_ANY`, `LV_STATE_ANY`, and `lv_obj_remove_style_all` are all within `extern "C"` blocks, so I was expecting G++ to treat them as 'C' code.

bryghtlabs-richard, 2024-08-02:
extern "C" only changes the linkage, not the language in the block, so any inline code in headers should be valid in any language we want to support including LVGL.

For this specific case(it's the most common one for me by far), I think we could move lv_obj_remove_style_all() out of the header and into the C file.

kisvegabor, 2024-08-13:
I caused warnings on other platforms too. As it's not a performance-critical function, we can move it to a C file without any issues. Do you have time to send a PR? If you send it, I can approve it, and we will be closer to the required 2 approval count :slightly_smiling_face:

bryghtlabs-richard, 2024-08-13:
It looks like https://github.com/lvgl/lvgl/pull/3709 (fix by moving to 'C') and https://github.com/lvgl/lvgl/commit/39687e808f (fix by casting) and supported 8.x branches just need a new release built to include the fix. From git + [support table](https://docs.lvgl.io/master/intro/index.html#version-support):

|Version|Finding|
|--------|--------|
|v8.2-|lacking fix, but also out of support|
|v8.3 |fixed with casting in 39687e808f, need new release version|
|v8.4 |fixed with casting in 39687e808f, need new release version|
|v9.0 |fixed by https://github.com/lvgl/lvgl/pull/3709|
|v9.1 |fixed by https://github.com/lvgl/lvgl/pull/3709|
|master|fixed by https://github.com/lvgl/lvgl/pull/3709|

bryghtlabs-richard, 2024-08-13:
Building a new release does fix the most common instance of this warning in my project, but this morning I converted a C file to C++, and had to add casting to `lv_style_selector_t` many times in the C++ code for every call to `lv_obj_set_style_bg_img_src()`. So, a change to the interfaces so that C++ could call these without casting would be welcome. Moving lv_part_t and lv_state_t into the same enum would work.

kisvegabor, 2024-08-17:
We will release v9.2 very soon. Unless there it's very important for any reason I'd fix in only in v9.2 and 8.4 to make hove too many too simple releases. 

> Moving lv_part_t and lv_state_t into the same enum would work.

It's a much larger change. Let's aim for the low hanging fruit first. :slightly_smiling_face:

Gabryxx, 2024-08-27:
I dont know if this can help, i use Arduino Gui for compile sketch
but also if i split the original string
lv_obj_set_style_pad_bottom(ui_InfoBox, 10, LV_PART_MAIN | LV_STATE_DEFAULT);
in 2 string
lv_obj_set_style_pad_bottom(ui_InfoBox, 10, LV_PART_MAIN );
lv_obj_set_style_pad_bottom(ui_InfoBox, 10, LV_STATE_DEFAULT);
i get same warning
warning: bitwise operation between different enumeration types '_lv_part_t' and '_lv_state_t' is deprecated

cleishm, 2025-12-16:
> Thank you; however, I'm still puzzled. It should be valid in C, and no C++ warnings should be generated from C files.

I think it's better to assume that it's just a matter of time until C compilers also warn on this. A bitwise operation between mixed types is indicative of a mistake in almost every case and thus a very reasonable thing to warn the user about. The only reason it's not a mistake when using the LVGL API is because of some very careful assignment of values across the different enum types.

So the real problem is that the LVGL API is representing these flags using different types, and then expecting the end-user to bitwise OR them. I think the proposal by @kisvegabor, to move `lv_part_t` and `lv_state_t` into the same enum, is the only reasonable solution.

Can this issue be re-opened and considered more broadly?

Note: I ran into this issue when using LVGL in an ESP-IDF project (with default configuration).

AndreCostaaa, 2025-12-17:
@cleishm which version of LVGL are you using?

AndreCostaaa, 2025-12-17:
Can you share the compilation error please?

Mwoua, 2026-11-03:
I know the issue is old, but I just got the error in C++ code.

Compilation error:
error: bitwise operation between different enumeration types 'lv_part_t' and 'lv_state_t' [-Werror=deprecated-enum-enum-conversion]

And the relevant code:
lv_obj_set_style_bg_opa( mTileView, LV_OPA_TRANSP, LV_PART_MAIN | LV_STATE_DEFAULT );

LVGL 9.5.0 with riscv32-esp-elf-g++ compiler (that use -std=gnu++26 by default)

AndreCostaaa, 2026-11-04:
Hi, you need to cast it to keep the c++ compiler happy:

```c
lv_obj_set_style_bg_opa(mTileView, LV_OPA_TRANSP, static_cast<lv_style_selector_t>(LV_PART_MAIN) | LV_STATE_DEFAULT);
```