lvgl/lvgl #5127: how use the LV_DRAW_BUF_DEFINE to load the image?

xinsuinizhuan, 2023-12-29:
[image]

i also inlucde the lv_draw_buf.h file,but  some compile error

xinsuinizhuan, 2023-12-29:
I also add the .c files, but still could not link:
[image]

XuNeo, 2023-12-29:
It looks like you haven't include the file to build. It complains no symbol can be found.

xinsuinizhuan, 2024-01-02:
> It looks like you haven't include the file to build. It complains no symbol can be found.

how to include the file to build? I also inlcude the .c file in my project.

xinsuinizhuan, 2024-01-02:
[image]

XuNeo, 2024-01-02:
I'm not familiar with vscode. But the error suggests the symbol is not linked in. You can try to add faulty code to img_hand.c and compile to see if it reports error.

xinsuinizhuan, 2024-01-03:
> I'm not familiar with vscode. But the error suggests the symbol is not linked in. You can try to add faulty code to img_hand.c and compile to see if it reports error.

I  add the faulty img_hand.c file to my project, if have some head file to add, or set some define to 1?

xinsuinizhuan, 2024-01-08:
I resolve the problem， i create the preject is c++, and the main file is .cpp file, the .cpp file to load the imge .c buffer file, load false, it need to add extern "C".

kisvegabor, 2024-01-22:
As the issue is resolved, I close it.

closed 2024-01-22 by kisvegabor