lvgl/lvgl #5137: Link to simulator unreachable

daixtrose, 2023-12-30:
### Perform all steps below and tick them with [x]

- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] ~~Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)~~

### Describe the bug

In the [README.md](https://github.com/lvgl/lvgl/blob/master/README.md#arrow_forward-get-started) in the `Getting Started` section the subsection `Start to Use LVGL` the link in 

```ghfm
    Set up a [Simulator](https://docs.lvgl.io/master/get-started/platforms/pc-simulator.html) (10 minutes)
```

leads to a 404 message. 

### To Reproduce

Visit https://docs.lvgl.io/master/get-started/platforms/pc-simulator.html, which yields a 404.

### Expected behavior

A webpage to read stuff.

### Screenshots or video

[image]

daixtrose, 2023-12-30:
AFAICS via digging through the docs, the link should be https://docs.lvgl.io/master/integration/ide/pc-simulator.html#simulator

C47D, 2023-12-30:
Hi, I've submitted a fix in #5139

kisvegabor, 2023-12-30:
Thanks! Let's continue there!

PR #5139 (2024-01-02): docs(README): fix link to the simulator
### Description of the feature or fix

Fixes #5137 

### Checkpoints
- [ ] Run `code-format.py` from the scripts folder. [astyle](http://astyle.sourceforge.net/install.html) needs to be installed.
- [ ] Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed
- [ ] Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant. 
- [ ] Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- [ ] If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://git [... cut]
closed 2023-12-30 by kisvegabor