lvgl/lvgl #5145: `set_ext_click_area()` no effect outside button's parent container

hyansuper, 2024-01-01:
```python

import lvgl as lv

import display_driver

cont = lv.obj(lv.scr_act())
cont.set_size(lv.SIZE.CONTENT,lv.SIZE.CONTENT)
cont.center()

btn = lv.btn(cont)
btn.center()

btn.set_ext_click_area(100)

cont.set_ext_click_area(100) # no effect
cont.add_flag(lv.obj.FLAG.CHECKABLE) # no effect
```
[image]

Button is clickable in the white area, but not in the grey area
How can I extend buttons clickable area even outside its container?
I think this is desired behaviour for extending clickable area

kisvegabor, 2024-01-01:
Hi,
Which LVGL version do you use?

hyansuper, 2024-01-02:
> Hi, Which LVGL version do you use?

online simulator v8.3 and v9 https://sim.lvgl.io
code for v9:
```python
import lvgl as lv

import display_driver

scr = lv.obj()

cont = lv.obj(scr)
cont.set_size(lv.SIZE_CONTENT, lv.SIZE_CONTENT)
cont.center()

btn = lv.button(cont)
btn.center()

btn.set_ext_click_area(100)

cont.set_ext_click_area(100) # no effect
cont.add_flag(lv.obj.FLAG.CLICKABLE) # no effect

lv.screen_load(scr)
```

kisvegabor, 2024-01-03:
There is not simple way to make the extra click area affect the parent too. However with this trick you can do it in v9:
```py
import display_driver

def event_cb(e):
    e.set_ext_draw_size(100)
    
scr = lv.obj()

cont = lv.obj(scr)
cont.set_size(lv.SIZE_CONTENT, lv.SIZE_CONTENT)
cont.center()

btn = lv.button(cont)
btn.center()

btn.set_ext_click_area(100)

cont.set_ext_click_area(100) # no effect
cont.add_flag(lv.obj.FLAG.CLICKABLE) # no effect
cont.add_event(event_cb, lv.EVENT.REFR_EXT_DRAW_SIZE, None)
cont.add_flag(lv.obj.FLAG.OVERFLOW_VISIBLE)
cont.refresh_ext_draw_size()

lv.screen_load(scr)
```

Note that this way the button won't be clipped to the boundaries of the parent.

hyansuper, 2024-01-03:
solved ❤