lvgl/lvgl #5153: v9 doxygen errors

kdschlosser, 2024-01-02:
Here is the list of things that doxygen is complaining about when compiling the documentation. These things needs to be fixed before moving onto the next release stage of LVGL v9. 

I wanted to have them as a reference so they can be checked off as they get fixed.

I am also going to generate all of the things that are missing docstrings all together.

```
WARNING: C:/Users/drsch/PycharmProjects/lvgl/src/draw/lv_draw_buf.h:143: warning: The following parameter of lv_draw_buf_clear(void *buf, uint32_t w, uint32_t h, lv_color_format_t color_format, const lv_area_t *a) is not documented:
  parameter 'buf'

WARNING: C:/Users/drsch/PycharmProjects/lvgl/src/draw/lv_draw_buf.h:102: warning: The following parameter of lv_draw_buf_malloc(size_t size_bytes, lv_color_format_t color_format) is not documented:
  parameter 'size_bytes'

WARNING: C:/Users/drsch/PycharmProjects/lvgl/src/draw/lv_draw_triangle.h:46: warning: The following parameter of lv_draw_triangle(struct _lv_layer_t *layer, const lv_draw_triangle_dsc_t *draw_dsc) is not documented:
  parameter 'draw_dsc'

WARNING: C:/Users/drsch/PycharmProjects/lvgl/src/draw/lv_draw_triangle.h:40: warning: The following parameter of lv_draw_triangle_dsc_init(lv_draw_triangle_dsc_t *draw_dsc) is not documented:
  parameter 'draw_dsc'

WARNING: C:/Users/drsch/PycharmProjects/lvgl/src/draw/sw/lv_draw_sw.h:94: warning: The following parameter of lv_draw_sw_image(lv_draw_unit_t *draw_unit, const lv_draw_image_dsc_t *draw_dsc, const lv_area_t *coords) is not documented:
  parameter 'draw_dsc'

WARNING: C:/Users/drsch/PycharmProjects/lvgl/src/draw/sw/lv_draw_sw.h:126: warning: The following parameter of lv_draw_sw_layer(lv_draw_unit_t *draw_unit, const lv_draw_image_dsc_t *draw_dsc, const lv_area_t *coords) is not documented:
  parameter 'draw_dsc'

WARNING: C:/Users/drsch/PycharmProjects/lvgl/src/draw/sw/lv_draw_sw.h:175: warning: The following parameter of lv_draw_sw_rgb565_swap(void *buf, uint32_t buf_size_px) is not documented:
  parameter 'buf'

WARNING: C:/Users/drsch/PycharmProjects/lvgl/src/draw/sw/lv_draw_sw.h:149: warning: The following parameter of lv_draw_sw_transform(lv_draw_unit_t *draw_unit, const lv_area_t *dest_area, const void *src_buf, int32_t src_w, int32_t src_h, int32_t src_stride, const lv_draw_image_dsc_t *draw_dsc, const lv_draw_image_sup_t *sup, lv_color_format_t cf, void *dest_buf) is not documented:
  parameter 'draw_dsc'

WARNING: C:/Users/drsch/PycharmProjects/lvgl/src/draw/sw/lv_draw_sw_gradient.h:42: warning: The following parameters of lv_gradient_color_calculate(const lv_grad_dsc_t *dsc, int32_t range, int32_t frac, lv_grad_color_t *color_out, lv_opa_t *opa_out) are not documented:
  parameter 'color_out'
  parameter 'opa_out'

WARNING: C:/Users/drsch/PycharmProjects/lvgl/src/draw/sw/lv_draw_sw_mask.h:240: warning: The following parameter of lv_draw_sw_mask_line_angle_init(lv_draw_sw_mask_line_param_t *param, int32_t p1x, int32_t py, int16_t angle, lv_draw_sw_mask_line_side_t side) is not documented:
  parameter 'p1x'

WARNING: C:/Users/drsch/PycharmProjects/lvgl/src/font/lv_binfont_loader.h:30: warning: The following parameter of lv_binfont_load(lv_font_t *font, const char *font_name) is not documented:
  parameter 'font_name'

WARNING: C:/Users/drsch/PycharmProjects/lvgl/src/font/lv_font.h:112: warning: The following parameters of lv_font_get_glyph_bitmap(const lv_font_t *font, lv_font_glyph_dsc_t *g_dsc, uint32_t letter, uint8_t *buf_out) are not documented:
  parameter 'font'
  parameter 'buf_out'

WARNING: C:/Users/drsch/PycharmProjects/lvgl/src/font/lv_font_fmt_txt.h:218: warning: The following parameter of lv_font_get_bitmap_fmt_txt(const lv_font_t *font, lv_font_glyph_dsc_t *g_dsc, uint32_t unicode_letter, uint8_t *bitmap_out) is not documented:
  parameter 'g_dsc'

WARNING: C:/Users/drsch/PycharmProjects/lvgl/src/libs/bin_decoder/lv_bin_decoder.h:48: warning: The following parameter of lv_bin_decoder_open(lv_image_decoder_t *decoder, lv_image_decoder_dsc_t *dsc, const lv_image_decoder_args_t *args) is not documented:
  parameter 'args'

WARNING: C:/Users/drsch/PycharmProjects/lvgl/src/misc/lv_anim.h:391: warning: The following parameters of lv_anim_set_bezier3_param(lv_anim_t *a, int16_t x1, int16_t y1, int16_t x2, int16_t y2) are not documented:
  parameter 'x2'
  parameter 'y2'

WARNING: C:/Users/drsch/PycharmProjects/lvgl/src/misc/lv_anim.h:322: warning: The following parameter of lv_anim_set_playback_duration(lv_anim_t *a, uint32_t duration) is not documented:
  parameter 'duration'

WARNING: C:/Users/drsch/PycharmProjects/lvgl/src/misc/lv_color.h:201: warning: The following parameter of lv_color_format_get_bpp(lv_color_format_t cf) is not documented:
  parameter 'cf'

WARNING: C:/Users/drsch/PycharmProjects/lvgl/src/misc/lv_color.h:208: warning: The following parameter of lv_color_format_get_size(lv_color_format_t cf) is not documented:
  parameter 'cf'

WARNING: C:/Users/drsch/PycharmProjects/lvgl/src/misc/lv_color_op.h:85: warning: The following parameter of lv_color_brightness(lv_color_t c) is not documented:
  parameter 'c'

WARNING: C:/Users/drsch/PycharmProjects/lvgl/src/others/observer/lv_observer.h:311: warning: The following parameter of lv_obj_bind_state_if_eq(lv_obj_t *obj, lv_subject_t *subject, lv_state_t state, int32_t ref_value) is not documented:
  parameter 'state'

[... 123 lines cut ...]
WARNING: C:/Users/drsch/PycharmProjects/lvgl/src/widgets/canvas/lv_canvas.h:55: warning: The following parameter of lv_canvas_set_buffer(lv_obj_t *obj, void *buf, int32_t w, int32_t h, lv_color_format_t cf) is not documented:
  parameter 'obj'

WARNING: C:/Users/drsch/PycharmProjects/lvgl/src/widgets/canvas/lv_canvas.h:152: warning: The following parameter of lv_canvas_fill_bg(lv_obj_t *obj, lv_color_t color, lv_opa_t opa) is not documented:
  parameter 'obj'

WARNING: C:/Users/drsch/PycharmProjects/lvgl/src/widgets/msgbox/lv_msgbox.h:81: warning: The following parameter of lv_msgbox_add_text(lv_obj_t *obj, const char *text) is not documented:
  parameter 'text'

WARNING: C:/Users/drsch/PycharmProjects/lvgl/src/widgets/msgbox/lv_msgbox.h:132: warning: The following parameter of lv_msgbox_close(lv_obj_t *mbox) is not documented:
  parameter 'mbox'

WARNING: C:/Users/drsch/PycharmProjects/lvgl/src/widgets/msgbox/lv_msgbox.h:138: warning: The following parameter of lv_msgbox_close_async(lv_obj_t *mbox) is not documented:
  parameter 'mbox'

WARNING: C:/Users/drsch/PycharmProjects/lvgl/src/widgets/scale/lv_scale.h:224: warning: The following parameter of lv_scale_section_set_range(lv_scale_section_t *section, int32_t minor_range, int32_t major_range) is not documented:
  parameter 'section'

WARNING: C:/Users/drsch/PycharmProjects/lvgl/src/widgets/scale/lv_scale.h:232: warning: The following parameter of lv_scale_section_set_style(lv_scale_section_t *section, uint32_t part, lv_style_t *section_part_style) is not documented:
  parameter 'section'

WARNING: C:/Users/drsch/PycharmProjects/lvgl/src/widgets/tabview/lv_tabview.h:79: warning: The following parameter of lv_tabview_set_tab_bar_size(lv_obj_t *obj, int32_t size) is not documented:
  parameter 'obj'

```

kdschlosser, 2024-01-03:
ok so you can ignore the above error list because it is only going to show you things that are missing from the items that have had some portion of a docstring provided.

Attached is a file with the definitive list of all things that are not documented. The list is too large to fit being pasted so I am attaching it as a file.

there are 3594 undocumented public things that are not documented. This includes macros, global variables, typedefs, enums, enum members, functions, function arguments, structures, structure fields, unions and union fields. I think that is all of them. 

[output.txt](https://github.com/lvgl/lvgl/files/13816127/output.txt)

C47D, 2024-01-03:
I've fixed the `scale` ones in #5141

kisvegabor, 2024-01-03:
Thanks @kdschlosser! It's great that now we see it together.  Fortunately it's better than it seems. E.g.
-  `_LV_STR_SYMBOL_*`s are reported 11 times from each header from where it's included. (682 reports instead of 62)
- Not comment is required in `lv_api_map.h` (174 reports)
- We should care about the external libs. (~1000 reports) 
- In `lv_style_gen.h` and `lv_obj_style_gen.h` we can auto generate the comments. (~500 reports)

So it should be around 1000 real issues.  

I have 2 quesitions:
1. How can we set up a CI to spot missed docs in the future
2. Can we use `//! @cond Doxygen_Suppress` to exclude parts such as `lv_api_map.h`?

kdschlosser, 2024-01-03:
Ohhh.... This nifty piece of code is going to spit out things that are missed when you generate the documentation. Doxygen catches things like incorrect arguments for functions but that is pretty much the limit of what it pucks up on. so if a function is not documented at all it is not going to complain. I coded into the documentation build system the ability to produce these errors and also the ability to have the docstrings pulled and added into that JSON file. Still working on the latter tho. 

I am going to see if I can eliminate the duplicated reporting. Gotta figure out the way to tell where the original definition is at in the doxygen xml output.

closed 2024-01-25 by lvgl-bot