lvgl/lvgl #5166: lv_fs_write  process cache error

kevinkang-Globe, 2024-01-03:
### LVGL version

v9.0.0 dev

### What happened?

if the write range is overlap with the cache range, need to invalidate the data in the cache.

kisvegabor, 2024-01-03:
Hi, 

Nice catch! Could you send a Pull request with the fix?

kevinkang-Globe, 2024-01-03:
Hi,

I would try to figure out a fix for your review.

kisvegabor, 2024-01-03:
Amazing, thank you!

kevinkang-Globe, 2024-01-03:
How about the below solution ? 

```c
lv_fs_res_t lv_fs_write(lv_fs_file_t * file_p, const void * buf, uint32_t btw, uint32_t * bw)
{
    if(bw != NULL) *bw = 0;

    if(file_p->drv == NULL) {
        return LV_FS_RES_INV_PARAM;
    }

    if(file_p->drv->write_cb == NULL) {
        return LV_FS_RES_NOT_IMP;
    }

    lv_fs_res_t res = LV_FS_RES_OK;

    /*Need to do FS seek before writing data to FS*/
    if(file_p->drv->cache_size) {
        res = file_p->drv->seek_cb(file_p->drv, file_p->file_d, file_p->cache->file_position, LV_FS_SEEK_SET);
        if(res != LV_FS_RES_OK) return res;
    }

    uint32_t bw_tmp = 0;
    res = file_p->drv->write_cb(file_p->drv, file_p->file_d, buf, btw, &bw_tmp);
    if(bw != NULL) *bw = bw_tmp;

    if(file_p->drv->cache_size && res == LV_FS_RES_OK){
		uint32_t start_position = file_p->cache->file_position;
		uint32_t end_position = file_p->cache->file_position + bw_tmp - 1;
		char * cache_buffer = file_p->cache->buffer;
		char * write_buffer = buf;
		
		if(start_position <= file_p->cache->start && end_position >= file_p->cache->end){
			lv_memcpy(  cache_buffer, 
						write_buffer + file_p->cache->start - start_position, 
						file_p->cache->end + 1 - file_p->cache->start);
		}
		else if(start_position >= file_p->cache->start && end_position <= file_p->cache->end){
			lv_memcpy(  cache_buffer + start_position - file_p->cache->start, 
						write_buffer, 
						end_position + 1 - start_position);
		}
		else if(end_position >= file_p->cache->start && end_position <= file_p->cache->end){
			lv_memcpy(  cache_buffer, 
						write_buffer + file_p->cache->start - start_position, 
						end_position + 1 - file_p->cache->start);
		}
		else if(start_position >= file_p->cache->start && start_position <= file_p->cache->end){
			lv_memcpy(  cache_buffer + start_position - file_p->cache->start, 
						write_buffer, 
						file_p->cache->end + 1 - start_position);
		}

		file_p->cache->file_position += bw_tmp;
	}

    return res;
}

```

kisvegabor, 2024-01-04:
Let's continue in the PR: #5172

closed 2024-01-04 by kisvegabor