lvgl/lvgl #5167: Use a JPEG-Binary as an Image-Source

dasarne, 2024-01-03:
### LVGL version

v8.2.0

### What happened?

If I pass a C array with a JPEG image as source to the `lv_img_set_src` method, the JPEG image is not decoded, but an attempt is made to display it as an array of pixels.

**Bug**
Passing a C array with a JPEG image results in a display of the bytes as pixels.

**Expectation**
Passing a C array with a JPEG image will result in a correct representation of the image.

**Documentation**
It must be said that the instructions for `lv_img_set_src` leave no doubt that JPEGs cannot be decoded, but that the OnlineConverter must be used. A change in the behavior of `lv_img_set_src` should therefore also lead to a change of the documentation of this function. 

### How to reproduce?

``` c
/**
 ******************************************************************************
 * @file    main.c
 * @author  Arne von Irmer
 * @version V1.0
 * @date    03-January-2024
 * @brief   Main function to easily test the passing of a C array with a JPEG to lv_img_set_src. 
 ******************************************************************************
 */
#include "lvgl.h"
#include "app_hal.h"

static const char peace_hand_map[] = {
    0xff, 0xd8, 0xff, 0xe0, 0x00, 0x10, 0x4a, 0x46, 0x49, 0x46, 0x00, 0x01, 0x02, 0x00, 0x00, 0x64, /*......JFIF.....d*/
    0x00, 0x64, 0x00, 0x00, 0xff, 0xfe, 0x00, 0x12, 0x41, 0x64, 0x6f, 0x62, 0x65, 0x20, 0x49, 0x6d, /*.d......Adobe Im*/
    0x61, 0x67, 0x65, 0x52, 0x65, 0x61, 0x64, 0x79, 0xff, 0xec, 0x00, 0x11, 0x44, 0x75, 0x63, 0x6b, /*ageReady....Duck*/
    0x79, 0x00, 0x01, 0x00, 0x04, 0x00, 0x00, 0x00, 0x3c, 0x00, 0x00, 0xff, 0xee, 0x00, 0x0e, 0x41, /*y.......<......A*/
    0x64, 0x6f, 0x62, 0x65, 0x00, 0x64, 0xc0, 0x00, 0x00, 0x00, 0x01, 0xff, 0xdb, 0x00, 0x84, 0x00, /*dobe.d..........*/
    0x06, 0x04, 0x04, 0x04, 0x05, 0x04, 0x06, 0x05, 0x05, 0x06, 0x09, 0x06, 0x05, 0x06, 0x09, 0x0b, /*................*/
    0x08, 0x06, 0x06, 0x08, 0x0b, 0x0c, 0x0a, 0x0a, 0x0b, 0x0a, 0x0a, 0x0c, 0x10, 0x0c, 0x0c, 0x0c, /*................*/
    0x0c, 0x0c, 0x0c, 0x10, 0x0c, 0x0e, 0x0f, 0x10, 0x0f, 0x0e, 0x0c, 0x13, 0x13, 0x14, 0x14, 0x13, /*................*/
    0x13, 0x1c, 0x1b, 0x1b, 0x1b, 0x1c, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, /*................*/
    0x01, 0x07, 0x07, 0x07, 0x0d, 0x0c, 0x0d, 0x18, 0x10, 0x10, 0x18, 0x1a, 0x15, 0x11, 0x15, 0x1a, /*................*/
    0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, /*................*/
    0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, /*................*/
    0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, /*................*/
    0x1f, 0xff, 0xc0, 0x00, 0x11, 0x08, 0x00, 0x67, 0x00, 0x46, 0x03, 0x01, 0x11, 0x00, 0x02, 0x11, /*.......g.F......*/
    0x01, 0x03, 0x11, 0x01, 0xff, 0xc4, 0x00, 0x75, 0x00, 0x00, 0x02, 0x02, 0x03, 0x01, 0x01, 0x00, /*.......u........*/
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x06, 0x04, 0x05, 0x02, 0x03, 0x07, /*................*/
    0x01, 0x08, 0x01, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, /*................*/
    0x00, 0x00, 0x00, 0x00, 0x10, 0x00, 0x01, 0x04, 0x02, 0x01, 0x02, 0x03, 0x05, 0x05, 0x07, 0x05, /*................*/
    0x01, 0x00, 0x00, 0x00, 0x00, 0x02, 0x01, 0x03, 0x04, 0x05, 0x00, 0x06, 0x11, 0x21, 0x12, 0x31, /*.............!.1*/
    0x13, 0x07, 0x41, 0x51, 0x61, 0x81, 0x22, 0x71, 0x91, 0x72, 0x82, 0x14, 0x32, 0x42, 0x62, 0x23, /*..AQa."q.r..2Bb#*/
    0x33, 0x43, 0x73, 0x52, 0x53, 0x83, 0xa3, 0x24, 0x15, 0x11, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, /*3CsRS..$........*/
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xff, 0xda, 0x00, 0x0c, 0x03, /*................*/
    0x01, 0x00, 0x02, 0x11, 0x03, 0x11, 0x00, 0x3f, 0x00, 0xfa, 0xa7, 0x00, 0xc0, 0xe6, 0xc1, 0xeb, /*.......?........*/
    0x53, 0x0a, 0x09, 0x68, 0xee, 0xb7, 0x68, 0xce, 0x9f, 0xe7, 0x79, 0x2b, 0xb4, 0x18, 0xb3, 0xfa, /*S..h..h...y+....*/
    0x74, 0x05, 0x35, 0x6d, 0x24, 0x2b, 0x28, 0xe2, 0xc8, 0x46, 0x3b, 0xbf, 0xb9, 0xd9, 0xc2, 0x27, /*t.5m$+(..F;....'*/
    0x55, 0xe9, 0x81, 0xd2, 0x51, 0x50, 0x91, 0x08, 0x57, 0x94, 0x5e, 0xa8, 0xa9, 0xe0, 0xa9, 0x80, /*U...QP..W.^.....*/
    0x60, 0x18, 0x06, 0x01, 0x80, 0x60, 0x18, 0x11, 0xad, 0x23, 0xb9, 0x26, 0xb2, 0x5c, 0x66, 0x89, /*`....`...#.&.\f.*/
    0x41, 0xc7, 0x99, 0x71, 0xb0, 0x24, 0x55, 0x45, 0x42, 0x30, 0x54, 0x45, 0x45, 0x4e, 0xa9, 0xe3, /*A..q.$UEB0TEEN..*/
    0x80, 0xb1, 0xe9, 0x8b, 0x10, 0xe6, 0x7a, 0x55, 0xaf, 0xc2, 0x79, 0xb1, 0x76, 0x3a, 0xd5, 0xb3, /*......zU..y.v:..*/
    0x0e, 0x5b, 0x06, 0x89, 0xda, 0x4a, 0xdb, 0x5e, 0x4b, 0xe0, 0x63, 0xf8, 0x84, 0x90, 0x93, 0x03, /*.[...J.^K.c.....*/
    0x2f, 0x4d, 0xa5, 0x48, 0x66, 0xaa, 0x56, 0xb7, 0x39, 0xe5, 0x76, 0xc7, 0x5a, 0x90, 0x50, 0x0c, /*/M.Hf.V.9.v.Z.P.*/
    0xcd, 0x7e, 0xb7, 0x22, 0xf1, 0xe6, 0x42, 0x75, 0x7d, 0xfd, 0xf1, 0xc8, 0x11, 0x57, 0xda, 0x42, /*.~."..Bu}....W.B*/
    0x58, 0x0d, 0xd8, 0x06, 0x05, 0x4e, 0xd7, 0xb2, 0xc1, 0xd6, 0xa8, 0xa4, 0xdb, 0xcc, 0x13, 0x74, /*X....N.........t*/
    0x19, 0xed, 0x06, 0x63, 0x32, 0x9d, 0xcf, 0x3e, 0xf3, 0xa4, 0x8d, 0xb4, 0xcb, 0x43, 0xed, 0x37, /*...c2..>.....C.7*/
    0x0c, 0x90, 0x53, 0x01, 0x59, 0x8d, 0xd3, 0x7c, 0xae, 0xb3, 0xaa, 0x1d, 0xaa, 0x82, 0x1c, 0x3a, /*..S.Y..|.......:*/
    0xbb, 0xa9, 0x21, 0x0a, 0x3b, 0x90, 0x66, 0x14, 0x97, 0xe2, 0xc8, 0x78, 0x48, 0x9b, 0x09, 0x20, /*..!.;.f....xH.. */
    0x6d, 0x34, 0x24, 0x85, 0xdb, 0xda, 0xa4, 0xd1, 0x2f, 0x0b, 0xf0, 0xc0, 0x7f, 0xc0, 0x30, 0x0c, /*m4$...../.....0.*/
[... 136 lines cut ...]
}

int main(void)
{
  lv_init();

  hal_setup();

  lv_disp_t *dispp = lv_disp_get_default();
  lv_theme_t *theme = lv_theme_default_init(dispp, lv_palette_main(LV_PALETTE_BLUE), lv_palette_main(LV_PALETTE_RED),
                                            false, LV_FONT_DEFAULT);
  lv_disp_set_theme(dispp, theme);
  ui_TestJPGScreen_screen_init();
  lv_obj_t *ui____initial_actions0 = lv_obj_create(NULL);

  lv_disp_load_scr(ui_TestJPGScreen);

  lv_img_set_src(ui_testJPG, &peace_hand);

  hal_loop();
}
```

This is the image to be displayed:
[image]

XuNeo, 2024-01-03:
`LV_IMG_CF_TRUE_COLOR` means it's already decoded. 
You have a try to use `LV_IMG_CF_RAW` instead.

dasarne, 2024-01-03:
In `lv_conf.h`  I set
```c
#define LV_USE_SJPG 1
```
And I enabled logging via 
```c
#define LV_USE_LOG 1
#define LV_LOG_LEVEL LV_LOG_LEVEL_TRACE
#define LV_LOG_PRINTF 1
```
But since there was no output at all at a log level "Trace", I became suspicious. And indeed, `lv_conf.h` does not seem to be included anywhere. If I intentionally insert a syntax error into the file, the source still compiles cleanly. 🤔
According to the instructions, I have placed `lv_conf.h`  as a copy of `lv_conf_template.h` next to the lvgl folder, i.e. under `.pio/libdeps/emulator_64bits/lv_conf.h`. 
[image]

Then I activated it using `#if 1` and made the changes above. 
If `lv_conf.h` wasn't compiled, it makes sense that JPEGs are not converted. 🥵
What am I doing wrong?
Here is the mentioned [lv_conf.h](https://github.com/lvgl/lvgl/files/13820650/lv_conf.h.md)

XuNeo, 2024-01-03:
I'm not familiar with `pio`, you can refer to[ this repo](https://github.com/rzeldent/esp32-smartdisplay) of how setting up build flags. And check which lv_conf.h in your project has been used.

dasarne, 2024-01-03:
Ok, the [Pio-Projekt](https://github.com/lvgl/lv_platformio) is years old and therefore looks unmaintained. 🤨 
You should remove it from your documentation.

**Solved**
Switching to the [vscode port of the simulator](https://github.com/lvgl/lv_port_pc_vscode) solves the problem.

For those who are searching for an example that works in the vscode port of the simulator, here is the modified main.c:
```c
/**
 ******************************************************************************
 * @file    main.c
 * @author  Ac6
 * @version V1.0
 * @date    01-December-2013
 * @brief   Default main function.
 ******************************************************************************
 */
#include "lvgl.h"
#include "lv_conf.h"
#include <pthread.h>

/**********************
 *  STATIC PROTOTYPES
 **********************/
static void hal_init(void);
static void hal_loop(void);
static void hal_deinit(void);
static void* tick_thread(void *data);

/**********************
 *  STATIC VARIABLES
 **********************/
static pthread_t thr_tick;    /* thread */
static bool end_tick = false; /* flag to terminate thread */

#if USE_SDL
  #define SDL_MAIN_HANDLED /*To fix SDL's "undefined reference to WinMain" issue*/
  #include <SDL2/SDL.h>
  #include "lv_drivers/sdl/sdl.h"
#elif USE_X11
  #include "lv_drivers/x11/x11.h"
#endif

void ui_TestJPGScreen_screen_init(void);

static const uint8_t peace_hand_map[] = {
    0xff, 0xd8, 0xff, 0xe0, 0x00, 0x10, 0x4a, 0x46, 0x49, 0x46, 0x00, 0x01, 0x02, 0x00, 0x00, 0x64, /*......JFIF.....d*/
    0x00, 0x64, 0x00, 0x00, 0xff, 0xfe, 0x00, 0x12, 0x41, 0x64, 0x6f, 0x62, 0x65, 0x20, 0x49, 0x6d, /*.d......Adobe Im*/
    0x61, 0x67, 0x65, 0x52, 0x65, 0x61, 0x64, 0x79, 0xff, 0xec, 0x00, 0x11, 0x44, 0x75, 0x63, 0x6b, /*ageReady....Duck*/
    0x79, 0x00, 0x01, 0x00, 0x04, 0x00, 0x00, 0x00, 0x3c, 0x00, 0x00, 0xff, 0xee, 0x00, 0x0e, 0x41, /*y.......<......A*/
    0x64, 0x6f, 0x62, 0x65, 0x00, 0x64, 0xc0, 0x00, 0x00, 0x00, 0x01, 0xff, 0xdb, 0x00, 0x84, 0x00, /*dobe.d..........*/
    0x06, 0x04, 0x04, 0x04, 0x05, 0x04, 0x06, 0x05, 0x05, 0x06, 0x09, 0x06, 0x05, 0x06, 0x09, 0x0b, /*................*/
    0x08, 0x06, 0x06, 0x08, 0x0b, 0x0c, 0x0a, 0x0a, 0x0b, 0x0a, 0x0a, 0x0c, 0x10, 0x0c, 0x0c, 0x0c, /*................*/
    0x0c, 0x0c, 0x0c, 0x10, 0x0c, 0x0e, 0x0f, 0x10, 0x0f, 0x0e, 0x0c, 0x13, 0x13, 0x14, 0x14, 0x13, /*................*/
    0x13, 0x1c, 0x1b, 0x1b, 0x1b, 0x1c, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, /*................*/
    0x01, 0x07, 0x07, 0x07, 0x0d, 0x0c, 0x0d, 0x18, 0x10, 0x10, 0x18, 0x1a, 0x15, 0x11, 0x15, 0x1a, /*................*/
    0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, /*................*/
    0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, /*................*/
    0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, 0x1f, /*................*/
    0x1f, 0xff, 0xc0, 0x00, 0x11, 0x08, 0x00, 0x67, 0x00, 0x46, 0x03, 0x01, 0x11, 0x00, 0x02, 0x11, /*.......g.F......*/
    0x01, 0x03, 0x11, 0x01, 0xff, 0xc4, 0x00, 0x75, 0x00, 0x00, 0x02, 0x02, 0x03, 0x01, 0x01, 0x00, /*.......u........*/
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x06, 0x04, 0x05, 0x02, 0x03, 0x07, /*................*/
    0x01, 0x08, 0x01, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, /*................*/
    0x00, 0x00, 0x00, 0x00, 0x10, 0x00, 0x01, 0x04, 0x02, 0x01, 0x02, 0x03, 0x05, 0x05, 0x07, 0x05, /*................*/
    0x01, 0x00, 0x00, 0x00, 0x00, 0x02, 0x01, 0x03, 0x04, 0x05, 0x00, 0x06, 0x11, 0x21, 0x12, 0x31, /*.............!.1*/
    0x13, 0x07, 0x41, 0x51, 0x61, 0x81, 0x22, 0x71, 0x91, 0x72, 0x82, 0x14, 0x32, 0x42, 0x62, 0x23, /*..AQa."q.r..2Bb#*/
    0x33, 0x43, 0x73, 0x52, 0x53, 0x83, 0xa3, 0x24, 0x15, 0x11, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, /*3CsRS..$........*/
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xff, 0xda, 0x00, 0x0c, 0x03, /*................*/
    0x01, 0x00, 0x02, 0x11, 0x03, 0x11, 0x00, 0x3f, 0x00, 0xfa, 0xa7, 0x00, 0xc0, 0xe6, 0xc1, 0xeb, /*.......?........*/
    0x53, 0x0a, 0x09, 0x68, 0xee, 0xb7, 0x68, 0xce, 0x9f, 0xe7, 0x79, 0x2b, 0xb4, 0x18, 0xb3, 0xfa, /*S..h..h...y+....*/
[... 303 lines cut ...]
{
  (void)data;

  while (!end_tick)
  {
    usleep(5000);
    lv_tick_inc(5); /*Tell LittelvGL that 5 milliseconds were elapsed*/
  }

  return NULL;
}

static void hal_loop(void)
{
  while (1)
  {
    /* Periodically call the lv_task handler.
     * It could be done in a timer interrupt or an OS task too.*/
    lv_timer_handler();
    usleep(5 * 1000);
  }

  hal_deinit();
}
```

closed 2024-01-03 by dasarne

Onzeisnottwice, 2024-02-09:
I think I am running in the same issue on Arduino IDE that the lv_conf.h file is not being used. Have you found a solution?

kisvegabor, 2024-02-11:
>  the lv_conf.h file is not being used. 

How do you mean it?

---

In the original issue I think the problem was that `LV_USE_SJPG` was removed in LVGL v9 and `LV_USE_TJPGD` shall be used instead.

Onzeisnottwice, 2024-02-11:
Well, the original lv_conv.h is placed within the lvgl folder. However, Arduino does not use this one. Once you put it into the src folder within LVGL than it is being used. It took me quite some time to find this out.
At least, this is my experience.

kisvegabor, 2024-02-12:
`lv_conf.h` should be placed into Arduino's library folder. See the docs [here](https://docs.lvgl.io/master/integration/framework/arduino.html#configure-lvgl).

Onzeisnottwice, 2024-02-12:
I first followed that direction. I my effort to simply display a png file from SD card, which I btw still have not been able to to, I dived into this config file.  I noticed that it is not being used if it is placed in the lvgl folder. Try it yourself and put some illegal code in the file and see if it is compiled. Until I moved it into the SRC file I noticed that this config file is finally being used.

kisvegabor, 2024-02-12:
>  I noticed that it is not being used if it is placed in the lvgl folder. 

It shouldn't be in the `lvgl` folder, but next to it in the `libraries` folder.