lvgl/lvgl #5168: Generating a JSON file that has the entire LVGL API mapped.

kdschlosser, 2024-01-03:
### Introduce the problem

There are 2 different issues that would be addressed with adding this feature.

It is become clear that developing a binding is challenging considering one of the things that is wanting to be done it the binding code gets automatically generated, I have written a python script that will read the header files in LVGL and it will produce a JSON file containing everything that should be needed to generate binding code.

The second portion is documentation. I have also included the docstrings that are located in the header files as well.

### Proposal

What I need to know if how we want the format of the JSON data set up. What information is needed?

This is what I have currently

This is for a function
```json
{
        "name": "lv_obj_update_flag",
        "args": [
            {
                "is_pointer": true,
                "is_const": false,
                "type": {
                    "name": "lv_obj_t",
                    "json_type": "lv_type"
                },
                "json_type": "arg",
                "name": "obj",
                "docstring": "pointer to an object "
            },
            {
                "is_const": false,
                "is_pointer": false,
                "type": {
                    "name": "lv_obj_flag_t",
                    "json_type": "lv_type"
                },
                "json_type": "arg",
                "name": "f",
                "docstring": "OR-ed values from lv_obj_flag_t to update. "
            },
            {
                "is_const": false,
                "is_pointer": false,
                "type": {
                    "name": "bool",
                    "json_type": "primitive_type"
                },
                "json_type": "arg",
                "name": "v",
                "docstring": "true: add the flags; false: remove the flags "
            }
        ],
        "type": {
            "is_const": false,
            "is_pointer": false,
            "type": {
                "name": "void",
                "json_type": "primitive_type"
            },
            "json_type": "retval",
            "docstring": null
        },
        "json_type": "function",
        "docstring": "Set add or remove one or more flags. "
    },
```

and a structure/union
```json
[... 209 lines cut ...]
            }
        ],
        "json_type": "enum",
        "docstring": "Chart types ",
        "is_const": false,
        "is_pointer": false
    },
```

typedef

```json
{
        "name": "lv_switch_t",
        "is_const": false,
        "is_pointer": false,
        "type": {
            "name": "_lv_switch_t",
            "json_type": "lv_type"
        },
        "json_type": "typedef"
    },
```

What are the thoughts or ideas?

XuNeo, 2024-01-03:
This looks really promising. 
Providing the json file is good for human reading. I'm also curious about the python script. Is it easy to use directly?

Only one question, how can we decide which APIs/Types to export? 
I guess APIs can be separated to several groups. 
- The core APIs that all bindings will need like lv_event_t related. 
- Widget APIs
- Low level APIs like how to setup display; how to draw a widget.

kdschlosser, 2024-01-03:
The script is as simple to use as

```
python3 gen_json.py /path/to/output/file/to
```

tit does everything that is needed to read the header files to collect all of the data that is needed to build the API and at the same time it leverages doxygen to collect the documentation for the items.

When I upgraded the documentation build system earlier this year I had written it in a manner that would allow me to load a portion of it from outside of the build system for the documentation. By being able to do that I am able to access the portion of the code that launches doxygen and provides me a way to access the stored docstrings in the header files.

kdschlosser, 2024-01-03:
The organization of the JSON is going to be an array of elements. that array is in the same exact order that the compiler compiles LVGL in. The C compiler is used to generate the preprocesor output and that output is what gets read to generate the JSON data. Each element is added in the same order in which it would be compiled.

There is some substitution that has been done for types that are forward declared.  Like functions that have _lv_obj_t being used before it has actually been created. If I left those strructures in place they would show up as fieldless structures as a function argument. because I know that LVGL would get compiled prior to any binding code I have replaced those declarations with a type name because that is what would be accessed. The same thing goes for enumerations and the type that gets set for the enumeration. 

The type that gets set for an enumeration is compiler dependent. Typically the type is an integer. In LVGL you see only a few places that an actual enum being used as a type. In most cases there is a typedef that represents an enum that is made and it is not of the type "int". Thios is to control memory use. Downside of doing this is there is no link between function arguments and enums because they are not actually tied together in any way. So what I did there is I stored the generated enums and when I cam across a typedef I had it look to see if the type had been declared with a preceding underscore. If there was and it was of the type enum then the enum members had their type changes to the name of the typedef.

kdschlosser, 2024-01-03:
I could separate things into different containers. Like "structures", "unions", "functions", etc.... If that would be easier to handle.

Macros is where I am having a bit of an issue. This is because macros are simple text replacement so when I generate the preprocessor output there are no macros because the text replacement has been done. The macros do not show up for me to read them.. That being said I can pull the macro names from doxygen. I have to look into how much information there is available for me to use for it to be added to the JSON output.

kisvegabor, 2024-01-04:
cc @nia-e @rafaelcaricio (from the Rust binding)

XuNeo, 2024-01-04:
> ```
> python3 gen_json.py /path/to/output/file/to
> ```

I wonder if I can use the `gen_json.py` directly. For example
```python
#  file: gen_lua_binding.py
 import lvgl_gen_json
```

ggoraa, 2024-01-04:
why pycparser btw? why not tree-sitter for Python? :D

kdschlosser, 2024-01-04:
because it's written in pure python and I am able to patch it in a very easy manner to make it behave the way I want. in 1500 lines of code I am to turn 55,572 lines of C code that is spread across 240 header files into 92,630 lines of JSON data.

The 1500 lines is going to be quite a bit smaller as this is my first run through it. I am doing to be able to condense those lines once I add the ability to look backwards through the tree from the outer most branches. This will allow me to do some pruning so to speak to remove the nesting and flatten the tree leaving me with only what is needed to build the JSON. 

It is amazing how deep that nesting is for something as simple as a field definition in a structure and most of the steps through the nest is something I do not need to care about at all if I have the ability to look backwards through the levels. I do this by subclassing the different classes in pycparser and overriding the parent classes in pycparser with the child classes I have created.

kdschlosser, 2024-01-04:
ok I got the code footprint down to 900 lines. I still have to add in collecting the docstrings but that is not going to add a large amount to it. I am thinking I should be somewhere around 1000 to 1050 or so. cut it down by 500 lines.

Here is the output without the docstrings.

[lvgl.zip](https://github.com/lvgl/lvgl/files/13835864/lvgl.zip)

kisvegabor, 2024-01-05:
In
```json
           "name": "_lv_anim_t",
            "json_type": "<class 'pycparser_monkeypatch.Decl'>",
            "type": {
                "name": "_lv_anim_t",
                "json_type": "lv_type"
            }
```
is `"<class 'pycparser_monkeypatch.Decl'>"` intentional?

kdschlosser, 2024-01-05:
It is internal. This is how I am identifying things that need to be corrected. The class that is holding that data is `Decl` which is being held by a `Typedef` class. in that case I need to have the Typedef class check to see if the class it is holding is a Decl class and if the names that are in those 2 classes match then propagate the data downwards and change the "json_type" to "typedef".  If that JSON example is in one of the root level elements and not nested up the tree at all then the data is properly being moved I just forgot to tell it to change the json_type field to typedef. If the data is a child of an element that has a json_type of typedef then the data is not being relocated either because I have not written the doce to tell it to do so or there is ome edge case that might be causing it to not be moved.

So this is the class arrangement for that specific structure.

```
Typedef(
      name='lv_anim_t',
      quals=[],
      storage=['typedef'],
      type=TypeDecl(
        declname='lv_anim_t',
        quals=[],
        align=None,
        type=Struct(
          name='_lv_anim_t',
          decls=[
            Decl(
              name='var',
              quals=[],
              align=[],
              storage=[],
              funcspec=[],
              type=PtrDecl(
                quals=[],
                type=TypeDecl(
                  declname='var',
                  quals=[],
                  align=None,
                  type=IdentifierType(names=['void'])
                )
              ),
              init=None,
              bitsize=None
            ),
            Decl(
              name='exec_cb',
              quals=[],
              align=[],
              storage=[],
              funcspec=[],
              type=TypeDecl(
                declname='exec_cb',
                quals=[],
                align=None,
                type=IdentifierType(names=['lv_anim_exec_xcb_t'])
              ),
              init=None,
              bitsize=None
            ),
            Decl(
              name='custom_exec_cb',
              quals=[],
              align=[],
              storage=[],
              funcspec=[],
              type=TypeDecl(
                declname='custom_exec_cb',
                quals=[],
                align=None,
                type=IdentifierType(names=['lv_anim_custom_exec_cb_t'])
              ),
              init=None,
              bitsize=None
            ),
            Decl(
              name='start_cb',
              quals=[],
              align=[],
              storage=[],
              funcspec=[],
[... 432 lines cut ...]
          )
        )
      )
    ),
```

specifically here

```
                type=TypeDecl(
                    declname=None,
                    quals=[],
                    align=None,
                    type=Struct(
                      name='_lv_anim_t',
                      decls=None
                    )
                  )
```

So now If I make it so that a TypeDecl look at what it is holding and sees it is holding a structure and then acts on that it is wrong. Because this is a forward declaration that is being done. In order to know that if the TypeDecl looks backwards 2 levels it would see that the "grandparent" so to speak is a `Typename` which means it is a forward declaration.

pycparser does not have built into it the passing of the parent object so there is no way to look backwards. The only way would be to load the entire things and then do recursive iteration over the entire tree to try and locate things. This gets done over and over again for each and every single object in the tree. Take a look at the gen_mpy.py file in the MicroPython binding and you will see what that ends up looking like to be able to work around this single "flaw" in the design of pycparser. This same exact flaw also exists in tree-sitter except I am not able to in place patch the code at runtime with tree-sitter because it is compiled C code and not runtime code. Having to do recursive iteration over the entire tree for each and every single object in the tree in order to build the json data is completely mind boggling. It's bad enough that I have to recurse the entire tree 2 times in order to handle this because of how the class instances are created....... OOoOO... that just gave me an idea on how to to do so that no iterations take place at all. I can do it as the classes are being constructed by pycparser. this would take place "realtime" as the code is being read by pycparser. Metaclasses to the rescue!!! I am going to override the instance creation of the classes so parent instances get populated into the children by constructing the classes that are being used in a dynamic manner and seting the parent as a class level variable. Then in the init methods of the classes I can look forward because the actual construction is done from the outer most first and with the addition of a method I can call that method in the children from the parents init method so the child will then have the ability to look backwards because it know what it has been assigned to a parent. I am pretty sure that will work and it will also be faster with less code.

Downside, it is what I call voodoo magic code. a data path is not going to be able to be followed through it by an IDE and it is not something that is going to be easily understood by looking at the code.

ggoraa, 2024-01-06:
> This same exact flaw also exists in tree-sitter

ggoraa, 2024-01-07:
@kdschlosser what about changing for example something like `"json_type": "stdlib_type"` to `"json_type": "stdlib"`? Because like why add the redundant `_type` at the end of each identifier? :)

kdschlosser, 2024-01-07:
The JSON API thing has not yet been added to LVGL. It will be I am pretty sure. I have not finished working on it yet. I have a bunch of projects I am working on so it will be a few more days before I get it finished up.

ggoraa, 2024-01-08:
For now I quite like how the JSON is made, to be honest. The only thing bothering me are those pesky <class "pycparsersomething">, but that's just a bug, right?

kdschlosser, 2024-01-08:
I am waiting on some changes being made in LVGL to forward declare structures and structure typedefs. once that is finished up I will be able to to handle those funky json_types. Those are not a bug, they are there to let me know that I need to do more work in the organization of the file. It is more of a way of showing me that there is an issue. If they show up then the code that generates the JSON data needs to be updated.

As a note, I am also going to be adding an option to output the JSON data to stdout. This way the output is able to be captured and used without having to deal with a file. This is beneficial if using Make or CMake.

as a note.. There are only 3 compilers that will work with the JSON generator. That is gcc, MSVC and clang. one of those MUST be available in order to generate the JSON data. This is because not all compilers output preprocessor code in the same manner and there are things that pycparser will pitch a fit about.

kdschlosser, 2024-01-08:
For the most part the structure of the JSON data is going to remain the same. The elements where you see that funky json_type are going to be removed and the branch on the tree will be shifted down one level. That is why I am saying it is best to hold off for the moment on writing code that uses the JSON file.  I don't want to make a heap of work for ya if you have to change your code.

kisvegabor, 2024-01-08:
> I am waiting on some changes being made in LVGL to forward declare structures and structure typedefs.

Probably I was lost somewhere. Exactly what changes are required?

ggoraa, 2024-01-10:
@kdschlosser will you write documentation on the file schema? bc it's not that easily understandable without docs

kdschlosser, 2024-01-10:
I already took care of the anonymous structures. everything is all fixed with the JSON generation. All docstrings pull in properly and the structure looks good. No more of those "pycparser" types. I also added in the macros but unfortunately the amount of information I am able to collect on those is pretty limited as I am dealing with preprocesor output which there are no macro definitions that are made available. I am using doxygen to collect the information I have. I have to dig into the doxygen XML and see if there is any additional information I might be able to pull out. Things like parameters if there are any. 

Here is what the final API looks like. 
[lvgl.zip](https://github.com/lvgl/lvgl/files/13888620/lvgl.zip)

The unions section is empty because there are no named unions only anonymous ones. I still have the script parse for them in the even a union gets added. There are unions declared inside of structures.

I also added a section for forward declarations to handle the empty structures. Function pointers also have their own json_type of "function_pointer"

Everything that would be needed to construct a binding is available in the JSON file right down to being able to create API documentation for a binding. 

and yes there are typedefs in LVGL that have the same name as the structure. that will be seen in the JSON output

kdschlosser, 2024-01-10:
@kisvegabor 

I was waiting for you to finish u p with the forward declarations of the typedefs and structs. There are places in the code where there are still _struct _lv_* types used as function parameters and/or structure fields and the type has a forward declaration.

ggoraa, 2024-01-10:
[image]
[image]

Well that looks funny

kdschlosser, 2024-01-10:
oopsie I must have a typo.

kdschlosser, 2024-01-10:
give this a try

[lvgl.zip](https://github.com/lvgl/lvgl/files/13890937/lvgl.zip)

ggoraa, 2024-01-10:
I also suggest adding a schema version number at the start of the file, ~~so it's easier to make breaking changes lmao~~

kdschlosser, 2024-01-10:
Hey this is not a release. I did say that to you didn't I? I told you it was going to change and I wouldn't be developing for it yet. You decided to have a go at it. This should be the final version of it. I believe everything is properly added and it provides the information that is needed in order to generate binding code using it. If the JSON is not able to be used for some reason there is access to the underlying code that reads the header files.

ggoraa, 2024-01-10:
Hey, hey, okay, sorry man, didn't want to make you mad, really

kdschlosser, 2024-01-10:
I'm not mad.  I'm just saying that I did let you know that it wasn't intended to be used yet. Just reminding you I did say that :wink:

closed 2024-01-12 by kdschlosser

kisvegabor, 2024-01-12:
Why is it closed?

ggoraa, 2024-01-15:
@kdschlosser what happened? 🤔

ggoraa, 2024-01-24:
please don't say that you ditched this...

ggoraa, 2024-01-26:
@kisvegabor do you know what happened?

kisvegabor, 2024-01-29:
I don't know it either. I've have already written to him on other channels, but no answer so far. I hope he is well... :slightly_frowning_face: