lvgl/lvgl #5173: The demo `lv_demo_music `shows an exception when the color is 24 bits or 16 bits,

IAMMX, 2024-01-04:
### LVGL version

v9.0.0

### What happened?

The demo `lv_demo_music `shows an exception when the color is 24 bits or 16 bits, and is displayed normally when it is 32 bits. May I ask if anyone can repair it?
### #define LV_COLOR_DEPTH 32
[image]
### #define LV_COLOR_DEPTH 24
[image]
### #define LV_COLOR_DEPTH 16
[image]

IAMMX, 2024-01-05:
@kisvegabor May I ask if this issue can be addressed?

IAMMX, 2024-01-07:
@kisvegabor Can you look at this issue?

kisvegabor, 2024-01-11:
Hi,

Probably I will remove the music player demo from v9 because:
1. It's a shame but the original assets were lost and I can covert new C array from them for v9
2. Due to some changes in v9, it would be quite complicated to implement the jumping strides.

XuNeo, 2024-01-11:
> It's a shame but the original assets were lost and I can covert new C array from them for v9

That's too unfortunate.
The LVGLImage.py can convert LVGL bin image back to PNG if you want.

```python

def test():
    f = "pngs/cogwheel.RGB565A8.png"
    img = LVGLImage().from_bin("output/cogwheel.RGB888.bin")
    img.to_png("output/cogwheel.RGB888.png.png")  # convert back to png
```

kisvegabor, 2024-01-11:
We have only the C arrays. :slightly_frowning_face: 

I could update the C assets by hand too, mainly just refactoring, but RGB565A8 is not supported in SW render anymore, so ARGB8888 needed to be used instead.

The difficulty in re-implementing the jumping strides is bothering me more. The correct solution would be to use vector graphics for those trapezoids but I'm afraid it would be too slow.

IAMMX, 2024-01-11:
It's a pity if the demo of the music player is gone, what a good demo!😭Could you please find some other pictures from the Internet to replace the missing pictures.

kisvegabor, 2024-01-11:
Seeing that it's important for you both, I'll give it an other try. Probably it can be updated only after the release of v9 though. :slightly_frowning_face:

lhdjply, 2024-01-12:
fix it in #5291, let's continue it

PR #5291 (2024-01-12): fix(music): fix image display errors in 16 bit and 24 bit errors
Help us review this PR! Anyone can [approve it or request changes](https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/reviewing-changes-in-pull-requests/approving-a-pull-request-with-required-reviews).

### Description of the feature or fix
fix #5173
fix(music): fix image display errors in 16 bit and 24 bit errors

### Checkpoints
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Test [... cut]
closed 2024-01-12 by kisvegabor