lvgl/lvgl #5174: Support color format ARGB8565 in v9

rabbitsaviola, 2024-01-04:
### Introduce the problem

DMA2D, PXP and VGLite doesn't support color format RGB565A8. If using ARGB8888 instead, more memory is needed. Though RGB565A8 is efficient for sw rendering, but it's not friendly to hw rendering.

### Proposal

Besides RGB565A8, ARGB8565 is also supported by v9.

XuNeo, 2024-01-04:
Hi, 
Can you prepare a PR to add ARGB8565 to lv_color.h? 
```
    /*RGB565 and variants*/
    LV_COLOR_FORMAT_RGB565            = 0x12,
    LV_COLOR_FORMAT_ARGB8565          = 0x13,
    LV_COLOR_FORMAT_RGB565A8          = 0x14    /**< Color array followed by Alpha array*/,

```

kisvegabor, 2024-01-04:
I believe you mean using the ARGB8565 for images, but I wonder what problem `ARGB8565` would solve:
- `ARGB8565` is not supported by GPUs either
- It has the same size as `RGB565A8`

XuNeo, 2024-01-04:
ARGB8565 is supported by some GPUs, for example, the latest version VG-Lite.

kisvegabor, 2024-01-04:
Actually we have introduced `RGB565A8` because it might be better supported by GPUs if the RGB565 part is blended by using an A8 mask. Isn't there a way to use `RGB565A8` like this in VG-Lite? 

@GorgonMeducer, @jeremy-baker  how does it work in Arm2D and Dave2D?

GorgonMeducer, 2024-01-04:
Arm-2D supports RGB565A8. 
It is hard to support ARGB8565 as it is 3bytes for each pixel and not friendly for the 128bit SIMD vector.

jeremy-baker, 2024-01-04:
Dave2D doesn't support ARG8565 or RGB565A8, it supports RGB565 (for RGB565 the Blue is used for Alpha as well)

kisvegabor, 2024-01-04:
These suggest me that RGB565A8 is a better choice. If it can be handled with the same performance in VG-Lite as ARGB8565, I would keep only RGB565A8.

XuNeo, 2024-01-05:
`ARGB8565` has its own benefits. It's can be supported directly by some GPUs. It's a real pixel color format while RGB565A8 is a map format. Thus I thinks it's reasonable to add to lv_color.h. Add support to software renderer however is no needed.

Using A8 mask may have some limitations too. For example, how we deal with multiple masks? Will the blend mode lvgl uses have same result with GPU?

GorgonMeducer, 2024-01-05:
Certainly, let's go over the analysis again in English.

### Analysis of ARGB8565 and RGB565 + Separate Alpha Array in C Language Image Processing

#### ARGB8565 Processing
- In the ARGB8565 format, the Alpha channel is tightly integrated with the color channels, encapsulated within each 24-bit pixel. This requires bit manipulation to extract or modify the Alpha value whenever a pixel is processed. Even simple operations on the Alpha channel necessitate handling the entire 24-bit pixel data.
- Frequent modifications of the Alpha value, such as in scenarios involving fade-in, fade-out, or transparency gradients, might lead to additional computational overhead in this format.

#### RGB565 + Separate Alpha Array Processing
- With this format, the color data (RGB565) and the Alpha data are separate. This allows for independent access and modification of Alpha values without the need for complex bit operations.
- For image processing algorithms, this separation can make handling the Alpha channel more straightforward and efficient, particularly when the color data does not need to be modified simultaneously.

#### Performance Considerations
- **Data Alignment**: The RGB565 + separate Alpha Array might offer better data alignment, especially in embedded systems where alignment significantly impacts processing performance.
- **Processing Complexity**: While integrated formats like ARGB8565 theoretically reduce the number of data accesses (since color and Alpha are in the same data block), the necessity for bit operations to separate these channels might not make it more efficient than a separate format.

#### Conclusion
In the context of C language image processing, particularly when dealing with frequent Alpha channel manipulations, RGB565 with a separate Alpha Array could provide higher efficiency in some scenarios. This is more apparent in embedded environments that favor aligned access and minimize bitwise operations. This highlights the importance of carefully considering specific application needs and processing features when selecting an image format.

kisvegabor, 2024-01-05:
In light of that, to put it really short we can say:
- RGB565A8 for SW render (including SIMD)
- ARGB8565 for GPU

So I'm ok with adding ARGB8565 format to the image converter and we can decide if we want to support it with SW render.

closed 2024-01-28 by lvgl-bot