lvgl/lvgl #5178: Can we add a macro definition for the selection of `bin_decoder`？

lhdjply, 2024-01-04:
### Introduce the problem

@XuNeo 
I find the line 198 in [lv_init.c](https://github.com/lvgl/lvgl/blob/master/src/lv_init.c#L198) ,bin_decoder must be use.
Can we add a macro definition for the selection of `bin_decoder`?
For example add define in `lv_conf.h`
```c
#define LV_USE_BIN_DECODER 0
```

XuNeo, 2024-01-04:
Without bin decoder, the whole draw image will not work. For example, lv_image_set_src will need to consult decoder for image header to get the image w,h etc.

lhdjply, 2024-01-04:
> Without bin decoder, the whole draw image will not work. For example, lv_image_set_src will need to consult decoder for image header to get the image w,h etc.

Okay, I understand. I don't think `lv_bin_decoder. c` should be placed in the lib folder because it is usually optional to use in this folder.

XuNeo, 2024-01-04:
The binary decoder was originally located in lv_image_decoder.c. However, due to its increased complexity, it has been moved to the libs directory alongside other decoders.

closed 2024-01-04 by lhdjply