lvgl/lvgl #5181: The table widget cannot scroll down with the focus when using encoder as input device

Maoqiming, 2024-01-04:
### LVGL version

8.3.11

### What happened?

The table widget cannot scroll down with the focus when using an encoder as the input device,but during simulation, it can.I used breakpoint tracing and found that when the focus moves outside the Table widget window, all conditional judgments in the function static void scroll_ to_ selected_ cell (lv_obj_t * obj) have not been triggered.The mcu what i used is stm32f407.LVGL runs on freeRTOS.The screen is monochrome OLED.

### How to reproduce?

extern lv_indev_t * indev_encoder; 
lv_group_t * group_default;

void AppTaskLVGL(void *argument)
{
	const uint16_t usFrequency = 5; /* delay period 5ms */
	uint32_t tick;

	lv_init(); 
	lv_port_disp_init(); 
	lv_port_indev_init();
	lv_disp_t* disp=lv_disp_get_default();
        lv_theme_t* theme=lv_theme_mono_init(disp /* hue */,false,LV_FONT_DEFAULT /* use LV_FONT_DEFAULT */);
        lv_disp_set_theme(disp,theme);
	
	page_table();

	/* Get current time */
	tick = osKernelGetTickCount(); 
	while(1)
	{
		lv_timer_handler();
		/* relative delay*/
		tick += usFrequency;                          
		osDelayUntil(tick);
	}	
}

void page_table(void)
{
	group_default = lv_group_create();
	lv_indev_set_group(indev_encoder, group_default);
        lv_example_table();
}

static void lv_example_table(void)
{
	 lv_group_t * group_table=group_default;

         lv_obj_t *table = lv_table_create(lv_scr_act());                                        
         lv_obj_center(table);                                                                  

        lv_table_set_cell_value(table, 0, 0, "Name");
        lv_table_set_cell_value(table, 1, 0, "Apple");
        lv_table_set_cell_value(table, 2, 0, "Banana");
        lv_table_set_cell_value(table, 3, 0, "Lemon");
        lv_table_set_cell_value(table, 4, 0, "Grape");
        lv_table_set_cell_value(table, 5, 0, "Melon");
        lv_table_set_cell_value(table, 6, 0, "Peach");
        lv_table_set_cell_value(table, 7, 0, "Nuts");
        lv_table_set_col_width(table, 0, scr_act_width()*3/4);
	lv_group_add_obj(group_table,table);/* 添加到组中 */
}

https://github.com/lvgl/lvgl/assets/36371676/126a86d4-9690-4472-91dc-23d3f3364814

kisvegabor, 2024-01-11:
Hi,

Sorry for the late reply. 

It happens because the size of the table is growing automatically according to the number of rows. It's just clipped because it doesn't fit the screen.

You can add `lv_obj_set_height(table, lv_pct(100));` to limit the height to the parent height, or just set some exact height in pixels.

Maoqiming, 2024-01-11:
Thank you very much for helping me solve this problem. Your help was very timely.

Maoqiming, 2024-01-11:
> Hi,
>
> Sorry for the late reply.
>
> It happens because the size of the table is growing automatically according to the number of rows. It's just clipped because it doesn't fit the screen.
>
> You can add [...]

Thank you very much for helping me solve this problem. Your help was very timely.

kisvegabor, 2024-01-11:
I'm happy it's resolved 🙂

closed 2024-01-11 by kisvegabor