lvgl/lvgl #5182: CI break due to micropython script update

XuNeo, 2024-01-04:
### Introduce the problem

A dirty fix would be https://github.com/lvgl/lv_binding_micropython/pull/309
I don't know why it doesn't report error before. There's a variable in ESP named `del` which is a keyword in python, thus script converted to `_del`.

XuNeo, 2024-01-04:
I guess the original script doesn't correctly generates all bindings. With latest script, it includes some new struct, thus `del` is now included.

XuNeo, 2024-01-04:
Script has been updated. Manually test passes now. 
The issue is that script incorrectly sanitizes the C struct field.

PGNetHun, 2024-01-04:
@XuNeo , I can test this also with unix and esp32-generic port, so please wait merging this.

PGNetHun, 2024-01-04:
> @XuNeo , I can test this also with unix and esp32-generic port, so please wait merging this.

Tested your PR only with unix port, it's OK, no compilation nor runtime errors.

kisvegabor, 2024-01-04:
Merged and the submodule is updated in `lv_micropython.`

Now the CI passes! :tada:

closed 2024-01-04 by kisvegabor