lvgl/lvgl #5187: Indicator area overflow over the bar or slider area in v9

rmshub, 2024-01-04:
### Introduce the problem

Code to reproduce the issue
### Bar
``` 
lv_obj_t *prog = lv_bar_create(design_panel);
lv_obj_set_size(prog, LV_PCT(94), LV_PCT(15));
lv_bar_set_value(prog, 0, LV_ANIM_ON);
lv_obj_set_style_radius(prog, 0, LV_PART_INDICATOR);
lv_obj_set_style_radius(prog, 5, 0);
lv_bar_set_range(prog, 0, 100);
```
### Slider
```
lv_obj_t *slider = lv_slider_create(brightness);
lv_obj_set_size(slider, 400, 50);
lv_slider_set_range(slider, 0, 100);
lv_slider_set_value(slider, lcd_brightness, LV_ANIM_ON);
lv_obj_remove_style(slider, NULL, LV_PART_KNOB); /* Be sure the knob is not displayed */
lv_obj_add_flag(slider, LV_OBJ_FLAG_ADV_HITTEST);
lv_obj_set_style_radius(slider, 10, LV_PART_MAIN | LV_STATE_DEFAULT);
lv_obj_set_style_radius(slider, 0, LV_PART_INDICATOR);
```
Branch: master
[image]
[image]
[image]

hyansuper, 2024-01-07:
@rmshub  Do you mean indicator overflow the rounded corner? but you set it so

> lv_obj_set_style_radius(prog, 0, LV_PART_INDICATOR);
lv_obj_set_style_radius(prog, 5, 0);

you can remove the fist line

rmshub, 2024-01-08:
Yes, indicator overflow at rounded corner. It is happen in v9. Same code doesn't generate overflow at corner in v8.3.11.  I want indicator as straight line while moving.

kisvegabor, 2024-01-08:
I'm working on it and will send a PR still today.

kisvegabor, 2024-01-08:
Should be fixed by #5229. Let's continue in the PR.

PR #5229 (2024-01-09): fix(bar): fix the clipping of the indicator in some corner cases
Help us review this PR! Anyone can [approve it or request changes](https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/reviewing-changes-in-pull-requests/approving-a-pull-request-with-required-reviews).

### Description of the feature or fix

Fixes #5187

On the test images you can see that in some cases we cannot draw border, shadow and outline as the indicator has a strange shape. 

The 2 commits shouldn't be squashed as they are independent.

### Checkpoints
- Update the [Documentation](https://github.com [... cut]
closed 2024-01-08 by kisvegabor