lvgl/lvgl #5188: Button and Label borders are not displayed when enabled

rmshub, 2024-01-04:
### Introduce the problem

### Label Style
```
    lv_style_init(&label_style);
    lv_style_set_width(&label_style, LV_PCT(100));
    lv_style_set_height(&label_style, LV_SIZE_CONTENT);
    lv_style_set_border_width(&label_style, 1);
    lv_style_set_border_side(&label_style, LV_BORDER_SIDE_BOTTOM);
    lv_style_set_text_color(&label_style, lv_palette_main(LV_PALETTE_DEEP_ORANGE));
    lv_style_set_text_font(&label_style, &lv_font_montserrat_26);
```
### Label
```
    lv_obj_t *cnt = lv_label_create(loom_status);
    lv_obj_add_style(weave_cnt, &label_style, 0);
    lv_label_set_text(weave_cnt, "48");
```
[image]

rmshub, 2024-01-04:
### Button Style used
```
    lv_style_init(&btn_style);
    lv_style_set_shadow_width(&btn_style, 0);
    lv_style_set_width(&btn_style, LV_PCT(100));
    lv_style_set_height(&btn_style, LV_PCT(30));
    lv_style_set_bg_opa(&btn_style, LV_OPA_TRANSP);
    lv_style_set_border_opa(&btn_style, LV_OPA_30);
    lv_style_set_border_width(&btn_style, 1);
    lv_style_set_border_side(&btn_style, LV_BORDER_SIDE_BOTTOM);
    lv_style_set_text_color(&btn_style, lv_theme_get_color_primary(NULL));
    lv_style_set_text_font(&btn_style, &lv_font_montserrat_20);

```

C47D, 2024-01-04:
## Label

I've added `lv_style_set_border_color(&label_style, lv_palette_main(LV_PALETTE_RED));` and got this image, is that what you were expecting?

[image]

kisvegabor, 2024-01-04:
Note that, border color is white by default.

rmshub, 2024-01-05:
Thanks for information.

closed 2024-01-05 by rmshub