lvgl/lvgl #5189: couldn't display symbol

codeElf01, 2024-01-04:
### LVGL version

v7.0.2

### What happened?

  ```
/* upload and download icon */
    static lv_obj_t * upload_speed_icon = lv_label_create(main_page,NULL);
    if(upload_speed_icon == NULL) Serial.println("NULL upload_speed_icon");
    lv_label_set_text(upload_speed_icon, LV_SYMBOL_UPLOAD);
    lv_obj_set_pos(upload_speed_icon, 30, 20);

```
    static lv_obj_t * download_speed_icon = lv_label_create(main_page,NULL);
    if(download_speed_icon == NULL) Serial.println("NULL download_speed_icon");
    lv_label_set_text(download_speed_icon, LV_SYMBOL_DOWNLOAD);
    lv_obj_set_pos(download_speed_icon, 100, 20);
   
    /*Create an chart*/
    chart = lv_chart_create(main_page, NULL);
    lv_obj_set_size(chart, 220, 70);
    lv_obj_align(chart, NULL, LV_ALIGN_CENTER, 0, -40);
    lv_chart_set_type(chart, LV_CHART_TYPE_LINE); /*Show lines and points too*/
    lv_chart_set_range(chart, 0, 100);
    lv_chart_set_point_count(chart, 10); 
    lv_chart_set_update_mode(chart, LV_CHART_UPDATE_MODE_SHIFT);

    /*Add a faded are effect*/
    lv_obj_set_style_local_bg_opa(chart, LV_CHART_PART_SERIES, LV_STATE_DEFAULT, LV_OPA_50); /*Max. opa.*/
    lv_obj_set_style_local_bg_grad_dir(chart, LV_CHART_PART_SERIES, LV_STATE_DEFAULT, LV_GRAD_DIR_VER);
    lv_obj_set_style_local_bg_main_stop(chart, LV_CHART_PART_SERIES, LV_STATE_DEFAULT, 255); /*Max opa on the top*/
    lv_obj_set_style_local_bg_grad_stop(chart, LV_CHART_PART_SERIES, LV_STATE_DEFAULT, 0);   /*Transparent on the bottom*/

    /*Add two data series*/
    upload_series = lv_chart_add_series(chart, LV_COLOR_RED);
    download_series = lv_chart_add_series(chart, LV_COLOR_GREEN);`

The symbol coudn't display in the main page, but the chart could refresh . Below the code
`    static lv_obj_t * upload_speed = lv_label_create(main_page,NULL);
    if(upload_speed == NULL) Serial.println("NULL upload_speed");
    lv_label_set_text(upload_speed, "0.00MB/s");
    lv_obj_set_pos(upload_speed, 20, 20);
` 
the 'MB' only show as 'M', the 'B' missed, I don't konw why.

kisvegabor, 2024-01-04:
Regarding the symbol, please be sure that  you IDE/editor has UTF-8 encoding.

codeElf01, 2024-01-05:
I use the visual studio code with the utf-8, but i change the default font @kisvegabor

codeElf01, 2024-01-05:
Thank you answer my question, I have resolved the problem，may be I reset the default font. @kisvegabor

closed 2024-01-05 by codeElf01